OSSystems/meta-browser #447: chromium-ozone-wayland freezes when context menu's are destroyed.

MastaG, 2021-01-25:
Hi there,

I'm having a long standing issue with chromium-ozone-wayland rev: 80256d1fd40e8bbe9e00c4e4c7d5b0210f3bf4bc on my ODroid XU3.
It's running in a wayland session on Gnome Desktop 3.38 (Fedora 33 Armv7).
The ODroid XU3 is using the ARM Mali T-628 wayland/gbm r12p004 driver (the binary blob from the ARM website).

I've built it with the latest openembedded-core / meta-openembeded /meta-clang / bitbake (all from master git) using: TOOLCHAIN ?= "clang"
```
Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "fedora-33"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "odroid-xu3"
DISTRO               = "odroid-os"
DISTRO_VERSION       = "release"
TUNE_FEATURES        = "arm vfp cortexa15 neon vfpv4 callconvention-hard"
TARGET_FPU           = "hard"
```
My flags are:
```
 --use-gl=egl --enable-gpu-rasterization --enable-impl-side-painting --ozone-platform=wayland --ignore-gpu-blocklist
```
Now when I add: --in-process-gpu everything is super smooth and nice, but it will always freeze up when closing (e.g. destroying context menu's) for example when closing the 3-dotted options menu or the menu that shows copy/paste etc when right-clicking somewhere.

It does not happen without: --in-process-gpu but then it will be less smooth overall.

I think I've reported this a long time before: https://bugs.chromium.org/p/chromium/issues/detail?id=949096
But it still seems to happen.
Any ideas on how to fix this?

I've attached screenshots of a context menu... it works when the menu is still showing but freezes/crashes when the menu goes away.
[image]

MastaG, 2021-01-26:
Okay, so I recompiled today's git and I was able to get the backtrace of the offending thread.
I can also verify it happens on gnome-shell but also on weston wayland compositor, the behaviour is exactly the same.
So for the record, without --in-process-gpu everything is fine, but if I add it, then it will freeze up when destroying a context menu (e.g. overlay window).

```
Thread 15 (Thread 0xa9a0dec0 (LWP 4125) "Chrome_InProcGp"):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0x6d7267c) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0xd3, cond=0x6d72650) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=cond@entry=0x6d72650, mutex=0xd3, mutex@entry=0x6d7262c) at pthread_cond_wait.c:638
#3  0xb3a1ae60 in read_events (display=0x6d725b0) at src/wayland-client.c:1497
#4  wl_display_read_events (display=0x6d725b0) at src/wayland-client.c:1567
#5  0xb4df6806 in dispatch_queue.constprop () at /usr/local/lib/libmali.so.1
#6  0xb4df6896 in window_surface_delete () at /usr/local/lib/libmali.so.1
#7  0xb4dee1e4 in eglp_window_surface_specific_deinitialization () at /usr/local/lib/libmali.so.1
#8  0xb4decd14 in eglp_delete_surface () at /usr/local/lib/libmali.so.1
#9  0xb4ded540 in eglDestroySurface () at /usr/local/lib/libmali.so.1
#10 0x03bec022 in Destroy() () at ../../ui/gl/gl_surface_egl.cc:1562
#11 0x018566da in ~GLSurfaceWayland () at ../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:70
#12 ~GLSurfaceWayland() () at ../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:69
#13 0x03be085e in DeleteInternal<gl::GLSurface> () at ../../base/memory/ref_counted.h:349
#14 Destruct () at ../../base/memory/ref_counted.h:315
#15 Release () at ../../base/memory/ref_counted.h:338
#16 Release () at ../../base/memory/scoped_refptr.h:322
#17 ~scoped_refptr () at ../../base/memory/scoped_refptr.h:224
#18 ~GLSurfaceAdapter() () at ../../ui/gl/gl_surface.cc:524
#19 0x041853fa in ~PassThroughImageTransportSurface () at ../../gpu/ipc/service/pass_through_image_transport_surface.cc:47
#20 ~PassThroughImageTransportSurface() () at ../../gpu/ipc/service/pass_through_image_transport_surface.cc:46
#21 0x0218e096 in DeleteInternal<gl::GLSurface> () at ../../base/memory/ref_counted.h:349
#22 Destruct () at ../../base/memory/ref_counted.h:315
#23 Release () at ../../base/memory/ref_counted.h:338
#24 Release () at ../../base/memory/scoped_refptr.h:322
```

Attached the full trace and also the clipboard info of about://gpu screen with and without --in-process-gpu.
[gdb_bt.txt](https://github.com/OSSystems/meta-browser/files/5873108/gdb_bt.txt)
[in-process-gpu.txt](https://github.com/OSSystems/meta-browser/files/5873110/in-process-gpu.txt)
[no-in-process-gpu.txt](https://github.com/OSSystems/meta-browser/files/5873112/no-in-process-gpu.txt)

MastaG, 2021-01-26:
There's a similar bug report on ARM from way before when using GTK applications on wayland.
https://community.arm.com/developer/tools-software/graphics/f/discussions/8146/r12p0-wayland-driver-odroid-xu3-frees-objects-too-early-leading-to-segm-fault

But I believe eventually the gnome team added a workaround, because I cannot reproduce it any longer and I'm still using the same r12p0 driver.

Basically the developer says:

This segfault can happen if the application frees the Wayland surface too early, specifically if the associated EGL surface is still current. If this is the case, the application is doing something like the following during clean up:

eglDestroySurface(egl_surface);
wl_egl_window_destroy(wl_egl_window_win);
wl_surface_destroy(wl_surface);

If egl_surface was either the draw or read argument in the previous call to eglMakeCurrent, egl_surface and wl_egl_window_win are only marked for deletion and are still in use. Destroying wl_surface results in the SEGFAULT when the driver subsequently needs to do something with the wl_surface (in this case, part of deletion). EGL spec 1.5 sections 3.5.5 and 3.2 cover the lifetime of EGL objects.

There are 2 possible application fixes you could consider:
 * Call eglMakeCurrent(display, EGL_NO_SURFACE, EGL_NO_SURFACE, EGL_NO_CONTEXT) before destroying the surface.
 * Call eglTerminate() instead of destroying the surfaces individually.

I don't know if this applies to chromium-ozone-wayland as well?
If yes, could this be added as a workaround?

ChrisGRSEB, 2021-02-02:
Hi @MastaG . Could you give us your chromium version used ?

MastaG, 2021-02-02:
Hi @ChrisGRSEB 
I'm using chromium-ozone-wayland_87.0.4280.141 built using this meta-browser layer without any changes.

So for the record, it only happens when using **--in-process-gpu** in both weston and gnome-shell.

I'm no C++ coder and not familiar with the code-base but judging from what the ARM developer stated.
Would something like this work? (just a while guess):
```
--- chromium-87.0.4280.141/ui/gl/gl_surface_egl.cc	2021-01-28 00:55:33.964646377 +0100
+++ chromium-87.0.4280.141/ui/gl/gl_surface_egl.cc	2021-01-28 01:00:06.472857212 +0100
@@ -2087,6 +2090,7 @@

 void PbufferGLSurfaceEGL::Destroy() {
   if (surface_) {
+    eglMakeCurrent(GetDisplay(), EGL_NO_SURFACE, EGL_NO_SURFACE, EGL_NO_CONTEXT);
     if (!eglDestroySurface(GetDisplay(), surface_)) {
       LOG(ERROR) << "eglDestroySurface failed with error "
                  << GetLastEGLErrorString();
```

MastaG, 2021-02-12:
any ideas here?

MastaG, 2021-03-10:
Updated to Fedora 34 with gnome 40 (although it also happens when running on weston).
I did a fresh build of ```88.0.4324.150``` with no external patches or whatsoever.
For the record it ```only``` happens when running with ```--in-process-gpu```.
When destroying a context (right-click) window:
```
Thread 16 (LWP 3733 "Chrome_InProcGp"):
#0  0xb6e76ef0 in __futex_abstimed_wait_common64 (futex_word=futex_word@entry=0x6ea4f68, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=0, private@entry=-1452621184, cancel=cancel@entry=true) at ../sysdeps/nptl/futex-internal.c:74
#1  0xb6e76fa0 in __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x6ea4f68, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ../sysdeps/nptl/futex-internal.c:123
#2  0xb6e6f738 in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x6ea4f1c, cond=0x6ea4f40) at pthread_cond_wait.c:504
#3  __pthread_cond_wait (cond=0x6ea4f40, mutex=0x6ea4f1c) at pthread_cond_wait.c:619
#4  0xb3cf9cc0 in wl_display_read_events () at /lib/libwayland-client.so.0
#5  0xb4cec806 in dispatch_queue.constprop () at /usr/local/lib/libmali.so.1
#6  0xb4cec896 in window_surface_delete () at /usr/local/lib/libmali.so.1
#7  0xb4ce41e4 in eglp_window_surface_specific_deinitialization () at /usr/local/lib/libmali.so.1
#8  0xb4ce2d14 in eglp_delete_surface () at /usr/local/lib/libmali.so.1
#9  0xb4ce3540 in eglDestroySurface () at /usr/local/lib/libmali.so.1
#10 0x03c97732 in Destroy() () at ../../ui/gl/gl_surface_egl.cc:1562
#11 0x018aec06 in ~GLSurfaceWayland () at ../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:70
#12 ~GLSurfaceWayland() () at ../../ui/ozone/platform/wayland/gpu/gl_surface_wayland.cc:69
#13 0x03c8c23a in DeleteInternal<gl::GLSurface> () at ../../base/memory/ref_counted.h:349
#14 Destruct () at ../../base/memory/ref_counted.h:315
#15 Release () at ../../base/memory/ref_counted.h:338
#16 Release () at ../../base/memory/scoped_refptr.h:322
#17 ~scoped_refptr () at ../../base/memory/scoped_refptr.h:224
#18 ~GLSurfaceAdapter() () at ../../ui/gl/gl_surface.cc:524
#19 0x04233832 in ~PassThroughImageTransportSurface () at ../../gpu/ipc/service/pass_through_image_transport_surface.cc:47
#20 ~PassThroughImageTransportSurface() () at ../../gpu/ipc/service/pass_through_image_transport_surface.cc:46
#21 0x021f8836 in DeleteInternal<gl::GLSurface> () at ../../base/memory/ref_counted.h:349
#22 Destruct () at ../../base/memory/ref_counted.h:315
#23 Release () at ../../base/memory/ref_counted.h:338
#24 Release () at ../../base/memory/scoped_refptr.h:322
#25 ~scoped_refptr () at ../../base/memory/scoped_refptr.h:224
#26 ~ReleaseCurrent () at ../../components/viz/service/display_embedder/skia_output_surface_impl_on_gpu.cc:334
#27 ~OptionalStorageBase () at ../../base/optional.h:58
#28 ~OptionalBase () at ../../base/optional.h:255
#29 ~SkiaOutputSurfaceImplOnGpu() () at ../../components/viz/service/display_embedder/skia_output_surface_impl_on_gpu.cc:462
#30 0x021f8a42 in viz::SkiaOutputSurfaceImplOnGpu::~SkiaOutputSurfaceImplOnGpu() () at ../../components/viz/service/display_embedder/skia_output_surface_impl_on_gpu.cc:437
#31 0x021f7758 in Run () at ../../base/callback.h:101
#32 operator() () at ../../components/viz/service/display_embedder/skia_output_surface_impl.cc:985
#33 Invoke<(lambda at ../../components/viz/service/display_embedder/skia_output_surface_impl.cc:975:7), std::vector<base::OnceCallback<void ()> >, base::WaitableEvent *, viz::SkiaOutputSurfaceImplOnGpu *, bool> () at ../../base/bind_internal.h:379
#34 MakeItSo<(lambda at ../../components/viz/service/display_embedder/skia_output_surface_impl.cc:975:7), std::vector<base::OnceCallback<void ()> >, base::WaitableEvent *, viz::SkiaOutputSurfaceImplOnGpu *, bool> () at ../../base/bind_internal.h:637
#35 RunImpl<(lambda at ../../components/viz/service/display_embedder/skia_output_surface_impl.cc:975:7), std::tuple<std::vector<base::OnceCallback<void ()> >, base::WaitableEvent *, viz::SkiaOutputSurfaceImplOnGpu *, bool>, 0, 1, 2, 3> () at ../../base/bind_internal.h:710
#36 RunOnce() () at ../../base/bind_internal.h:679
#37 0x03e43dda in Run () at ../../base/callback.h:101
#38 RunNextTask() () at ../../gpu/command_buffer/service/scheduler.cc:577
#39 0x02fccd8a in Run () at ../../base/callback.h:101
#40 RunTask() () at ../../base/task/common/task_annotator.cc:163
#41 0x02fd8db6 in DoWorkImpl() () at ../../base/task/sequence_manager/thread_controller_with_message_pump_impl.cc:357
#42 0x02fd8af4 in DoWork() () at ../../base/task/sequence_manager/thread_controller_with_message_pump_impl.cc:268
#43 0x02fd923e in non-virtual thunk to base::sequence_manager::internal::ThreadControllerWithMessagePumpImpl::DoWork() ()
#44 0x02fa45ba in Run() () at ../../base/message_loop/message_pump_default.cc:39
#45 0x02fd94ac in Run() () at ../../base/task/sequence_manager/thread_controller_with_message_pump_impl.cc:477
#46 0x02fbc106 in Run() () at ../../base/run_loop.cc:124
#47 0x02fea8be in ThreadMain() () at ../../base/threading/thread.cc:382
#48 0x030052cc in ThreadFunc() () at ../../base/threading/platform_thread_posix.cc:87
#49 0xb6e67c84 in start_thread (arg=0xa96ad000) at pthread_create.c:473
#50 0xb61f9b58 in  () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
```

Is there any workaround possible? (perhaps based on my earlier quote from the ARM mali developer?)
Some quick and dirty workaround or something?

Because I'd really like to run it with --in-process-gpu as that makes the rendering smooth as butter.

msisov, 2021-03-10:
Sorry, I haven't had a chance to look into this issue yet. I will try to find time for this.

StefanLellwitz, 2021-03-17:
Hi,

I am experiencing the exact same behavior as @MastaG on i.MX8QM. Running Chromium 88.0.4324.150 on Weston.

I have to use `--in-process-gpu` because unfortunately I haven't found any other way to use hardware acceleration on i.MX8QM. Please see #232 for some more details.

When running Chromium with `--disable-gpu` the problem does not occur and everything seems fine. So I guess I can confirm that this only occurs in combination with `--in-process-gpu`.

MastaG, 2021-03-19:
Hi @msisov 

With the latest version (as of today) it doesn't segfault any longer but still doesn't force hardware compositing any longer when ```--in-process-gpu``` and ```--ignore-gpu-blocklist``` are given.

As for the previous version, could the crash be related to the fact that the mali drivers are missing dmabuf support in wayland?
Here's my wayland-info:
````
interface: 'mali_buffer_sharing',                        version:  4, name:  1
interface: 'wl_compositor',                              version:  4, name:  2
interface: 'wl_shm',                                     version:  1, name:  3
	formats: 'XB30'(0x30334258) 'AB30'(0x30334241) 'XR30'(0x30335258) 'AR30'(0x30335241) RGB565 XRGB8888 ARGB8888
interface: 'wl_output',                                  version:  2, name:  4
	x: 0, y: 0, scale: 1,
	physical_width: 600 mm, physical_height: 340 mm,
	make: 'ACR', model: 'XB273K GP',
	subpixel_orientation: unknown, output_transform: normal,
	mode:
		width: 1920 px, height: 1080 px, refresh: 60.000 Hz,
		flags: current preferred
interface: 'zxdg_output_manager_v1',                     version:  3, name:  5
	xdg_output_v1
		output: 4
		name: 'HDMI-1'
		description: 'Acer Technologies 27"'
		logical_x: 0, logical_y: 0
		logical_width: 1920, logical_height: 1080
interface: 'wl_data_device_manager',                     version:  3, name:  6
interface: 'zwp_primary_selection_device_manager_v1',    version:  1, name:  7
interface: 'gtk_primary_selection_device_manager',       version:  1, name:  8
interface: 'wl_subcompositor',                           version:  1, name:  9
interface: 'xdg_wm_base',                                version:  3, name: 10
interface: 'zxdg_shell_v6',                              version:  1, name: 11
interface: 'wl_shell',                                   version:  1, name: 12
interface: 'gtk_shell1',                                 version:  4, name: 13
interface: 'wp_viewporter',                              version:  1, name: 14
interface: 'zwp_pointer_gestures_v1',                    version:  1, name: 15
interface: 'zwp_tablet_manager_v2',                      version:  1, name: 16
interface: 'wl_seat',                                    version:  5, name: 17
	name: seat0
	capabilities: pointer keyboard
	keyboard repeat rate: 33
	keyboard repeat delay: 500
interface: 'zwp_relative_pointer_manager_v1',            version:  1, name: 18
interface: 'zwp_pointer_constraints_v1',                 version:  1, name: 19
interface: 'zxdg_exporter_v1',                           version:  1, name: 20
interface: 'zxdg_importer_v1',                           version:  1, name: 21
interface: 'zwp_keyboard_shortcuts_inhibit_manager_v1',  version:  1, name: 22
interface: 'zwp_text_input_manager_v3',                  version:  1, name: 23
interface: 'gtk_text_input_manager',                     version:  1, name: 24
interface: 'wl_eglstream_controller',                    version:  1, name: 26
````

As you can see ```zwp_linux_dmabuf_v1``` is missing, but I do have a ```mali_buffer_sharing``` extension.

msisov, 2021-03-22:
Can you share gpu logs Chromium generates?

iMonZ, 2021-03-23:
I have tried Nativefier and there is this flag not available
error: unknown option '--in-process-gpu'

MastaG, 2021-04-02:
Hi @msisov it still happens with todays git (which switched over to the llvm runtime).
It just hangs forever when I close a context menus when ```--in-process-gpu``` is enabled.
Which is a shame as the difference between running with this flash is night and day.
Scrolling through pages and even watching youtube and netflix is fast and smooth even though the video streams are software decoded on the cpu.
I'm now rebuilding it with ```enable_gpu_client_logging=true``` and  ```enable_gpu_service_logging=true``` GN flags.
I really hope you can find some hint on why it happens.

msisov, 2021-04-02:
The reason why it hangs is the deadlock. I didn’t have time to work on that yet because of other priorities. I’m sorry.

The reason why —in-process-gpu is way faster for you because out of process gpu cannot properly work on your system (lack of either libgbm/drm render nodes/wayland dmabuf protocols) and chromium falls back to sw compositing. What’s more, only out of process gpu is heavily developed, which means delegated compositing, zero copy, and other features.

MastaG, 2021-04-06:
No problem @msisov 

I've created the logs and the backtrace.
[backtrace.txt](https://github.com/OSSystems/meta-browser/files/6263184/backtrace.txt)
[gpu-client-log.txt](https://github.com/OSSystems/meta-browser/files/6263185/gpu-client-log.txt)
[gpu-service-log.txt](https://github.com/OSSystems/meta-browser/files/6263189/gpu-service-log.txt)

For the record, the context-menu does actually disappear when you close it.. it's just that the browser locks up.
If I can help by providing more information, please let me know.

MastaG, 2021-04-19:
I wanted to say that the fix from @msisov is still going strong.
Chromium is now usable like a desktop pc :)

cchsu23, 2021-04-27:
> I wanted to say that the fix from @msisov is still going strong.
> Chromium is now usable like a desktop pc :)

Hi , can you share the commit that can solve this issue?

msisov, 2021-04-28:
https://chromium-review.googlesource.com/c/chromium/src/+/2844573

This will be included in the next meta-browser update.

cchsu23, 2021-05-06:
Hi Msisov:

I already upgrade to 90.0.4430.93 stable version, the issue is gone.
 
Thanks
Dennis

closed 2021-05-10 by msisov