OSSystems/meta-browser #451: chromium: support for 32 bit

yifan19, 2021-02-01:
try to build chromium-x11 using `lib32` for `intel-x86-64` as an image for `wrlinux-image-std`

There is a warning for relocation and the resulting binary segfaults when run

Any thoughts on supporting 32-bit machines?

```
root@intel-x86-64:~# chromium --no-sandbox --disable-gpu

[1465:1478:1216/060845.676256:FATAL:gpu_data_manager_impl_private.cc(445)] GPU process isn't usable. Goodbye.

Trace/breakpoint trap

root@intel-x86-64:~# tail -n10  /var/log/syslog

2020-12-16T06:15:50.463854+00:00 intel-x86-64 kernel: VizCompositorTh[2301]: segfault at 0 ip 0000000000000000 sp 00000000f2b5a590 error 14 in chromium-bin[565fe000+104a000]
``

MastaG, 2021-02-03:
That's strange.
I'm building for 32bit armv7 and get no such issues.

rakuco, 2021-02-03:
> Is there an intent to support 32 bit?

Upstream itself stopped providing official 32-bit Linux packages long ago, and I don't think this setup is being actively tested by them. Can you share more information about your setup? Is this a 32-bit Chromium build running in a x86-64 kernel + userland, or a x86-64 kernel with a fully 32-bit userland?

yifan19, 2021-02-04:
I think Its 32 bit chromium running in a x86-64 kernel with everything else being 32 bit
basically i built something similar to a lib32-core-image-sato but using wrlinux.

Windriver has no customer interest in intel86 32 bit machines , and i don't think anyone would want to throw out the 64bit userspace for a 32bit one.

> I'm building for 32bit armv7 and get no such issues.
I have heard that 32bit arm has more interest.

If there is no one that cares about 32bit, then i won't be digging in to fixing some of these issues the testing team and I found.

rakuco, 2021-02-05:
I wouldn't go as far as saying "nobody cares", support for it here (and maybe upstream) just doesn't seem as high-priority as x86-64, but we and upstream would be happy to review whatever fixes you may have come up with.

yifan19, 2021-02-08:
its not my priority either, Windriver testing team went through the common cases and uncovered  these kind of issues.
given the low priority given to 32 bit intel (both here, upstream and windriver), i will leave it as it is

rakuco, 2021-02-25:
I'm reopening this one for a while because there's also a discussion about issues with 32-bit Linux in the [chromium-packagers](https://groups.google.com/a/chromium.org/g/chromium-packagers/c/EwWwdXmCcgY/m/pM5y9ppVAQAJ) mailing list; if some patches end up landing we might be able to revisit this.

yifan19, 2021-04-07:
qemux86 no longer have "binary relocation errors" with latest builds.