OSSystems/meta-browser #454: chromium: Changing "use-vaapi" defaults?

rakuco, 2021-02-03:
Now that we've updated the recipe to M88, I'd like to discuss whether it makes sense to default the "use-vaapi" to true on at least the Intel targets. That's what upstream seems to be doing according to https://bugs.chromium.org/p/chromium/issues/detail?id=1097029.

I'm not familiar with this area myself, so I wondering if @msisov can provide more information:
* What kind of hardware works worse with VAAPI support enabled?
* Can we rely on the upstream default of enabling it for IA and disabling it on other architectures?

msisov, 2021-02-03:
Well, even though upstream has va-api enabled in built time, it's still disabled on runtime. VA-API is mostly for ChromeOS and X11 path may not always work (it may break and somebody may need to fix that).

Secondly, we do have va-api patches that enable VA-API for ozone/wayland. Basically, it reuses ChromeOS' path that uses vaapi drm.

Regarding hardware, personally, I tried with AMD RX460 (my workstation) and with some intel based boards - gen 8 and gen 9 if I'm not mistaken.

Gen8 had some problems with damaged frames that I partly fixed by adding more reference frames and explicit synchronization.

Last  but not least, the reason why I decided to disable va-api by default is that it brings unnecessary libva dependency that not everyone may need. That's why I think that it should stay as optional setting.

rakuco, 2021-02-03:
> Last but not least, the reason why I decided to disable va-api by default is that it brings unnecessary libva dependency that not everyone may need. That's why I think that it should stay as optional setting.

Yeah, I was talking about enabling it by default on chromium-x11 when targeting IA while still making it possible to disable it.

> Well, even though upstream has va-api enabled in built time, it's still disabled on runtime.

Does that apply to X11 too? In other words, is it also possible that a user enables the "use-vaapi" knob and Chromium still refuses to use it?

msisov, 2021-02-03:
HW accelerated video decoding with vaapi was changed from compile to runtime (see https://bugs.chromium.org/p/chromium/issues/detail?id=1097029)

It's still disabled unless you enable hw accelerated video decoding via a runtime flag. See https://source.chromium.org/chromium/chromium/src/+/master:chrome/browser/about_flags.cc;l=2942?q=enable-accelerated-video-decod&ss=chromium and https://source.chromium.org/chromium/chromium/src/+/master:media/base/media_switches.cc;drc=8ffe0a431f557f8f0b99820374fd9cfddab4af53;l=458

msisov, 2021-02-03:
> Does that apply to X11 too? In other words, is it also possible that a user enables the "use-vaapi" knob and Chromium still refuses to use it?

Applies to Linux.

rakuco, 2021-02-03:
> HW accelerated video decoding with vaapi was changed from compile to runtime (see https://bugs.chromium.org/p/chromium/issues/detail?id=1097029)

That's the bug I was looking at. The latest commits say "Enable vaapi decoding on linux x86/64 by default", but if I understood you correctly that's misleading, and actually enabling it also depends on passing extra flags, right?

msisov, 2021-02-03:
> That's the bug I was looking at. The latest commits say "Enable vaapi decoding on linux x86/64 by default", but if I understood you correctly that's misleading, and actually enabling it also depends [...]

It'll be very problematic to have it enabled by default. I believe the commit message should've been changed to "switch from compile to runtime check to enable va-api".

rakuco, 2021-02-03:
I'll take that as a "yes" to my question :-) Thanks for the explanation, I guess it makes sense to just keep things as-is then.

closed 2021-02-03 by rakuco