OSSystems/meta-browser #463: chromium: update to 89.0.4389.72

msisov, 2021-03-03:
Stable release announcements since our latest update:
https://chromereleases.googleblog.com/2021/03/stable-channel-update-for-desktop.html

Targets and platforms we need to check:

**chromium-x11:**
- dunfell, clang, MACHINE="intel-core2-32"
- dunfell, clang, MACHINE="intel-corei7-64"
- dunfell, clang, MACHINE="raspberrypi3"
- gatesgarth, MACHINE="raspberrypi3"

**chromium-ozone-wayland:**

- dunfell, clang, MACHINE="intel-corei7-64"
- dunfell, clang, MACHINE="qemuarm64"
- gatesgarth, clang, MACHINE="raspberrypi3"
- dunfell, clang, MACHINE="m3ulcb"

kraj, 2021-03-03:
do we test master ? clang has been updated to 12.0.0-rc2 on master and I would be interested to know if there are any issues with master/meta-clang

MastaG, 2021-03-09:
I'm currently building against master for armv7.
Will report back shortly.

rwmacleod, 2021-03-10:
With 88.0.4324.150 on recent-ish master, building for R Pi, we see:
```
aarch64-wrs-linux-clang++ -target aarch64-wrs-linux  -mcpu=cortex-a72 -march=armv8-a+crc+crypto   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=TOPDIR/tmp-glibc/work/cortexa72-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot -MMD -MF obj/base/third_party/double_conversion/double_conversion/fixed-dtoa.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -I../.. -Igen -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -Xclang -fdebug-compilation-dir -Xclang . -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-max-tokens -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wno-unused-const-variable -Wno-unused-function -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=TOPDIR/tmp-glibc/work/cortexa72-wrs-linux/chromium-x11/88.0.4324.150-r0=/usr/src/debug/chromium-x11/88.0.4324.150-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/cortexa72-wrs-linux/chromium-x11/88.0.4324.150-r0=/usr/src/debug/chromium-x11/88.0.4324.150-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/cortexa72-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot=                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/cortexa72-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot-native= -g1  -c ../../base/third_party/double_conversion/double-conversion/fixed-dtoa.cc -o obj/base/third_party/double_conversion/double_conversion/fixed-dtoa.o
../../base/third_party/double_conversion/double-conversion/fixed-dtoa.cc:28:10: fatal error: 'cmath' file not found
#include <cmath>
         ^~~~~~~
```
And for intel-x86-64:
```
clang++ -target x86_64-linux  -MMD -MF src/base/value_iterators.o.d -I../../../tools/gn/src -I. -isystem/home/wrlbuild/builds/builds-941c6660-6b1b-4311-bae5-d3c659183007/wrlinux_intel-x86-64_std_chromium_predefined_20/tmp-glibc/work/x86_64-linux/gn-native/88.0.4324.150-r0/recipe-sysroot-native/usr/include -O2 -pipe -isystem/home/wrlbuild/builds/builds-941c6660-6b1b-4311-bae5-d3c659183007/wrlinux_intel-x86-64_std_chromium_predefined_20/tmp-glibc/work/x86_64-linux/gn-native/88.0.4324.150-r0/recipe-sysroot-native/usr/include -O2 -pipe -DNDEBUG -O3 -fdata-sections -ffunction-sections -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -std=c++17 -c ../../../tools/gn/src/base/value_iterators.cc -o src/base/value_iterators.o
In file included from ../../../tools/gn/src/base/value_iterators.cc:5:
../../../tools/gn/src/base/value_iterators.h:8:10: fatal error: 'memory' file not found
#include <memory>
         ^~~~~~~~ 
```

yifan19, 2021-03-10:
running clang with -v shows:
```
#include <...> search starts here:
 ../..
 gen
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot-native/usr/lib/clang/12.0.0/include
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot/usr/include
 ```

kraj, 2021-03-10:
you perhaps need to add your vedor name to clang see

https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/clang/0029-llvm-Recognize-yoe-and-poky-as-OE-distro.patch

msisov, 2021-03-10:
> do we test master ? clang has been updated to 12.0.0-rc2 on master and I would be interested to know if there are any issues with master/meta-clang

I only test dunfell and gatesgarth. Do you think the master branch should also be tested?

yifan19, 2021-03-10:
> you perhaps need to add your vendor name to clang see
> 
> https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/clang/0029-llvm-Recognize-yoe-and-poky-as-OE-distro.patch
```
include "..." search starts here:
#include <...> search starts here:
 ../..
 gen
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot/usr/lib64/x86_64-wrs-linux/10.2.0/../../../include/c++/10.2.0
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot/usr/lib64/x86_64-wrs-linux/10.2.0/../../../include/c++/10.2.0/x86_64-wrs-linux
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot/usr/lib64/x86_64-wrs-linux/10.2.0/../../../include/c++/10.2.0/backward
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot-native/usr/lib/clang/12.0.0/include
 /yow-lpggp32/yyu1/b/build/tmp-glibc/work/core2-64-wrs-linux/chromium-x11/88.0.4324.150-r0/recipe-sysroot/usr/include
``` 
THX

closed 2021-03-12 by msisov

shr-project, 2021-04-07:
> > you perhaps need to add your vendor name to clang see
> > https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/clang/0029-llvm-Recognize-yoe-and-poky-as-OE-distro.patch
> 
> THX

@yifan19 
You have probably noticed already, but just in case:
with latest meta-clang master you don't need to add this vendor manually, it should be added from TARGET_VENDOR variable since:
https://github.com/kraj/meta-clang/pull/446 and follow-up fix in https://github.com/kraj/meta-clang/pull/451