OSSystems/meta-browser #468: Firefox segfault on startup

MastaG, 2021-03-10:
I've built Firefox with the latest meta-rust/meta-clang/meta-llvm but it crashes when trying to run it.
I'm using: ```GDB_BACKEND=wayland```:

```
#0  0xb217e6fa in GeckoCrash () at /usr/lib/firefox/libxul.so
#1  0xb26f204e in gkrust_shared::panic_hook () at /usr/lib/firefox/libxul.so
#2  0xb26f1fa8 in core::ops::function::Fn::call () at /usr/lib/firefox/libxul.so
#3  0xb2761aea in std::panicking::rust_panic_with_hook () at /usr/lib/firefox/libxul.so
#4  0xb275a484 in std::panicking::begin_panic_handler::{{closure}} () at /usr/lib/firefox/libxul.so
#5  0xb275a42a in std::sys_common::backtrace::__rust_end_short_backtrace () at /usr/lib/firefox/libxul.so
#6  0xb27605e4 in rust_begin_unwind () at /usr/lib/firefox/libxul.so
#7  0xb25f0ff2 in core::panicking::panic_fmt () at /usr/lib/firefox/libxul.so
#8  0xb25f0fd0 in core::panicking::panic () at /usr/lib/firefox/libxul.so
#9  0xb28cad12 in style::gecko_properties::ComputedValues::default_values () at /usr/lib/firefox/libxul.so
#10 0xb269c3ae in Servo_StyleSet_Init () at /usr/lib/firefox/libxul.so
#11 0xb19e3b02 in mozilla::ServoStyleSet::ServoStyleSet(mozilla::dom::Document&) () at /usr/lib/firefox/libxul.so
#12 0xb0e09094 in mozilla::dom::Document::Init() () at /usr/lib/firefox/libxul.so
#13 0xb14fc0c4 in nsHTMLDocument::Init() () at /usr/lib/firefox/libxul.so
#14 0xb14fbd90 in NS_NewHTMLDocument(mozilla::dom::Document**, bool) () at /usr/lib/firefox/libxul.so
#15 0xb1bb16f6 in nsContentDLF::CreateBlankDocument(nsILoadGroup*, nsIPrincipal*, nsDocShell*) () at /usr/lib/firefox/libxul.so
#16 0xb1fb7686 in nsDocShell::CreateAboutBlankContentViewer(nsIPrincipal*, nsIURI*, bool, bool) () at /usr/lib/firefox/libxul.so
#17 0xb1fb792e in non-virtual thunk to nsDocShell::CreateAboutBlankContentViewer(nsIPrincipal*) () at /usr/lib/firefox/libxul.so
#18 0xb1fcb3d8 in nsWebShellWindow::Initialize(nsIXULWindow*, nsIXULWindow*, nsIURI*, int, int, bool, nsIRemoteTab*, mozIDOMWindowProxy*, nsWidgetInitData&) ()
    at /usr/lib/firefox/libxul.so
#19 0xb1fca2a8 in nsAppShellService::JustCreateTopWindow(nsIXULWindow*, nsIURI*, unsigned int, int, int, bool, nsIRemoteTab*, mozIDOMWindowProxy*, nsWebShellWindow**) ()
    at /usr/lib/firefox/libxul.so
#20 0xb1fca700 in nsAppShellService::CreateTopLevelWindow(nsIXULWindow*, nsIURI*, unsigned int, int, int, nsIRemoteTab*, mozIDOMWindowProxy*, nsIXULWindow**) ()
    at /usr/lib/firefox/libxul.so
#21 0xb21108ee in nsAppStartup::CreateChromeWindow2(nsIWebBrowserChrome*, unsigned int, nsIRemoteTab*, mozIDOMWindowProxy*, unsigned long long, bool*, nsIWebBrowserChrome**)
    () at /usr/lib/firefox/libxul.so
#22 0xb21639a6 in nsWindowWatcher::CreateChromeWindow(nsTSubstring<char> const&, nsIWebBrowserChrome*, unsigned int, nsIRemoteTab*, mozIDOMWindowProxy*, unsigned long long, nsIWebBrowserChrome**) () at /usr/lib/firefox/libxul.so
#23 0xb216318a in nsWindowWatcher::OpenWindowInternal(mozIDOMWindowProxy*, char const*, char const*, char const*, bool, bool, bool, nsIArray*, bool, bool, bool, nsDocShellLoadState*, mozIDOMWindowProxy**) () at /usr/lib/firefox/libxul.so
#24 0xb216246a in nsWindowWatcher::OpenWindow(mozIDOMWindowProxy*, char const*, char const*, char const*, nsISupports*, mozIDOMWindowProxy**) () at /usr/lib/firefox/libxul.so
#25 0xb05fe340 in NS_InvokeByIndex () at /usr/lib/firefox/libxul.so
#26 0xa5047680 in  ()
```

ashie, 2021-04-22:
Hmm, it's reproduced also on my build.

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.7"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:13f4ddf50eccaeed96a40a5f1a1d4173e677e98a"
meta-oe              
meta-python          
meta-networking      = "dunfell:430ef96fe65f62d8da995f446d5b9b093544f031"
meta-clang           = "dunfell:e63d6f9abba5348e2183089d6ef5ea384d7ae8d8"
meta-rust            = "master:920db7b045c1f721efcc2e1d891516b515b7e7a8"
meta-firefox         = "master:1bff166b36072a8cacc0f29f02de5fa9748824e9"
```

core-image-sato on Virtual Box.
I didn't follow recent changes so I don't know yet whats wrong.

ashie, 2021-04-23:
> Hmm, it's reproduced also on my build.

Now I've confirmed that I can build workable binary when I use Rust 1.43 (meta-rust:7f235b6f8973cc5269448375f2a8f9867bb2a369). Not yet test other versions.