OSSystems/meta-browser #473: Chromium missing fonts

yifan19, 2021-03-17:
built with latest poky + meta-clang

MACHINE: qemux86-64
bitbake core-image-sato

```
runqemu qemux86-64 kvm qemuparams="-m 4000 -display sdl -vga std" serialstdio
```

run `DISPLAY=:0.0 chromium 1.1.1.1` as a user and you see no font display, image works ok.
https://imgur.com/a/lITEEKE

runs with --no-sandbox as user fixes the problem (the error/warning msg stay the same)

error msg (doesn't impact browsing)
```
^Cqemux86-64:~$ DISPLAY=:0.0 chromium 1.1.1.1  
[1932:1932:0317/143707.902184:ERROR:browser_dm_token_storage_linux.cc(94)] Error: /etc/machine-id contains 0 characters (32 were expected).
[1963:1963:0317/143707.915188:ERROR:gl_implementation.cc(283)] Failed to load libGLESv2.so.2: libGLESv2.so.2: cannot open shared object file: No such file or directory
[1963:1963:0317/143707.927250:ERROR:viz_main_impl.cc(150)] Exiting GPU process due to errors during initialization
[1932:1961:0317/143707.953809:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1932:1961:0317/143707.953841:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1932:1961:0317/143707.997092:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1932:1961:0317/143707.998560:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1932:2016:0317/143708.166145:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1932:2016:0317/143708.168306:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1932:2016:0317/143708.171132:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
```

rakuco, 2021-03-17:
Are other pages affected as well? At least `font-family: sans-serif` should have resolved to an existing font.

yifan19, 2021-03-17:
> Are other pages affected as well? At least `font-family: sans-serif` should have resolved to an existing font.

Some webpages (I think YouTube... does display font properly, perhaps they provide their own font). (I'll do some more testing on an actual machine) https://imgur.com/a/WNCy9VC

sometimes, this error pops up, seems related to the issue
```
Fontconfig error: Cannot load default config file: No such file: (null)
```

Also it doesn't happen when I build it for rpi4

otavio, 2021-03-17:
@yifan19 check if your image has any default font installed.

yifan19, 2021-03-24:
here is an strace of what I think causes
`Fontconfig error: Cannot load default config file: No such file: (null)` occurs

reproducing steps:
build a core-image-sato image for qemux-86-64 including chromium-x11
boot it in qemu
`runqemu qemux86-64 kvm qemuparams="-m 4000 -display sdl -vga std" serialstdio`
in the terminal type
`DISPLAY=:0.0 chromium` 
search for anything in the url bar, since we don't have DNS, it will error out, but font will be missing

```
qemux86-64:~$ less data_follow.str | grep /etc/fonts/fonts.conf -A10  | grep access
[pid  2711] access("/etc/fonts/fonts.conf", R_OK) = 0
[pid  2711] access("/etc/fonts/fonts.conf", R_OK) = 0
[pid  2711] access("/etc/fonts/conf.d", R_OK) = 0
[pid  2711] access("/etc/fonts/conf.d", R_OK) = 0
[pid  2711] access("/etc/fonts/fonts.conf", R_OK <unfinished ...>
[pid  2711] <... access resumed>)       = 0
[pid  2711] access("/etc/fonts/fonts.conf", R_OK <unfinished ...>
[pid  2711] <... access resumed>)       = 0
[pid  2914] access("/etc/fonts/fonts.conf", R_OK <unfinished ...>
[pid  2914] <... access resumed>)       = -1 EPERM (Operation not permitted)
```
no such error happens with --sandbox

yifan19, 2021-03-24:
@otavio  I didn't install any extra font packages, but I have the "default"

```
root@qemux86-64:~# ls /etc/fonts/conf.d/     
10-hinting-slight.conf      51-local.conf
10-scale-bitmap-fonts.conf  60-generic.conf
20-unhint-small-vera.conf   60-latin.conf
30-liberation-aliases.conf  65-fonts-persian.conf
30-metric-aliases.conf      65-nonlatin.conf
40-nonlatin.conf            69-unifont.conf
45-generic.conf             80-delicious.conf
45-latin.conf               90-synthetic.conf
49-sansserif.conf           README
50-user.conf                
```
```
root@qemux86-64:~# ls /usr/share/fonts/ttf/
LiberationMono-Bold.ttf        LiberationSans-Italic.ttf
LiberationMono-BoldItalic.ttf  LiberationSans-Regular.ttf
LiberationMono-Italic.ttf      LiberationSerif-Bold.ttf
LiberationMono-Regular.ttf     LiberationSerif-BoldItalic.ttf
LiberationSans-Bold.ttf        LiberationSerif-Italic.ttf
LiberationSans-BoldItalic.ttf  LiberationSerif-Regular.ttf
```

Escalion, 2021-03-29:
Hi yifan, it looks like a simple permissions error.
Your fonts should be set to 755.
Verify with ls -la /usr/share/fonts/ttf/

yifan19, 2021-04-01:
i do remember changing permissions to 777, and it still failed, i'll try again

MarkusVolk, 2021-04-05:
Hi,
i had the same issue on aarch64 built with upcoming hardknott release. Problem seems to be caused by changes in glibc-2.33. For me the solution was rebuilding chromium using this patch:

https://src.fedoraproject.org/rpms/chromium/c/82ff440ce80f6a03b150f768a2cba9469a8ec4f4?branch=rawhide

yifan19, 2021-04-05:
> Hi,
> i had the same issue on aarch64 built with upcoming hardknott release. Problem seems to be caused by changes in glibc-2.33. For me the solution was rebuilding chromium using this patch: [...]

thanks! I think that is the same issue. I'll rebuild

yifan19, 2021-04-07:
fixed in https://github.com/OSSystems/meta-browser/commit/cb3278e31340c7f081e8deb0683df2145da515c9

closed 2021-04-07 by yifan19

MarkusVolk, 2021-07-22:
Looks like there finally is an official patch for this

https://chromium.googlesource.com/chromium/src/+/60d5e803ef2a4874d29799b638754152285e0ed9%5E%21/

I`ll give it a try within the next days and will report if the problem gets fixed with it. Hopefully it will also fix things up for Raspberry Pi 32bit where the fedora patch didn't make a difference. Chromium-ozone-wayland doesn't run, at least  it didn't when i tested it about one month ago, so there probably is also another issue but chromium-x11 did run under xwayland ( with --disable-sandbox flag set). Hopefully i can get rid of this flag

rakuco, 2021-07-22:
Please note that you'll probably need the preceding patch in that bug too: https://bugs.chromium.org/p/chromium/issues/detail?id=1164975#c29

MarkusVolk, 2021-07-23:
I have included the patches from here:
https://github.com/archlinux/svntogit-packages/tree/packages/chromium/trunk
and removed the glibc-2.33 patch. Rebuild for rockchip aarch64 was successful, Chromium is running and all the text is shown correctly.