OSSystems/meta-browser #479: Drop Down list menu will show White Screen, or Full-screen mode both on v.86.0.4240.198   and v.88.0.4324.150

cchsu23, 2021-03-18:
CHROMIUM_EXTRA_ARGS += "
--in-process-gpu
--enable-wayland-ime
--disable-restore-session-state
--disable-session-crashed-bubble
--high-dpi-support=1
--force-device-scale-factor=1
--touch-devices=enabled
--touch-events=enabled
--no-first-run
--noerrdialogs
--disable-pinch
--disable-quic
--ignore-gpu-blocklist
--disable-infobars
--disable-translate
--no-sandbox
"

GN_ARGS += "
use_lld=false
use_gold=true
enable_widevine=true
ffmpeg_branding="Chrome"
proprietary_codecs=true
is_component_ffmpeg=true
media_use_ffmpeg=true
"
exec chromium:
/usr/bin/chromium --app=https://www.netflix.com/login

Steps:
1.Netflixlogin page -> on the bottom has the language drop-down list

symptom
1. Fullscreen WhiteScreen       (wrong)
[log](https://drive.google.com/file/d/1XMtlyXZr9T4UC7fvVn7zVLpYd77XjxHr/view?usp=sharing)
[video](https://drive.google.com/file/d/1ozEM4rBt6JRE4JjgWI6YzI7hLw_9PBfn/view?usp=sharing)
2. Fullscreen Drop-Down list   (yes, but size is wrong )
[log](https://drive.google.com/file/d/1By9ABAWCuGrSljZGRL-p0E5ngpNV2JBe/view?usp=sharing)
[video](https://drive.google.com/file/d/1j-RjQJk_n6b0MJuBW5eeKWMUOA2piJ0R/view?usp=sharing)
3. will redirect to new web page (90% will crash)
see symptom2 crash...

msisov, 2021-03-18:
Html select tags were fixed in https://crbug.com/1146445 . And this patch is part of Chromium 89.0.4389.90, which is part of meta-browser already.

I would encourage you to test with that branch and report if the issue is still there.

cchsu23, 2021-03-18:
chromium-89.0.4389.82  has the same issue.....
You update new version Chromium 89.0.4389.90, right?
I will test again...

cchsu23, 2021-03-18:
or I use this patch first
 https://chromium-review.googlesource.com/c/chromium/src/+/2562141/2/third_party/blink/renderer/core/exported/web_page_popup_impl.cc

cchsu23, 2021-03-18:
chromium-89.0.4389.82
has already this patch,  but still fail...
  // Do not set the same |window_rect| more than once. In case of backends that
  // can reposition native windows during initialization of popups, |this| can
  // override position of the popup on screen, which will result in
  // RenderWidgetHostViewAura setting wrong bounds for aura::Window. In turn,
  // that can result in offset and the content will not be properly positioned.
  if (window_rect == WindowRectInScreen())
    return;

cchsu23, 2021-03-18:
> Html select tags were fixed in https://crbug.com/1146445 . And this patch is part of Chromium 89.0.4389.90, which is part of meta-browser already.
> 
> I would encourage you to test with that version and report if the issue is still there.

Hi, this patch seems to fix symptom1 (white screen), but still has symptom 2 problem

msisov, 2021-03-18:
The links to logs and videos are not accessible. I sent you requests to have an access to them.

cchsu23, 2021-03-18:
sorry about that,
I already enable sharing.. please try again.

cchsu23, 2021-03-18:
I can share new log to you. 
Apply your https://crbug.com/1146445 to fix white screen problem , but still has symptom 2 problem and symptom 3 problem
log[](https://drive.google.com/file/d/19yStL9C18z-EJLbqR1FqlsQ9QFKIUkJ-/view?usp=sharing)

cchsu23, 2021-03-18:
[log is here]( https://drive.google.com/file/d/19yStL9C18z-EJLbqR1FqlsQ9QFKIUkJ-/view?usp=sharing)

msisov, 2021-03-18:
Thanks. I will need to further triage these problems and then see how we can fix them. Thanks for your reports.

cchsu23, 2021-03-18:
Thanks,   
this crash problem is the same as #446 which was reported by me..

cchsu23, 2021-03-27:
Hi Msisov,
I use kiosk-shell as our shell (based on ivi-shell), 
symptom 2 problem is solved, size is right.
but symptom 3 problem is still happens. 
[log](https://drive.google.com/file/d/1J29bNFIIc04DLqulSiPOUJpGJfFoHwiD/view?usp=sharing) 

[3913226.483]  -> xdg_popup@35.destroy()

[3913226.581]  -> xdg_surface@33.destroy()

[3913226.608]  -> wl_surface@30.attach(nil, 0, 0)

[3913226.664]  -> wl_surface@30.commit()

[3913230.698]  -> wl_surface@30.attach(nil, 0, 0)

[3913230.757]  -> wl_surface@30.commit()

[3913230.796] wl_buffer@36.release()

[3913230.893]  -> wl_buffer@36.destroy()

cchsu23, 2021-03-27:
desktop-shell has the same problem
[log is here](https://drive.google.com/file/d/1gJF1oRBuhUOabmYBK6g9yJwY9-Wd-gkt/view?usp=sharing)

cchsu23, 2021-04-03:
Hi Msisov:
 I have 100% chance to reproduce this issue. 
 Use mouse.  This method what is touch device will often happened.

1. Netflix login page -> on the bottom has the language drop-down list, 
2. Click and popup menu, just slide over sub-menu, do not click, and choose back original menu.

p.s. if you choose one of language every time by mouse, the issue will not happened.

[mp4](https://drive.google.com/file/d/1qEXYYNnYPHhIJkkAxlVmc3nXK8ZNt7VP/view?usp=sharing)
[log](https://drive.google.com/file/d/1GGaIRYshsfCwH3MbmnbVMEVcgUVHsggz/view?usp=sharing)

cchsu23, 2021-04-06:
I see another one issue. I use their flag to test on my issue.
--disable-gpu    cannot happen..    (But more cpu loading....)
--in-process-gpu.   which I use,  can will has this issue.   (We need this to improve the performance)

msisov, 2021-04-12:
Ok. I see now it's the same problem as https://github.com/OSSystems/meta-browser/issues/447 . So, I will merge this issue into that. 
Duplicate of #447

closed 2021-04-12 by msisov