OSSystems/meta-browser #485: clang-12: error: invalid linker name in argument '-fuse-ld=gold'

twoerner, 2021-03-21:
Apparently chromium can't be built (can no longer be built?) with gold linker.

The following must be set for the chromium build to succeed:
    DISTRO_FEATURES += "ld-is-lld"

kraj, 2021-03-22:
hmmm there seems to be something more going on here. Can you post the relevant logs? In fact I built chromium with clang-12 yesterday for x86_64 of course you will need to apply https://github.com/OSSystems/meta-browser/pull/482
whats your `TARGET_VENDOR` set to ?

rakuco, 2021-03-22:
I'm also interested in the logs. The upstream bits for gold support don't seem to have changed lately.

twoerner, 2021-03-22:
I've verified the issue both with and without the patch for #482.

```
Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "opensuseleap-15.2"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53"
TARGET_FPU           = ""
meta-raspberrypi     = "master:fdf02accfe3f68d11fa178d72eb377f9ca96411c"
meta-tweaks          = "master:f5b342baa0e0a5914fa9e618b8529a51975f56d6"
meta-seccamviewer    = "master:b1d1ce50de17eb5337827f6d1db268aee945f555"
meta                 = "master:27cb44e83b4bc63484bbea9988701e9d9b3f7142"
meta-oe              
meta-python          
meta-networking      = "master:6e83f210ff34d60a9422b4d619939c0386453058"
meta-chromium        = "master:0d7073186806f98ba470257edb8a04f9856c46fd"
meta-clang           = "master:e8a0bada69637ff7a767d387b803a42123aead0b"
meta-python2         = "master:5f4db648e00cf1c41e0999d3b184a02cde225143"
```
[run.do_compile.24589.txt](https://github.com/OSSystems/meta-browser/files/6185546/run.do_compile.24589.txt)
[local.conf.txt](https://github.com/OSSystems/meta-browser/files/6185548/local.conf.txt)
[bblayers.conf.txt](https://github.com/OSSystems/meta-browser/files/6185550/bblayers.conf.txt)

kraj, 2021-03-22:
it seems native piece is failing 
```
[15128/50648] python "../../build/toolchain/gcc_link_wrapper.py" --output="yocto_native/brotli" -- clang++ -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=gold -Wl,--threads -Wl,--thread-count=4 -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,--disable-new-dtags -L/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -L/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,-O1 -o "yocto_native/brotli" -Wl,--start-group @"yocto_native/brotli.rsp"  -Wl,--end-group  -ldl -lpthread -lrt
FAILED: yocto_native/brotli 
python "../../build/toolchain/gcc_link_wrapper.py" --output="yocto_native/brotli" -- clang++ -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=gold -Wl,--threads -Wl,--thread-count=4 -m64 -Wl,-O2 -Wl,--gc-sections -rdynamic -pie -Wl,--disable-new-dtags -L/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -L/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/z/build-master/6951/seccamviewer/build/tmp-glibc/work/cortexa53-oe-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,-O1 -o "yocto_native/brotli" -Wl,--start-group @"yocto_native/brotli.rsp"  -Wl,--end-group  -ldl -lpthread -lrt
clang-12: error: invalid linker name in argument '-fuse-ld=gold'

```

twoerner, 2021-03-23:
> it seems native piece is failing, do you have gold linker installed on host ?

I'm pretty sure I don't.
It's just invoked with "gold"? If so, I don't.

I'm also pretty sure I don't have clang installed natively (i.e. via my distros package feed).

As I mentioned before, specifying:
```
DISTRO_FEATURES += "ld-is-lld"
```
works as a work-around for me.

I perform this build roughly once a week. It compiled fine the previous weekend, but then failed this past one.

kraj, 2021-03-23:
clang expects ld.gold to be in same directory as itself so in this case if you add binutils-native to dependencies then I think it will work, I am not able to reproduce it here, I built it for qemuarm64 with ld-is-gold here as well. So I would still like to understand where the difference lies. Using lld instead of gold perhaps is a good idea regardless.

yifan19, 2021-03-29:
Saw a segmentation issue with yocto/native build with ld.gold 
meta-browser 0d7073186806f98ba470257edb8a04f9856c46fd
meta-clang e8a0bada69637ff7a767d387b803a42123aead0b

in this configuration, dynamic layer in meta-clang wasn't removed yet and meta-browser hasn't added the GN ldd flag.
```
FAILED: yocto_native/flatc 
python "../../build/toolchain/gcc_link_wrapper.py" --output="yocto_native/flatc" -- clang++ -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -
Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -fuse-ld=gold -Wl,--threads -Wl,--thread-count=4 -m64 -Wl,-O2 -Wl,--gc
-sections -rdynamic -pie -Wl,--disable-new-dtags -L/yow-lpggp32/yyu1/yocto/b_poky/build/tmp/work/core2-64-poky-linux/chromium-x11/89.0.4389.
90-r0/recipe-sysroot-native/usr/lib                         -L/yow-lpggp32/yyu1/yocto/b_poky/build/tmp/work/core2-64-poky-linux/chromium-x11
/89.0.4389.90-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/yow-lpggp
32/yyu1/yocto/b_poky/build/tmp/work/core2-64-poky-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/usr/lib                         -
Wl,-rpath-link,/yow-lpggp32/yyu1/yocto/b_poky/build/tmp/work/core2-64-poky-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot-native/lib     
                    -Wl,-rpath,/yow-lpggp32/yyu1/yocto/b_poky/build/tmp/work/core2-64-poky-linux/chromium-x11/89.0.4389.90-r0/recipe-sysroot
-native/usr/lib                         -Wl,-rpath,/yow-lpggp32/yyu1/yocto/b_poky/build/tmp/work/core2-64-poky-linux/chromium-x11/89.0.4389.
90-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/yow-lpggp32/yyu1/yocto/b_p
oky/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -o "yocto_native/flatc" -Wl,--start-group @"yocto_native/flatc.rsp"  
-Wl,--end-group  -ldl -lpthread -lrt
clang-12: error: unable to execute command: Segmentation fault
```

if I revert b8ef5c875c789e1e704d22aca80ba05e9583b9bf in meta-clang, the build is fine (tested it on wrlinux, not poky)

kraj, 2021-03-29:
@yifan19 we have been using lld forever but then layer rearchitecting dropped the change we were carrying in meta-clang accidentally, this change is now proposed https://github.com/OSSystems/meta-browser/pull/482 please apply this locally and this should return you to normal days.

yifan19, 2021-03-30:
@kraj 15228b0 is the direct fix, i'll run some builds with using llvm runtime lib
@twoerner native clang uses host gcc toolchain to build, so if you have the gold linker, it should be somewhere in your bin directory. But the better solution would be backporting or updating to 15228b0 to use the more tested ldd linker
```
which ld.gold
/usr/bin/ld.gold
```

twoerner, 2021-04-08:
Yes, it appears to work for me now without having to explicitly set "ld-is-lld" in my configuration.
I assume #489 is the reason.

closed 2021-04-08 by twoerner

shr-project, 2021-04-08:
@yifan19 @twoerner for binutils-2.35.1 currently in gatesgarth you might find this fix useful:
https://lists.openembedded.org/g/openembedded-core/message/150302