OSSystems/meta-browser #486: meta-python2 blacklisting everything except python2-native

yifan19, 2021-03-22:
building chromium seems to break with newest meta-python2 blacklisting efforts

`https://git.openembedded.org/meta-python2/commit/?id=10c65fc18998d302f34634ffceaf31ab335056dd`

```
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'python' (but /yow-lpggp32/yyu1/wr_linux/layers/meta-browser/recipes-browser/chromi/)
python was skipped: Recipe is blacklisted: python2 is out of support for long time, read https://www.pythont"
NOTE: Runtime target 'chromium-x11' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['chromium-x11', 'python']
NOTE: Runtime target 'lib32-wrlinux-image-std' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['lib32-wrlinux-image-std', 'wrlinux-image-std', 'chromium-x11 ]
ERROR: Required build target 'wrlinux-image-std' has no buildable providers.
Missing or unbuildable dependency chain was: ['wrlinux-image-std', 'chromium-x11', 'python']
```__

shr-project, 2021-03-22:
No it doesn't, upgrade meta-browser to use meta-chromium with https://github.com/OSSystems/meta-browser/commit/ae3258d43e06cc0eaf8b3994e30b9d011c4f1b87

yifan19, 2021-03-23:
oops I forgot to pull meta-browser, will test again...

closed 2021-03-23 by yifan19