OSSystems/meta-browser #488: chromium: Switch to building with libc++

rakuco, 2021-03-25:
This was originally discussed in #482, which covered a build issue with the upcoming libstdc++ from GCC 11.

I thought we had already been using libc++ ever since we started building with clang in #290, but apparently we only switched compilers back then. libstdc++ is not officially supported or tested upstream, so it is up to the community to send patches when things break, and the issue in #482 is another example. Switching to libc++ would get us closer to what upstream builds and tests and allow us to remove a few patches we currently carry.

Things to consider:
* Whether to use Chromium's own bundled libc++ and compiler_rt copy. Doing so would get us even closer to what upstream builds and ships, but using bundled libraries is something frowned upon by OE and Yocto and there may be a risk that things don't work when mixing them with the older clang from meta-clang ("older" compared to the upstream LLVM snapshot that Chromium ships and uses as a compiler).
* What to do with the native parts of the build. Setting `RUNTIME = "llvm"` in `chromium.inc` does not cause libc++ to be used for the native parts of the build, so we still risk running into libstdc++ build issues. Not sure if this is something that needs to be fixed in meta-clang.
* Testing. I've been doing builds with `RUNTIME = "llvm"` here, but help running the builds and doing more builds under other settings is also appreciated.

rakuco, 2021-03-25:
cc @kraj @msisov

kraj, 2021-03-25:
FWIW, I think using  `RUNTIME = "llvm"` is a good first step and improves the situation, as far as using vendored libc++, it also mean we should be using vendored llvm/clang as well.  I would say lets run with this for considerable time and see how some of chromium upgrades pan out, if we start seeing issues due to different versions of libc++ then we have to solve it, otherwise we will cross the bridge when we reach there.

I have been building for qemuarm64/musl, qemux86-64/musl with RUNTIME = "llvm"  using Clang 12.0.0-RC3 from meta-clang master without issues.

rakuco, 2021-03-25:
Do you know if anything needs to be done in meta-clang to support building the native bits with libc++ too?

kraj, 2021-03-25:
> Do you know if anything needs to be done in meta-clang to support building the native bits with libc++ too?

yes, right now we only build `clang-native` but none of runtime is built for native machine.

MastaG, 2021-03-25:
I've also tested a RUNTIME="llvm" build of Chromium a couple of times for armv7 and it worked fine.

msisov, 2021-03-25:
going to trigger our bot with this change.

rakuco, 2021-03-25:
For the record, I've test-built the following configurations so far:
* chromium-x11:
  * dunfell, intel-core-i7-64
  * gatesgarth, intel-core2-32
* chromium-ozone-wayland:
  * gatesgarth, qemuarm64
 
I'd appreciate it if people could also run their builds to make sure nothing crashes weirdly.

rakuco, 2021-03-25:
> yes, right now we only build `clang-native` but none of runtime is built for native machine.

Is that in your todo list, or is there a workaround we could apply to the recipe in the meantime? Otherwise I think we'll need to keep a few libstdc++ patches around for the native binaries.

kraj, 2021-03-25:
> > yes, right now we only build `clang-native` but none of runtime is built for native machine.
> 
> Is that in your todo list, or is there a workaround we could apply to the recipe in the meantime? Otherwise I think we'll need to keep a few libstdc++ patches around for the native binaries.

No its not in backlog I think it would be nice if you can create a github issue for this enhancement for meta-clang.

r1mikey, 2021-03-25:
FWIW, I'm going to run dunfell, intel-core-i7-64(ish) with RUNTIME="llvm" on a large subset of $WORK devices from Friday.  I'll let that soak for a few days and report back.

rakuco, 2021-03-25:
Awesome, thank you!

kraj, 2021-03-26:
chromium-x11 - (master branches with glibc)
qemux86_64 and qemuarm64 works too.

msisov, 2021-03-26:
> going to trigger our bot with this change now and report once it's done (tomorrow).

works fine with all the targets I typically use (mentioned in the commit messages in chromium recipe updates)

r1mikey, 2021-03-26:
I'm still running my testing, but early results (about 2 hours soak time) shows no regressions (0.45% crash rate over two hours in the control group, 0.15% in the test group).

r1mikey, 2021-03-26:
A four hour soak test with a large set of fairly Javascript-heavy applications (mostly React) in a test group shows no regressions with RUNTIME="llvm".  The test group sees a 4 hour crash rate of 0.71%, which the much larger control group sees a four hour crash rate of 0.81%.

rakuco, 2021-03-26:
Thanks a lot @r1mikey, that's very useful! Do you know what's causing these crashes? Is there something we need to check either here in meta-browser or upstream?

r1mikey, 2021-03-26:
> Thanks a lot @r1mikey, that's very useful! Do you know what's causing these crashes? Is there something we need to check either here in meta-browser or upstream?

Nothing to look into just yet.  We get quite a few out-of-memory issues due to displaying user content, and the next one is Weston seeming to race with Chromium, where we're missing some env vars at startup.  There are a few others, but we're in the process of building a categoriser for the errors so we can better identify crashing paths.  Once we've got that in place I can start sharing details.

r1mikey, 2021-03-31:
Wrapping this up before we start our weekly push and make the data dirty... looking back over a 96 hour window, which is equivalent to about 120k hours of machine time, we see no regressions that affect any of our metrics with this change (alongside with the lld change) running in a test group.

rakuco, 2021-03-31:
Thanks everyone for your reports! I've submitted #492 to switch `RUNTIME` to `"llvm"`.

PR #492 (2021-03-31): chromium: Switch to using libc++ and compiler_rt in target builds.
This follows what upstream has done in its official builds for years. We had
already been using clang in our builds, but with libstdc++. This has
occasionally caused a number of build issues over the years (e.g. #482), so
it makes sense to just follow upstream's toolchain choices more closely.

Due to https://github.com/kraj/meta-clang/issues/449 the native parts of the
build are still using libstdc++ though, so not all libstdc++-related patches
could be removed in this commit.

This change has been tested exhaustively by multi [... cut]
commit 49beed91 (2021-03-31): chromium: Switch to using libc++ and compiler_rt in target builds.

This follows what upstream has done in its official builds for years. We had
already been using clang in our builds, but with libstdc++. This has
occasionally caused a number of build issues over the years (e.g. #482), so
it makes sense to just follow upstream's toolchain choices more closely.

Due to https://github.com/kraj/meta-clang/issues/449 the native parts of the
build are still using libstdc++ though, so not all libstdc++-related patches
could be removed in this commit.

This change has been tested exhaustively by mult [... cut]
closed 2021-03-31 by rakuco