OSSystems/meta-browser #490: chromium: virtual keyboard extension recipe

yifan19, 2021-03-26:
Came across the extension
https://chrome.google.com/webstore/detail/virtual-keyboard/pflmllfnnabikmfkkaddkoolinlfninn
which has its source code available at 
https://github.com/xontab/chrome-virtual-keyboard

I made a small recipe for installing that into rootfs and linked the extension 

are you interested in maintaining this extension recipe?

rakuco, 2021-04-06:
@otavio @kraj 
I don't have any strong opinion either way. Upstream hasn't seen any activity in years so there wouldn't be much maintenance here, OTOH we've never packaged any other extensions before.

kraj, 2021-04-06:
If it is stable and not stale then we can think about it but usually someone needs to keep testing it with updates and upgrades we go through so we if there is some commitment for that then it would work but otherwise we should not apply it as it will bitrot of fail on us and cause unwanted work

schnitzeltony, 2021-04-06:
Looks like it can be installed by webstore - so why add a recipe for it?

yifan19, 2021-04-06:
> Looks like it can be installed by webstore - so why add a recipe for it?

* deploying/scaling is tricky if the target machines don't have access to the Internet, i did look up chromium policies, but I think it still required some kind of local server/mirror 
* not all extensions are open source and/or permissively licensed and getting from the chrome web store would also involve going through their terms and services.
https://developer.chrome.com/docs/webstore/terms/

diegorondini, 2021-04-07:
@yifan19 can you share the recipe? I'm interested in giving it a shot...

diegorondini, 2021-04-09:
> @diegorondini I have pushed it to
> [yifan19@e51dc82](https://github.com/yifan19/meta-browser/commit/e51dc829c38691cadecd37b552c1dff915815bda)

Thank you! It built just fine.

yifan19, 2021-04-22:
I have done a little bit more work to integrate it in chromium 
it is now available as a pull request https://github.com/OSSystems/meta-browser/pull/503