OSSystems/meta-browser #491: chromium: Compilation with ICECC fails

Escalion, 2021-03-29:
When attempting to compile with ICECC, ICECC reports unable to get ICECC_CC or ICECC_CXX - which results in the package falling back to local compilation.
To compound this issue, ninja is then executed with the -j value that ICECC was working with, in my case -j 200. This results in a very fast memory starvation.

I suspect this is due to the fact that ICECC is not aware of the toolchain that is created in gn-utils.inc.

I haven't had much time to look into this, but thought I would post the issue here as a placeholder if I find the cause.

rakuco, 2021-04-06:
I haven't used icecc myself in years, so I don't know how well it works with upstream these days. Do you know if it currently works fine with clang?

Escalion, 2021-04-15:
@rakuco - I don't have the foggiest. I haven't dived too deeply into it as I am currently trying to integrate a system. I'm also unable to devtool modify chromium-ozone-wayland due to a patch being unable to apply to tools/v8_context_snapshot/BUILD.gn

Escalion, 2021-04-15:
Just as a note I found this which may be useful, will check if this works at some point and revert:

https://mkollaro.github.io/2015/05/08/compiling-chromium-with-clang-and-icecc.html