OSSystems/meta-browser #505: Input filed with password property will delete 3 characters by backspace one time

cchsu23, 2021-04-29:
step:
1.usr/bin/chromium --app="http://www.netflix.com/login"
2.Input some characters into password filed. 
3.And then , when password is invisible, you press backspace, you will find that UI will delete 3 characters one time.
4. when password is visible, do not have this symptom. delete one characters one by one.

 [video is here](https://drive.google.com/file/d/1c_IydYq_OSM5TrQ2mlSAEXopa4QDpAOL/view?usp=sharing)

rakuco, 2021-04-30:
When filing bugs, please specify at least the recipe you're referring to. In this case the description at least mentions Chromium, but it's still not clear if this is chromium-x11 or chromium-ozone-wayland.

msisov, 2021-04-30:
Also, I don't believe this repo is right place for reporting bugs. Please report them to crbug.com instead. Thanks.

cchsu23, 2021-04-30:
Hi, 
I'am using chromium-ozone-wayland.
OK, I will file the bug also in  crbug.com

cchsu23, 2021-04-30:
https://bugs.chromium.org/p/chromium/issues/detail?id=1204433

cchsu23, 2021-05-06:
still has this issue on 90.0.4430.93

msisov, 2021-05-10:
This will be handled in crbug. closing this.