OSSystems/meta-browser #508: How can I get the newest chromium version?

jwzl, 2021-05-20:
I found #507  fixed the freezen issue, but I use the yocto zeus.
and I found the  LAYERSERIES_COMPAT_chromium-browser-layer = "dunfell gatesgarth hardknott"

and not found zeus.
and how can I apply the M90 chromium in my zeus yocto project?

thank you very much!

msisov, 2021-05-20:
It should probably compile fine if you just add zeus to the list of compatible layers.

jwzl, 2021-05-21:
I meet the issue:
 ../../base/containers/checked_iterators.h:241:8: error: explicit specialization of undeclared template struct '__is_cpp17_contiguous_iterator'

msisov, 2021-05-21:
are you using clang? you can try to use one from the master branch.. who knows..maybe it will work :) otherwise, you'll have to fix these compilation failures.

msisov, 2021-05-24:
i think this can be closed

closed 2021-05-24 by msisov

jwzl, 2021-05-26:
I build the chromium okay, but generate image failed:
 imx-image-full-1.0-r0 do_rootfs: Postinstall scriptlets of ['chromium-ozone-wayland'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.