OSSystems/meta-browser #509: build the newest chromium issue at zeus

jwzl, 2021-05-26:
continus #508
I build the chromium okay, but generate image failed:
imx-image-full-1.0-r0 do_rootfs: Postinstall scriptlets of ['chromium-ozone-wayland'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
how can I fix it?
thank you very much!
Regards

jwzl, 2021-05-28:
@msisov  thank you

msisov, 2021-05-28:
@otavio @kraj @rakuco  any ideas?

kraj, 2021-05-28:
can you dig more details from your rootfs logs ?

jwzl, 2021-05-31:
this is that I can found the only error in log file

jwzl, 2021-05-31:
DEBUG: Executing python function rootfs_deb_bad_recommendations
DEBUG: Python function rootfs_deb_bad_recommendations finished
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-support/bmap-tools/bmap-tools_3.5.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/apt/apt-native_1.2.31.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/dpkg/dpkg_1.19.7.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-extended/pbzip2/pbzip2_1.1.13.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/meta-imx/meta-bsp/recipes-bsp/u-boot/u-boot-imx_2020.04.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/meta-imx/meta-bsp/recipes-fsl/m4-demos/imx-m4-demos_2.8.0.1.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/meta-imx/meta-bsp/recipes-bsp/imx-mkimage/imx-boot_1.0.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-core/glibc/cross-localedef-native_2.30.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/e2fsprogs/e2fsprogs_1.45.3.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', 'virtual:native:/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.2.bb:do_populate_sysroot', '/home/Work/imx8/yocto_zeus/sources/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot']
NOTE: Installed into sysroot: ['prelink-native', 'makedevs-native', 'ldconfig-native', 'apt-native', 'dpkg-native', 'pbzip2-native', 'mklibs-native', 'update-rc.d-native', 'pigz-native', 'opkg-utils-native', 'systemd-systemctl-native', 'binutils-native', 'elfutils-native', 'shadow-native', 'curl-native', 'db-native', 'perl-native', 'gdk-pixbuf-native', 'gtk+3-native', 'desktop-file-utils-native', 'mkfontdir-native', 'mkfontscale-native', 'debianutils-native', 'nss-native', 'flex-native', 'shared-mime-info-native', 'libjpeg-turbo-native', 'libpng-native', 'gobject-introspection-native', 'fontconfig-native', 'libxdamage-native', 'libxi-native', 'pango-native', 'libxrandr-native', 'libxext-native', 'at-spi2-atk-native', 'libxcursor-native', 'libxrender-native', 'atk-native', 'libxcomposite-native', 'libx11-native', 'libxfixes-native', 'cairo-native', 'util-macros-native', 'xorgproto-native', 'freetype-native', 'libfontenc-native', 'nspr-native', 'libxml2-native', 'intltool-native', 'cmake-native', 'expat-native', 'gperf-native', 'fribidi-native', 'harfbuzz-native', 'libxft-native', 'dbus-native', 'at-spi2-core-native', 'xtrans-native', 'libxcb-native', 'libsm-native', 'pixman-native', 'font-util-native', 'libxml-parser-perl-native', 'icu-native', 'libxtst-native', 'libxau-native', 'libpthread-stubs-native', 'libxdmcp-native', 'xcb-proto-native', 'libice-native']
NOTE: Skipping as already exists in sysroot: ['bmap-tools-native', 'pseudo-native', 'u-boot-imx', 'imx-m4-demos', 'imx-boot', 'qemuwrapper-cross', 'cross-localedef-native', 'e2fsprogs-native', 'depmodwrapper-cross', 'python3-six-native', 'python3-setuptools-native', 'python3-native', 'quilt-native', 'gnu-config-native', 'libtool-native', 'automake-native', 'autoconf-native', 'qemu-native', 'glib-2.0-native', 'gettext-native', 'xz-native', 'gettext-minimal-native', 'zlib-native', 'bzip2-native', 'pkgconfig-native', 'gcc-runtime', 'glibc', 'kmod-native', 'optee-os', 'imx-atf', 'firmware-imx-8m', 'openssl-native', 'patch-native', 'attr-native', 'util-linux-native', 'texinfo-dummy-native', 'unzip-native', 'gdbm-native', 'sqlite3-native', 'libffi-native', 'readline-native', 'libtirpc-native', 'libnsl2-native', 'm4-native', 'libpcre-native', 'meson-native', 'gtk-doc-native', 'ninja-native', 'libgcc', 'linux-libc-headers', 'libcap-ng-native', 'ncurses-native', 're2c-native']
DEBUG: sed -e 's:^[^/]*/:/home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/imx8mmlpddr4evk-poky-linux/imx-image-full/1.0-r0/recipe-sysroot-native/:g' /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/sysroots-components/x86_64/apt-native/fixmepath /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/sysroots-components/x86_64/dpkg-native/fixmepath /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/sysroots-components/x86_64/elfutils-native/fixmepath /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/sysroots-components/x86_64/curl-native/fixmepath /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/sysroots-components/x86_64/perl-nat [... cut ...]
WARNING: Postinstall for package chromium-ozone-wayland failed with 1:

ERROR: Postinstall scriptlets of ['chromium-ozone-wayland'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
Details of the failure are in /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/imx8mmlpddr4evk-poky-linux/imx-image-full/1.0-r0/temp/log.do_rootfs.
DEBUG: Python function do_rootfs finished

jwzl, 2021-06-01:
final I found it can cause by the chromium.inc:27  inherit mime-xdg 
and I found the chromium.inc of the old version  inherit the setuptools.
and my yocto not use mime-xdg.
How Can I fix this issue?

@msisov @kraj

jwzl, 2021-06-02:
Hello ?  thanks!

r1mikey, 2021-06-02:
You can simply remove the "inherit mime-xdg" from the recipe.

jwzl, 2021-06-03:
It's also a way, but I don't want to maintain a branch.
If the master branch will reomve it  in future?

otavio, 2021-06-03:
@jwzl upgrade your layer; so it avoids this mess and reduce maintenance work for the future.

jwzl, 2021-06-04:
thanks @otavio @r1mikey  @msisov
I fixed the issue by add a chromium-ozone-wayland_%.bbappend
`
mime_xdg_postinst(){
	#override the mime_xdg_postinst
}
`
It works well!

Thank you very much!

closed 2021-06-04 by jwzl