OSSystems/meta-browser #510: build failed when disable gbm

jwzl, 2021-06-03:
when I build chromium-ozone-wayland 90.0.4430.93 , I add some
` GN_ARGS_append_mx8 = " use_system_minigbm=false use_wayland_gbm=false"`
 it has many errors:
/home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: warning: /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot/usr/lib//aarch64-poky-linux/9.2.0/../../libc++.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
| /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: warning: /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot/usr/lib//aarch64-poky-linux/9.2.0/../../libc++.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
| /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: warning: /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot/usr/lib//aarch64-poky-linux/9.2.0/../../libc++.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
| /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: warning: /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot/usr/lib//aarch64-poky-linux/9.2.0/../../libc++.so.1: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
| /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: warning: /home/Work/imx8/yocto_zeus/imx8mmevkxwayland/tmp/work/aarch64-mx8mm-poky-linux/chromium-ozone-wayland/90.0.4430.93-r0/recipe-sysroot/usr/lib//aarch64-poky-linux/9.2.0/../../libc++abi.so: unsupported GNU_PROPERTY_TYPE (5) type: 0xc0000000
| clang-10: error: linker command failed with exit code 1 (use -v to see invocation)

jwzl, 2021-06-04:
I tested the gbm library can't work well for imx8mm , so we must disable the gbm.
@otavio  @r1mikey @msisov

msisov, 2021-06-04:
You don't need to disable gbm at all. Please revert the change and let use_system_minigbm=true. imx8 does have libgbm, but it doesn't have drm render nodes, which makes the dmabuf unusable in Ozone/Wayland.

More over, if one platform doesn't support that path, it doesn't mean we should disable that on compile time.

jwzl, 2021-06-04:
@msisov 
I have compiled it  without the line "GN_ARGS_append_mx8 = " use_system_minigbm=false use_wayland_gbm=false""
but I found chromium has a black square when I moving the mouse on the web view.

as you said imx8 no drm render nodes, and what should I do for this issue? 
thank you very much!

msisov, 2021-06-04:
If no drm render nodes are found, there are two paths -

If --in-process-gpu is not passed, Chromium will try to use dmabuf path (requires drm render nodes) or sw composition path.
If --in-process-gpu is passed, Chromium will try to use dmabuf path, then Wayland EGL, then sw composition.

Given that imx8 doesn't have drm render nodes (though, AFAIK you can enable those if you use etnaviv), you need to pass --in-process-gpu and Chromium will simply use Wayland EGL path.

jwzl, 2021-07-14:
Dear msisov:
I found NXP disabled system and wayland GBM path in the bbapend file.
and imx8mm dose have  drm render nodes named renderD128.
I guess the GBM not work well for chromunim, So they  disable it.
1. how to disable GBM and used wayland-egl?
2. when I build chromim with gbm enabled, the chromium seem cause weston crash some times,
and I restart weston, it will okay! 

3.   
`FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI_append = " file://0002-chromium-fix-build-after-y2038-changes-in-glibc.patch \
                   file://Fix-build-breaks-on-non-gbm-machines.patch \
"

REQUIRED_DISTRO_FEATURES = "wayland"

DEPENDS += "\
        libxkbcommon \
        virtual/egl \
        wayland \
        wayland-native \
        at-spi2-atk \
"
GN_ARGS += ' \
        system_wayland_scanner_path="${STAGING_BINDIR_NATIVE}/wayland-scanner" \
        use_lld=false \
'
# gbm is availiable only for mx8, but still drop it for 8MQ in 2020 Q4 release
GN_ARGS_append_mx6 = " use_system_minigbm=false use_wayland_gbm=false"
GN_ARGS_append_mx7 = " use_system_minigbm=false use_wayland_gbm=false"
GN_ARGS_append_mx8 = " use_system_minigbm=false use_wayland_gbm=false"

CHROMIUM_EXTRA_ARGS_append = " --disable-features=VizDisplayCompositor --in-process-gpu --disable-gpu-compositing"`

jwzl, 2021-07-15:
@msisov

msisov, 2021-07-19:
use_wayland_gbm=false is enough to disable the dmabuf path. But don't forget to run the browser with --in-process-gpu flag then.

closed 2022-07-22 by jwzl