OSSystems/meta-browser #511: chromium-x11 90.0.4430.93 cannot open web-pages

schnitzeltony, 2021-06-06:
My build environment:
* All layers at latest / bleeding-edge master
* Machine RaspberryPi 4 / cortexa72 (aarch64) / rt-kernel 5.10.35-rt38
* PACKAGECONFIG = "impl-side-painting proprietary-codecs"

When starting from console I see loads of complaints as:
> [1411:1411:0604/121512.753371:ERROR:network_service_instance_impl.cc(330)] Network service crashed, restarting service.

Rolling back to 89.0.4389.114 makes chromium-x11 work again (this issue was created on Raspi)

Googled around a bit but similar issues were not found - all search hits were years ago

schnitzeltony, 2021-06-20:
FWIW: 

On current master/91.0.4472.77 problem was not solved: chromium-x11 remains unusable

Now PACKAGECONFIG = "proprietary-codecs" (impl-side-painting was just removed)

msisov, 2021-06-21:
I'm sorry you experience problems with Chromium, but I don't believe this is the best place to report bugs. Please report them to 
the Chromium bugtracker instead. Thanks!

TonyTascioglu, 2021-06-28:
I'm also experiencing this issue on a build with meta-browser on master and the other layers on hardknott.

You can wget a page and open it in chromium, but if you try loading a page in chromium, it gives the following error:
`network_service_instance_impl.cc(331)] Network service crashed, restarting service.`

This happens with the latest chromium 91.0.4472.114, I can test older versions if needed but it looks like it's happened for a bit.

rakuco, 2021-06-29:
Are you able to get any backtraces or more logging messages? It's not clear if the same process is crashing and being restarted in a loop or if it's something else.

schnitzeltony, 2021-06-29:
Will start a build for latest chromium. Is there a knob to increase verbosity?

rakuco, 2021-06-29:
I'd start with what's described [here](https://chromium.googlesource.com/chromium/src/+/HEAD/docs/linux/debugging.md#logging). If it's possible to do a debug build, it should output even more information by default.

TonyTascioglu, 2021-07-05:
FWIW: Running the same chromium version in qemux86_64 seems to work fine, but fails on bare metal on the Raspberry Pi. I'll test qemuarm64 as well to see if it works.

I have some screenshots from the Raspberry Pi generated with the verbose setting from above, in the 6th one, you can see the network service crashed error. I'll try to get the actual text log files off as well to make it easier, apologies in advance for the quality.

https://ibb.co/XXJ6F2s
https://ibb.co/hH3MDZs
https://ibb.co/frsWFLQ
https://ibb.co/Bjcc1Fp
https://ibb.co/swtwZQy
https://ibb.co/SyqqgPz

rakuco, 2021-07-06:
Does starting chromium with `--no-sandbox` make it stop crashing? It could be some seccomp policy that's only failing on aarch64 (or the specific kernel used by the rpi4).

I also see https://crbug.com/1221442 mention network service crashes, but I guess it's unrelated since we're not seeing anything related to prctl().

TonyTascioglu, 2021-07-06:
update: loading websites from chromium in qemu works from both qemux86_64 and qemuarm64.
When I tried it with --no-sandbox on the Pi last week, I don't believe it worked, but I'll give it another try and get the full logs.

TonyTascioglu, 2021-07-08:
No luck with --no-sandbox, it results in the same error. Tested with the most recent version (Chromium 91.0.4472.114 stable), which worked on qemuarm64 and and qemux86_64.
I have also checked that the pi is not out of memory from meminfo (https://pastebin.com/EzHJMhau).

rakuco, 2021-07-08:
OK, thanks for checking. I guess the only option left is to try to get a backtrace :/

schnitzeltony, 2021-08-18:
Now that I can build again (https://github.com/OSSystems/meta-browser/pull/542), I can come back to this. If I do start with chromium with `--enable-features=NetworkServiceInProcess`

> morona@raspberrypi4-64:~$ chromium --enable-features=NetworkServiceInProcess
> [1888:1888:0818/213818.555185:FATAL:double_fork_and_exec.cc(131)] execv /usr/lib/chromium/crashpad_handler: No such file or directory (2)
> Illegal instruction
> 

My thoughts:
* `FATAL:double_fork_and_exec.cc...` is https://github.com/OSSystems/meta-browser/issues/541
* For the `Illegal instruction`: Since chromium seems to work on qemuarm64: There are different tunes: qemuarm64 has `aarch64 armv8a crc cortexa57` while Raspi4-64 has `aarch64 armv8a crc crypto cortexa72` - maybe a pointer?

schnitzeltony, 2022-01-01:
After long time I think this was an raspberrypi-64 problem.
Currently chromium-x11 runs fine and I think it was solved by [1-2]. I will close this issue. If others still see problems feel free to re-open.

[1] https://github.com/agherzan/meta-raspberrypi/issues/964
[2] https://github.com/agherzan/meta-raspberrypi/commit/cd234925fa59197a52f4cb5011902ab5867c684f

closed 2022-01-01 by schnitzeltony

rakuco, 2022-01-06:
Awesome, thanks for keeping track of this issue, glad to hear everything is working now!