OSSystems/meta-browser #513: chromium-ozone-wayland doesn't run on raspberrypi4 (yocto/dunfell)?

matt-hammond-001, 2021-06-07:
Has anyone been successful in using chromium-ozone-wayland on a raspberrypi4? I'm trying to get it working in wayland for `core-image-minimal` on poky `dunfell`.

I have the following config that works on a RPi3 with MACHINE=raspberrypi3 but fails on a RPi4. Is it something I'm doing wrong? Or is there a bug? On the RPi4, wayland starts, but chromium does not launch correctly.

Any suggestions/advice much appreciated.

Command line and error output:
```
$ chromium --no-sandbox
[709:709:0607/152414.179250:ERROR:browser_dm_token_storage_linux.cc(94)] Error: /etc/machine-id contains 0 characters (32 were expected).
[709:740:0607/152414.303758:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[709:740:0607/152414.303912:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[735:735:0607/152414.351041:ERROR:gpu_init.cc(426)] Passthrough is not supported, GL is egl
[709:784:0607/152414.714769:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.
UPower was not provided by any .service files
[709:784:0607/152414.715953:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesk
top.UPower was not provided by any .service files
[709:784:0607/152414.717031:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesk
top.UPower was not provided by any .service files
drmPrimeFDToHandle() failed: Cannot allocate memory
Failed to create scanout resource.
[735:735:0607/152414.987369:ERROR:gl_surface_egl.cc(778)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
[735:735:0607/152414.991994:ERROR:gl_image_egl.cc(38)] Error creating EGLImage: EGL_BAD_ALLOC
[735:735:0607/152414.992545:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(264)] Failed to create GLImage 945x1060, RGBA_8888, usage SCANOUT
drmPrimeFDToHandle() failed: Cannot allocate memory
Failed to create scanout resource.
[735:735:0607/152415.036303:ERROR:gl_surface_egl.cc(778)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
[735:735:0607/152415.036772:ERROR:gl_image_egl.cc(38)] Error creating EGLImage: EGL_BAD_ALLOC
[735:735:0607/152415.038605:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(264)] Failed to create GLImage 945x1060, RGBA_8888, usage GPU_READ
[735:735:0607/152415.039036:ERROR:shared_image_backing_factory_gl_texture.cc(536)] CreateSharedImage: Failed to create bindable image
[735:735:0607/152415.039421:ERROR:shared_image_factory.cc(582)] CreateSharedImage: could not create backing.
[735:735:0607/152415.057591:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.058214:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.058980:ERROR:shared_image_stub.cc(452)] SharedImageStub: context already lost
[735:735:0607/152415.061472:ERROR:shared_image_stub.cc(452)] SharedImageStub: context already lost
[735:735:0607/152415.068997:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.075935:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.076601:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.077121:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.077670:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[735:735:0607/152415.079215:ERROR:raster_decoder.cc(1064)]   RasterDecoderImpl: Context lost during MakeCurrent.
[709:753:0607/152415.171316:ERROR:wayland_event_watcher.cc(249)] Fatal Wayland communication error: Broken pipe
drmPrimeFDToHandle() failed: Cannot allocate memory
Failed to create scanout resource.
[735:735:0607/152415.449951:ERROR:gl_surface_egl.cc(778)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
[735:735:0607/152415.450432:ERROR:gl_image_egl.cc(38)] Error creating EGLImage: EGL_BAD_ALLOC
[735:735:0607/152415.450878:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(264)] Failed to create GLImage 945x1060, RGBA_8888, usage SCANOUT
drmPrimeFDToHandle() failed: Cannot allocate memory
Failed to create scanout resource.
[735:735:0607/152415.482919:ERROR:gl_surface_egl.cc(778)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
[735:735:0607/152415.483383:ERROR:gl_image_egl.cc(38)] Error creating EGLImage: EGL_BAD_ALLOC
[735:735:0607/152415.483811:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(264)] Failed to create GLImage 945x1060, RGBA_8888, usage GPU_READ
[735:735:0607/152415.484205:ERROR:shared_image_backing_factory_gl_texture.cc(536)] CreateSharedImage: Failed to create bindable image
[735:735:0607/152415.484569:ERROR:shared_image_factory.cc(582)] CreateSharedImage: could not create backing.
[797:797:0607/152415.716259:ERROR:gpu_init.cc(426)] Passthrough is not supported, GL is egl
drmPrimeFDToHandle() failed: Cannot allocate memory
Failed to create scanout resource.
[797:797:0607/152416.048405:ERROR:gl_surface_egl.cc(778)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
[797:797:0607/152416.049116:ERROR:gl_image_egl.cc(38)] Error creating EGLImage: EGL_BAD_ALLOC
[797:797:0607/152416.049755:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(264)] Failed to create GLImage 945x1060, RGBA_8888, usage SCANOUT
drmPrimeFDToHandle() failed: Cannot allocate memory
Failed to create scanout resource.
[797:797:0607/152416.082102:ERROR:gl_surface_egl.cc(778)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
[797:797:0607/152416.082585:ERROR:gl_image_egl.cc(38)] Error creating EGLImage: EGL_BAD_ALLOC
[797:797:0607/152416.083044:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(264)] Failed to create GLImage 945x1060, RGBA_8888, usage GPU_READ
[797:797:0607/152416.083457:ERROR:shared_image_backing_factory_gl_texture.cc(536)] CreateSharedImage: Failed to create bindable image
[797:797:0607/152416.083839:ERROR:shared_image_factory.cc(582)] CreateSharedImage: could not create backing.
[797:797:0607/152416.085128:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[797:797:0607/152416.085653:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[797:797:0607/152416.099455:ERROR:skia_output_surface_impl_on_gpu.cc(1495)] Failed to make current.
[... 127 lines cut ...]

DISTRO_FEATURES_remove = "\
    x11 \
"

ENABLE_UART = "1"
HDMI_FORCE_HOTPLUG = "1"
DISABLE_OVERSCAN = "1"
DISABLE_SPLASH = "1"
BOOT_DELAY = "0"
BOOT_DELAY_MS = "0"

PACKAGE_CLASSES = "package_deb"
```

local.conf (RPi4) ... same but with these additions/changes
```
MACHINE = "raspberrypi4"

RPI_KERNEL_DEVICETREE = "bcm2711-rpi-4-b.dtb"
RPI_KERNEL_DEVICETREE_OVERLAYS_append = " overlays/vc4-kms-v3d-pi4.dtbo"
VC4DTBO = "vc4-kms-v3d-pi4"
```

And have also tried when setting boot/config.txt to `dtoverlay=vc4-fkms-v3d` and rebooting, but the same result. Weston starts but chromium generates the same error output.

msisov, 2021-06-09:
Mind to report to the upstream instead? Thanks.

matt-hammond-001, 2021-06-09:
You mean the chromium project?

msisov, 2021-06-09:
Exactly.

rakuco, 2021-06-09:
Specifically, https://bugs.chromium.org -- you can also paste the bug number here so that we can subscribe to it and maybe CC the relevant people.

bestgaurav007, 2022-01-21:
Hi, I am also facing the same issue.
Do you get any solution to start chromium on raspberry pi 4 ?

Escalion, 2023-04-18:
For me it's working using the FKMS driver (although I have some issues with blotting/not-redrawing)

Himaniyocto, 2024-07-17:
Hello All

i am facing similar issue for raspberry pi 5 can you please help

MaxIhlenfeldt, 2024-07-18:
Does `chromium-bin --ozone-platform=wayland --use-angle=opengles --no-sandbox` work?

MaxIhlenfeldt, 2024-07-18:
@Himaniyocto can you try running `chromium-bin --ozone-platform=wayland --use-angle=opengles --no-sandbox` instead of `chromium --no-sandbox`?

Himaniyocto, 2024-07-18:
> @Himaniyocto can you try running `chromium-bin --ozone-platform=wayland --use-angle=opengles --no-sandbox` instead of `chromium --no-sandbox`?

ok sure will try to run but can you have a look do i need to modify something inlocal.conf?