OSSystems/meta-browser #514: build

fangxiaoying, 2021-06-09:


msisov, 2021-06-09:
M83 is really old in terms of Ozone development and it's advised to use the latest recipe.

However, why do you need ozone_platform_x11 set? Can you disable that?

fangxiaoying, 2021-06-09:
> M83 is really old in terms of Ozone development and it's advised to use the latest recipe.
>  Our yocto release is use it. I will try latest recipe.
> However, why do you need ozone_platform_x11 set? Can you disable that?
Because Customer meets an issue, if build chromium-ozone-wayland, when click  more tools -> extensions -> load uppacked
the chromium will crash.    I think may chromium without gtk cause it.   So I try to build it with GTK. but if ozene_platfrom_x11=false may meet many issue.

Could you tell me how to build chromium-ozone-wayland with GTK.  which bb files do I need to change?

msisov, 2021-06-09:
The recipe with Chromium 83.0.4103.116 already has gtk enabled.

`$ git grep -ni "use_gtk"
chromium-ozone-wayland_83.0.4103.116.bb:24:        use_gtk=true \`

fangxiaoying, 2021-06-09:
my branch is false in default. I just modify it.
diff --git a/recipes-browser/chromium/chromium-ozone-wayland_83.0.4103.116.bb b/recipes-browser/chromium/chromium-ozone-wayland_83.0.4103.116.bb
index de9274d..ca91cff 100644
--- a/recipes-browser/chromium/chromium-ozone-wayland_83.0.4103.116.bb
+++ b/recipes-browser/chromium/chromium-ozone-wayland_83.0.4103.116.bb
@@ -21,7 +21,7 @@ GN_ARGS += "\
         use_system_libwayland=true \
         use_system_minigbm=true \
         use_system_libdrm=true \
-        use_gtk=false \
+        use_gtk=true \
 "
meet error,
assert(use_x11 || ozone_platform_x11)
^-----
See //ui/gtk/BUILD.gn:119:5: which caused the file to be included.
    "//ui/events/platform/x11",
    ^-------------------------
WARNING: exit code 1 from a shell command.

Do I need to add GN_ARGS "use_x11 = true" in recipes-browser/chromium/chromium-ozone-wayland_83.0.4103.116.bb?

msisov, 2021-06-09:
I'd suggest you to look at the tree of commits in https://source.chromium.org/chromium/chromium/src/+/main:ui/gtk/BUILD.gn;bpv=1 and identify the commit that fixed that issue. It's unnecessary to enable x11 if you don't need that.

fangxiaoying, 2021-06-15:
I have update chromium version to  V89. and enable GTK.  but It cannot run on my chip. (i.MX8MMEVK)

root@imx8mmevk:~# chromium --no-sandbox
[1038:1104:0920/101925.489837:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1038:1104:0920/101925.490962:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1038:1104:0920/101925.492203:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files

msisov, 2021-06-15:
Run with $ chromium --no-sandbox --ozone-platform=wayland --in-process-gpu

fangxiaoying, 2021-06-15:
same error on my side.

fangxiaoying, 2021-06-15:
This is Debug log. Do you have any idea to debug ? Thx
root@imx8mmevk:~# WAYLAND_DEBUG=1 chromium --no-sandbox --ozone-platform=wayland --in-process-gpu
[1796176.021]  -> wl_display@1.get_registry(new id wl_registry@2)
[1796176.144]  -> wl_display@1.sync(new id wl_callback@3)
[1796176.341] wl_display@1.delete_id(3)
[1796176.360] wl_registry@2.global(1, "wl_compositor", 4)
[1796176.415]  -> wl_registry@2.bind(1, "wl_compositor", 4, new id [unknown]@4)
[1796176.461] wl_registry@2.global(2, "wl_subcompositor", 1)
[1796176.493]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@5)
[1796176.523] wl_registry@2.global(3, "wp_viewporter", 1)
[1796176.552]  -> wl_registry@2.bind(3, "wp_viewporter", 1, new id [unknown]@6)
[1796176.582] wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[1796176.605] wl_registry@2.global(5, "wp_presentation", 1)
[1796176.631]  -> wl_registry@2.bind(5, "wp_presentation", 1, new id [unknown]@7)
[1796176.660] wl_registry@2.global(6, "zwp_alpha_compositing_v1", 1)
[1796176.683] wl_registry@2.global(7, "zwp_relative_pointer_manager_v1", 1)
[1796176.703] wl_registry@2.global(8, "zwp_pointer_constraints_v1", 1)
[1796176.726] wl_registry@2.global(9, "zwp_input_timestamps_manager_v1", 1)
[1796176.746] wl_registry@2.global(10, "wl_data_device_manager", 3)
[1796176.771]  -> wl_registry@2.bind(10, "wl_data_device_manager", 3, new id [unknown]@8)
[1796176.801] wl_registry@2.global(11, "wl_shm", 1)
[1796176.828]  -> wl_registry@2.bind(11, "wl_shm", 1, new id [unknown]@9)
[1796176.857] wl_registry@2.global(12, "weston_touch_calibration", 1)
[1796176.879] wl_registry@2.global(13, "wl_viv", 1)
[1796176.901] wl_registry@2.global(14, "wl_drm", 2)
[1796176.926]  -> wl_registry@2.bind(14, "wl_drm", 2, new id [unknown]@10)
[1796176.957]  -> wl_display@1.sync(new id wl_callback@11)
[1796176.974] wl_registry@2.global(15, "wl_seat", 7)
[1796177.001]  -> wl_registry@2.bind(15, "wl_seat", 5, new id [unknown]@12)
[1796177.031]  -> wl_data_device_manager@8.get_data_device(new id wl_data_device@13, wl_seat@12)
[1796177.056] wl_registry@2.global(16, "zwp_linux_dmabuf_v1", 3)
[1796177.085]  -> wl_registry@2.bind(16, "zwp_linux_dmabuf_v1", 3, new id [unknown]@14)
[1796177.115]  -> wl_display@1.sync(new id wl_callback@15)
[1796177.131] wl_registry@2.global(17, "weston_direct_display_v1", 1)
[1796177.154] wl_registry@2.global(18, "zwp_linux_explicit_synchronization_v1", 2)
[1796177.182]  -> wl_registry@2.bind(18, "zwp_linux_explicit_synchronization_v1", 2, new id [unknown]@16)
[1796177.212] wl_registry@2.global(19, "weston_content_protection", 1)
[1796177.236] wl_registry@2.global(20, "zwp_hdr10_metadata_v1", 1)
[1796177.256] wl_registry@2.global(21, "wl_output", 3)
[1796177.283]  -> wl_registry@2.bind(21, "wl_output", 3, new id [unknown]@17)
[1796177.315] wl_registry@2.global(22, "zwp_input_panel_v1", 1)
[1796177.337] wl_registry@2.global(23, "zwp_input_method_v1", 1)
[1796177.357] wl_registry@2.global(24, "zwp_text_input_manager_v1", 1)
[1796177.386]  -> wl_registry@2.bind(24, "zwp_text_input_manager_v1", 1, new id [unknown]@18)
[1796177.415] wl_registry@2.global(25, "xdg_wm_base", 1)
[1796177.442]  -> wl_registry@2.bind(25, "xdg_wm_base", 1, new id [unknown]@19)
[1796177.471] wl_registry@2.global(26, "zxdg_shell_v6", 1)
[1796177.499]  -> wl_registry@2.bind(26, "zxdg_shell_v6", 1, new id [unknown]@20)
[1796177.528] wl_registry@2.global(27, "wl_shell", 1)
[1796177.549] wl_registry@2.global(28, "weston_desktop_shell", 1)
[1796177.570] wl_registry@2.global(29, "weston_screenshooter", 1)
[1796177.592] wl_callback@3.done(1180)
[1796177.922] wl_display@1.delete_id(11)
[1796177.948] wl_display@1.delete_id(15)
[1796177.963] wl_drm@10.device("/dev/dri/card0")
[1796178.032]  -> wl_drm@10.authenticate(1)
[1796178.055]  -> wl_display@1.sync(new id wl_callback@3)
[1796178.073] wl_drm@10.format(808669761)
[1796178.092] wl_drm@10.format(808669784)
[1796178.106] wl_drm@10.format(808665665)
[1796178.117] wl_drm@10.format(808665688)
[1796178.131] wl_drm@10.format(875713089)
[1796178.142] wl_drm@10.format(875713112)
[1796178.156] wl_drm@10.format(909199186)
[1796178.168] wl_drm@10.format(961959257)
[1796178.181] wl_drm@10.format(825316697)
[1796178.192] wl_drm@10.format(842093913)
[1796178.204] wl_drm@10.format(909202777)
[1796178.214] wl_drm@10.format(875713881)
[1796178.224] wl_drm@10.format(842094158)
[... 184 lines cut ...]
[1796637.608]  -> wl_region@31.add(1, 1057, 943, 2)
[1796637.629]  -> wl_region@31.add(3, 1059, 939, 1)
[1796637.651]  -> wl_surface@15.set_opaque_region(wl_region@31)
[1796637.664]  -> wl_region@31.destroy()
[1796637.697]  -> wl_compositor@4.create_region(new id wl_region@30)
[1796637.712]  -> wl_region@30.add(3, 0, 939, 1)
[1796637.735]  -> wl_region@30.add(1, 1, 943, 2)
[1796637.756]  -> wl_region@30.add(0, 3, 945, 1054)
[1796637.776]  -> wl_region@30.add(1, 1057, 943, 2)
[1796637.797]  -> wl_region@30.add(3, 1059, 939, 1)
[1796637.818]  -> wl_surface@15.set_input_region(wl_region@30)
[1796637.829]  -> wl_region@30.destroy()
[1796638.105]  -> xdg_surface@28.set_window_geometry(0, 0, 945, 1060)
[1796638.165] xdg_surface@28.configure(1181)
[1796638.182]  -> xdg_surface@28.ack_configure(1181)
[1796638.754] wl_keyboard@3.repeat_info(40, 400)
[1796638.828] wl_keyboard@3.keymap(1, fd 110, 47298)
[1796678.814]  -> zwp_linux_explicit_synchronization_v1@24.get_synchronization(new id zwp_linux_surface_synchronization_v1@27, wl_surface@15)
[... cut ...]
[1555:1631:0920/104428.975714:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1555:1631:0920/104428.976514:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files

fangxiaoying, 2021-07-05:
After upgrade chromium to V89 and disable GPU, The issue is fixed on NXP i.MX8M series chips.  GPU have some issue need GPU vendor to fix. Thanks for your help!

closed 2021-07-05 by fangxiaoying