OSSystems/meta-browser #515: chromium: update to 91.0.4472.77

msisov, 2021-06-09:
Stable release announcements since our latest update:
https://chromereleases.googleblog.com/2021/05/stable-channel-update-for-desktop_25.html

Targets and platforms we need to check:

**chromium-x11:**
- dunfell, clang, MACHINE="intel-core2-32"
- dunfell, clang, MACHINE="intel-corei7-64"
- dunfell, clang, MACHINE="raspberrypi3"
- gatesgarth, MACHINE="raspberrypi3"

**chromium-ozone-wayland:**

- dunfell, clang, MACHINE="intel-corei7-64"
- dunfell, clang, MACHINE="qemuarm64"
- gatesgarth, clang, MACHINE="raspberrypi3"
- dunfell, clang, MACHINE="m3ulcb"

PS: testing with hardknott is pending to be added.

r1mikey, 2021-06-16:
This can be closed.

msisov, 2021-06-17:
indeed. thanks

closed 2021-06-17 by msisov