OSSystems/meta-browser #523: Chromium GPU acceleration on RPI3/RPI4

nandra, 2021-06-29:
Hi,

this is not an issue more question. I would like to ask if anyone have experience with gpu acceleration in chromium using this layer. Using proprietary broadcom or VC4? How about playing like 4K video on RPI4? 

Thanks.

rakuco, 2021-06-29:
cc @msisov who might know more about this

schnitzeltony, 2021-06-29:
On Raspi 4:
Had trouble with (use-)egl. After disabling [1] - chrome://gpu/ says it is accelerated and WebGL [2] is up to 10 times faster as with egl.

[1] https://github.com/schnitzeltony/meta-mortsgna/blob/master/appends/meta-browser/chromium-x11_%25.bbappend
[2] https://threejs.org/examples

nandra, 2021-06-30:
@schnitzeltony thanks for info. How about 4k video? Did anybody try it? Thanks.