OSSystems/meta-browser #528: IMX8MM chromium-ozone-wayland configure error

steven-smpct, 2021-07-26:
I am building a wayland OS for a IMX8MM on Ubuntu 20.04.

When I compile I get the following error.

`Loading cache: 100% |#########################################################################################################################################| Time: 0:00:00
Loaded 4851 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "imx8mm-var-dart"
DISTRO               = "fslc-wayland"
DISTRO_VERSION       = "3.3"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:9df1d44de50b175c736d9f7e52f730fac25e3c12"
meta-oe              
meta-multimedia      
meta-python          
meta-filesystems     
meta-gnome           
meta-networking      = "HEAD:9ee0e08ba2395b9cb93f2273bb1018adb3630407"
meta-freescale       = "HEAD:7f12b5e235e67bfe28891d460cb6b1f3025bd0ea"
meta-freescale-3rdparty = "HEAD:7f23af99cb97a12134a46b5b9d497f05b758bf0c"
meta-freescale-distro = "HEAD:916df6d24c0a33a3b1533bde70b6a2724ec77af4"
meta-qt5             = "HEAD:a00af3eae082b772469d9dd21b2371dd4d237684"
meta-swupdate        = "HEAD:2c2c57d3aaae42999b6ef6dcdf67fe52b12735c3"
meta-virtualization  = "HEAD:cd431b8f4498a77a800c19a857eb46717e7c71d4"
meta-variscite-fslc  = "HEAD:eff54271f37b78a7d5e751b3dbbef0e67bd40582"
meta-python2         = "HEAD:8db9e4f6ceae33d7a4f55453d31e69f9858af4eb"
meta-clang           = "hardknott:7ef47d048267cf755c496a3962b34314f6f1f1e1"
meta-chromium        = "master:ecbd3786959592e19fff7a3c82506a04bf1ef970"
meta-smpct           = "hardknott:2e97f4bc92abf4fee4f0a3cc988390817b63627d"

Initialising tasks: 100% |####################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 0 Local 0 Network 0 Missed 0 Current 423 (0% match, 100% complete)
NOTE: Executing Tasks
ERROR: chromium-ozone-wayland-91.0.4472.164-r0 do_configure: Execution of '/home/steven/Work/var-fslc-yocto-hardknott/build_wayland/tmp/work/cortexa53-crypto-mx8mm-fslc-linux/chromium-ozone-wayland/91.0.4472.164-r0/temp/run.do_configure.3383419' failed with exit code 1:
ERROR at //build/config/linux/dri/BUILD.gn:11:20: Disallowed exec_script call.
  dri_driver_dir = exec_script(pkg_config_script,
                   ^-----------------------------
The use of exec_script use is restricted in this build. exec_script
is discouraged because it can slow down the GN run and is easily
abused.

Generally nontrivial work should be done as build steps rather than
when GN is run. For example, if you need to compute a nontrivial
preprocessor define, it will be better to have an action target
generate a header containing the define rather than blocking the GN
run to compute the value.

The allowed callers of exec_script is maintained in the "//.gn" file
if you need to modify the whitelist.
See //content/gpu/BUILD.gn:141:18: which caused the file to be included.
    configs += [ "//build/config/linux/dri" ]
                 ^-------------------------
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/steven/Work/var-fslc-yocto-hardknott/build_wayland/tmp/work/cortexa53-crypto-mx8mm-fslc-linux/chromium-ozone-wayland/91.0.4472.164-r0/temp/log.do_configure.3383419
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-graphics/wayland/wayland_1.19.0.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-multimedia/webp/libwebp_1.2.0.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-extended/xz/xz_5.2.5.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-support/atk/at-spi2-atk_2.38.0.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/bison/bison_3.7.5.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/desktop-file-utils/desktop-file-utils_0.26.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-openembedded/meta-oe/recipes-support/nspr/nspr_4.29.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-gnome/hicolor-icon-theme/hicolor-icon-theme_0.17.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-graphics/jpeg/libjpeg-turbo_2.0.6.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-core/glibc/glibc_2.33.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-browser/meta-chromium/recipes-browser/chromium/gn-native_91.0.4472.164.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-core/expat/expat_2.2.10.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-python2/recipes-devtools/python/python-setuptools_42.0.2.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-freescale/recipes-graphics/imx-gpu-viv/imx-gpu-viv_6.4.3.p1.4-aarch64.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-python2/recipes-devtools/python/python-native_2.7.18.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-extended/gperf/gperf_3.1.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-bsp/pciutils/pciutils_3.7.0.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-core/dbus/dbus_1.12.20.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-graphics/cairo/cairo_1.16.0.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-core/glib-2.0/glib-2.0_2.66.7.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/gcc/libgcc_10.2.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-support/libxslt/libxslt_1.1.34.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/qemu/qemu-native_5.2.0.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-openembedded/meta-oe/recipes-support/nspr/nspr_4.29.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-graphics/xorg-lib/libxkbcommon_1.0.3.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-support/atk/atk_2.36.0.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-gnome/gdk-pixbuf/gdk-pixbuf_2.40.0.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.25.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-openembedded/meta-oe/recipes-support/nss/nss_3.64.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-graphics/freetype/freetype_2.10.4.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.25.bb:do_populate_sysroot', 'virtual:native:/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-openembedded/meta-oe/recipes-support/nss/nss_3.64.bb:do_populate_sysroot', '/home/steven/Work/var-fslc-yocto-hardknott/sources/meta-freescale/recipes-graphics/drm/libdrm_2.4.102.imx.bb:do_populate_sysroot', 'vi [... cut ...]
NOTE: Tasks Summary: Attempted 1991 tasks of which 1990 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds

Summary: 1 task failed:
  /home/steven/Work/var-fslc-yocto-hardknott/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_91.0.4472.164.bb:do_configure
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.`

It seems to have something to do with the DRI. I am not sure if wayland uses it

steven-smpct, 2021-07-26:
I turned off the DRI and it seemed to compile. I did this by add extra and in content/gpu/BUILD.gn for the DRI to check for use_x11. It configured, compilied and got running on the processor. Only thing is that I got some errors about EGL. I don't know if this will be a problem in the future.

`chromium --no-sandbox
EGL: Warning: No default display support on wayland
[596:596:0726/235207.974853:ERROR:gl_surface_egl.cc(1348)] EGL display query failed with error EGL_SUCCESS
[596:596:0726/235207.975125:ERROR:gl_surface_egl.cc(1373)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[596:596:0726/235207.975220:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[596:596:0726/235208.011875:ERROR:viz_main_impl.cc(160)] Exiting GPU process due to errors during initialization
[608:608:0726/235210.034567:ERROR:gpu_init.cc(440)] Passthrough is not supported, GL is swiftshader
[568:568:0726/235210.599175:ERROR:cursor_loader.cc(107)] Failed to load a platform cursor of type kNull
[568:642:0726/235211.446373:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBs
[568:642:0726/235211.447047:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktos
[568:642:0726/235211.447578:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktos`

msisov, 2021-07-27:
On this board, please run chromium with --in-process-gpu.

steven-smpct, 2021-07-27:
With that option, I got a black screen with a blinking cursor and the following below. Maybe I am missing a recipe?

chromium --no-sandbox --in-process-gpu
[489:489:0727/175430.010973:ERROR:cursor_loader.cc(107)] Failed to load a platform cursor of type kNull
[489:554:0727/175430.529283:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBs
[489:554:0727/175430.530531:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktos
[489:554:0727/175430.531488:ERROR:object_proxy.cc(622)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktos
[     1] ES30: some draw get error and skipped during validation
[     2] ES30: some draw get error and skipped during validation
[     3] ES30: some draw get error and skipped during validation
[     4] ES30: some draw get error and skipped during validation
[     5] ES30: some draw get error and skipped during validation
[     6] ES30: some draw get error and skipped during validation
[     7] ES30: some draw get error and skipped during validation
[     8] ES30: some draw get error and skipped during validation
[     9] ES30: some draw get error and skipped during validation
[    10] ES30: some draw get error and skipped during validation
[    11] ES30: some draw get error and skipped during validation

msisov, 2021-07-27:
That’s a known issue and it has been fixed in M92. I’m going to submit an update to this repo for M92 tomorrow or after tomorrow. The update is almost ready.

rakuco, 2021-09-09:
Given we've moved to M92 now, is this still an issue?

steven-smpct, 2021-09-09:
I am doing this for work. I am on another project right now. When I come back to this I will update this issue.

x19chen, 2022-03-16:
How is it going with this issue?

I am facing a similar error
```
ERROR: chromium-ozone-wayland-98.0.4758.102-r0 do_configure: Execution of '/workdir/build_wayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/98.0.4758.102-r0/temp/run.do_configure.1711' failed with exit code 1
ERROR: Logfile of failure stored in: /workdir/build_wayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/98.0.4758.102-r0/temp/log.do_configure.1711
Log data follows:
| DEBUG: Executing shell function do_configure
| ERROR at //build/config/linux/dri/BUILD.gn:11:20: Script returned non-zero exit code.
|   dri_driver_dir = exec_script(pkg_config_script,
|                    ^----------
| Current dir: /workdir/build_wayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/98.0.4758.102-r0/chromium-98.0.4758.102/out/Release/
| Command: python3 /workdir/build_wayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/98.0.4758.102-r0/chromium-98.0.4758.102/build/config/linux/pkg-config.py --dridriverdir dri
| Returned 1 and printed out:
|
| Error from pkg-config.
|
| stderr:
|
| Package dri was not found in the pkg-config search path.
| Perhaps you should add the directory containing `dri.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'dri' found
|
| See //content/gpu/BUILD.gn:132:18: which caused the file to be included.
|     configs += [ "//build/config/linux/dri" ]
|                  ^-------------------------
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/workdir/build_wayland/tmp/work/cortexa9t2hf-neon-mx6qdl-fslc-linux-gnueabi/chromium-ozone-wayland/98.0.4758.102-r0/temp/run.do_configure.1711' failed with exit code 1
ERROR: Task (/workdir/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_98.0.4758.102.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4346 tasks of which 4335 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 4 seconds

Summary: 1 task failed:
  /workdir/sources/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_98.0.4758.102.bb:do_configure
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```
without trying anything else at the moment.

Villemoes, 2022-10-27:
This seems to still be a problem with 105.0.5195.102. To get chromium-ozone-wayland to build, based on the patch posted upthread, I'm applying
```
--- a/content/gpu/BUILD.gn      2021-07-30 16:02:54.298106046 -0700
+++ b/content/gpu/BUILD.gn      2021-07-30 16:03:19.318535373 -0700
@@ -135,7 +135,7 @@
   }
 
   # Use DRI on desktop Linux builds.
-  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux &&
+  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux && ozone_platform_x11 &&
       !is_castos) {
     configs += [ "//build/config/linux/dri" ]
   }
```
(and the same thing previously, just with a slightly different context since the !is_castos is relatively new)

xjordanx, 2024-05-21:
This trail of breadcrumbs has helped me. Thankyou very much. I am now building chromium-ozone-wayland 117.0.5938.132 over nanbield 6.6.3-1.0.0 (Current as of today for IMX YOCTO USER GUIDE from NXP). I will report back here when I get chromium to actually run on the i.MX8 QuadXPlus MEK.

xjordanx, 2024-05-21:
... and to save other people the time rummaging... the file you need to make a change to is deep in the temporary sources working directory tree. Go to the specific directory:
`[**your linux computer or WSL**]\home\<**your username**>\<**your_yocto_dir**>\<**your_build_dir**>\tmp\work\armv8a-mx8-poky-linux\chromium-ozone-wayland\117.0.5938.132\chromium-117.0.5938.132\media\gpu\sandbox`
...then edit the file `BUILD.gn` to add the `&& ozone_platform_x11` test as outlined by @Villemoes above.
This is the contents of my new working `BUILD.gn`:
`# Copyright 2021 The Chromium Authors
# Use of this source code is governed by a BSD-style license that can be
# found in the LICENSE file.

import("//build/config/chromeos/ui_mode.gni")
import("//media/gpu/args.gni")

source_set("sandbox") {
  sources = []
  if (is_linux || is_chromeos_ash) {
    sources += [
      "hardware_video_decoding_sandbox_hook_linux.cc",
      "hardware_video_decoding_sandbox_hook_linux.h",
    ]
  }
  if (is_linux || is_chromeos) {
    sources += [
      "hardware_video_encoding_sandbox_hook_linux.cc",
      "hardware_video_encoding_sandbox_hook_linux.h",
    ]
  }
  deps = [
    "//base",
    "//media/gpu:buildflags",
    "//sandbox/policy:policy",
  ]
  if (use_vaapi) {
    deps += [ "//media/gpu/vaapi:common" ]
  }
  if (use_v4l2_codec) {
    deps += [ "//media/gpu/v4l2" ]
  }
  if (current_cpu != "s390x" && current_cpu != "ppc64" && is_linux && ozone_platform_x11 &&
      !is_castos) {
    # For DRI_DRIVER_DIR.
    configs += [ "//build/config/linux/dri" ]
  }
}`

Asadeds, 2024-11-05:
HI @xjordanx @Villemoes 
i got the same errror, n i tried to resolve as you mentioned here. not resolved and i got the same result or erros