OSSystems/meta-browser #533: Firefox: Can't build with Wayland

apbr, 2021-08-10:
When trying to build Firefox on a yocto setup with Wayland I get the following error:
```
ERROR: Nothing PROVIDES 'startup-notification' (but /work/work/sources/poky/../meta-browser/dynamic-layers/rust-layer/recipes-browser/firefox/firefox_68.9.0esr.bb DEPENDS on or otherwise requires it)
startup-notification was skipped: missing required distro feature 'x11' (not in DISTRO_FEATURES)
ERROR: Required build target 'firefox' has no buildable providers.
Missing or unbuildable dependency chain was: ['firefox', 'startup-notification']
```

It looks like Firefox depends on `startup-notification` which requires x11.
Is Firefox only supported with x11?
Did I choose some wrong configurations?
I appreciate any help to fix this.

This is the full output:
```
$ bitbake firefox
Parsing recipes: 100% |##############################################################################################################################################################################| Time: 0:05:54
Parsing of 3391 .bb files complete (0 cached, 3391 parsed). 4895 targets, 508 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'startup-notification' (but /work/work/sources/poky/../meta-browser/dynamic-layers/rust-layer/recipes-browser/firefox/firefox_68.9.0esr.bb DEPENDS on or otherwise requires it)
startup-notification was skipped: missing required distro feature 'x11' (not in DISTRO_FEATURES)
ERROR: Required build target 'firefox' has no buildable providers.
Missing or unbuildable dependency chain was: ['firefox', 'startup-notification']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

uart-aplex, 2021-12-08:
I also encountered the same error message on a system that only supports wayland. Does anyone know a solution?

ashie, 2021-12-08:
Firefox doesn't support building without X11 even if you want to use only Wayland backend: https://bugzilla.mozilla.org/show_bug.cgi?id=1661450

kwavnet, 2024-06-05:
The next ESR might finally enable that use case, FF 121+ should support it (see https://www.mozilla.org/en-US/firefox/121.0/releasenotes/)