OSSystems/meta-browser #536: rg2h chromium-ozone-wayland compile error

blithe2015, 2021-08-13:
error message:
[33236/33633] aarch64-niic-linux-g++  -march=armv8-a -mtune=cortex-a57 --sysroot=/home/fred/yocto/build-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot -
MMD -MF obj/ui/views/views/message_box_view.o.d   -O2 -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/fred/yocto/build-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108
.igalia.1-r0=/usr/src/debug/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0 -fdebug-prefix-map=/home/fred/yocto/build-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.
1-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/fred/yocto/build-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot= -g1 -DVIEWS_IMPLEMENTATION -DV8_DE
PRECATION_WARNINGS -DUSE_UDEV -DUSE_AURA=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DNO_TCMALLOC -DFULL_SAFE_BROWSING -DSAFE_BROWSING_CSD -DSAFE_BROWSING_DB_LOCAL -DOFFICIAL_BUILD -DCHROMIUM_BUILD -D_FILE_OFFSET_BIT
S=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DTOOLKIT_VIEWS=1 -DUSE_EGL -DU_USING_I
CU_NAMESPACE=0 -DU_ENABLE_DYLOAD=0 -DU_STATIC_IMPLEMENTATION -DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_FILE -DUCHAR_TYPE=uint16_t -DSK_IGNORE_LINEONLY_AA_CONVEX_PATH_OPTS -DSK_HAS_PNG_LIBRARY -DSK_HAS_WEBP_LIBRARY
-DSK_HAS_JPEG_LIBRARY -DSK_SUPPORT_GPU=1 -DSK_GPU_WORKAROUNDS_HEADER=\"gpu/config/gpu_driver_bug_workaround_autogen.h\" -DGOOGLE_PROTOBUF_NO_RTTI -DGOOGLE_PROTOBUF_NO_STATIC_INITIALIZER -DHAVE_PTHREAD -DLEVE
LDB_PLATFORM_CHROMIUM=1 -I../.. -Igen -Igen/shim_headers/libwebp_shim -I../../third_party/khronos -I../../gpu -I../../third_party/libyuv/include -I../../third_party/ced/src -I../../third_party/icu/source/com
mon -I../../third_party/icu/source/i18n -I../../skia/config -I../../skia/ext -I../../third_party/skia/include/c -I../../third_party/skia/include/config -I../../third_party/skia/include/core -I../../third_par
ty/skia/include/effects -I../../third_party/skia/include/encode -I../../third_party/skia/include/gpu -I../../third_party/skia/include/images -I../../third_party/skia/include/lazy -I../../third_party/skia/inc
lude/pathops -I../../third_party/skia/include/pdf -I../../third_party/skia/include/pipe -I../../third_party/skia/include/ports -I../../third_party/skia/include/utils -I../../third_party/skia/src/gpu -I../../
third_party/skia/src/sksl -I../../third_party/skia/modules/skottie/include -I../../third_party/protobuf/src -I../../third_party/libwebm/source -I../../third_party/protobuf/src -Igen/protoc_out -I../../third_
party/leveldatabase -I../../third_party/leveldatabase/src -I../../third_party/leveldatabase/src/include -I../../third_party/mesa_headers -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funw
ind-tables -fPIC -pipe -pthread -Wall -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -fno-delete-null-pointer-checks -Wno-comments -Wno-missing-field-initializers -Wno-unuse
d-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -std=gnu++14 -Wno-narrowing -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -c ../../ui/vi
ews/controls/message_box_view.cc -o obj/ui/views/views/message_box_view.o
[33237/33633] python "../../build/toolchain/gcc_link_wrapper.py" --output="./v8_context_snapshot_generator" -- aarch64-niic-linux-g++  -march=armv8-a -mtune=cortex-a57 --sysroot=/home/fred/yocto/build-rg2/tm
p/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,now -Wl,-z,relro -Wl,-z,defs -Wl,--as-
needed -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-rpath-link=. -Wl,--disable-new-dtags -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o "./v8_context_snapshot_generator" -Wl,--start-group @"./v8_context_snapsho
t_generator.rsp"  -Wl,--end-group   -latomic -ldl -lpthread -lrt -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -ljpeg -lxml2 -lxslt -lm -lresolv -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lfreetype -
lwebpdemux -lwebpmux -lwebp -lexpat -luuid -lxkbcommon -lasound -lz -lwayland-egl -lwayland-client -lgbm -lpci -ldbus-1
[33238/33633] python ../../build/gn_run_binary.py ./v8-qemu-wrapper.sh ./v8_context_snapshot_generator --output_file=v8_context_snapshot.bin
FAILED: v8_context_snapshot.bin
python ../../build/gn_run_binary.py ./v8-qemu-wrapper.sh ./v8_context_snapshot_generator --output_file=v8_context_snapshot.bin
././v8-qemu-wrapper.sh: line 7:  8613 Segmentation fault      (core dumped) PSEUDO_UNLOAD=1 qemu-aarch64 -r 3.14 -L /home/fred/yocto/build-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r
589108.igalia.1-r0/recipe-sysroot -E LD_LIBRARY_PATH=/home/fred/yocto/build-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot/usr/lib64:/home/fred/yocto/bu
ild-rg2/tmp/work/aarch64-niic-linux/chromium-ozone-wayland/71.0.3545.0.r589108.igalia.1-r0/recipe-sysroot/lib64 "$@"
./v8-qemu-wrapper.sh failed with exit code 139

kraj, 2021-08-13:
it seems you are using gcc to build, could you try using clang ?

msisov, 2021-08-13:
Hi. It's a very old 71.0.3545.0 build. Is there a reason you still use that?

blithe2015, 2021-08-13:
> 看来您正在使用 gcc 来构建，您可以尝试使用 clang 吗？

I don't currently use clang to compile, do I need to fill in meta-clang to compile

blithe2015, 2021-08-13:
> 你好。这是一个非常古老的 71.0.3545.0 版本。你还有理由使用它吗？

> 你好。这是一个非常流行的 71.0.3545.0 版本。你还有理由使用它吗？

The version of yocto used by Renesas is relatively low, if you can use the latest version is also possible

blithe2015, 2021-08-13:
> 你好。这是一个非常流行的 71.0.3545.0 版本。你还有理由使用它吗？

我也想用新的版本吖，瑞萨的yocto bsp版本较低升级的话需要很多工作

blithe2015, 2021-08-13:
> Hi. It's a very old 71.0.3545.0 build. Is there a reason you still use that?

The version of yocto used by Renesas is relatively low, if you can use the latest version is also possible

blithe2015, 2021-08-13:
> it seems you are using gcc to build, could you try using clang ?

What else do I need to do to add meta-clang?

rakuco, 2021-09-09:
@blithe2015 is this still happening? You essentially just need to update to a newer meta-browser version and check out the meta-clang branch corresponding to the OE/Yocto release you are using.