OSSystems/meta-browser #541: Chromium does not run due to missing crashpad handler?

MastaG, 2021-08-18:
Hi there,

After updating to 92.0.4515.107 I can't seem to run Chromium any longer.

[21439:21439:0818/092005.587570:FATAL:double_fork_and_exec.cc(131)] execv /usr/lib/chromium/crashpad_handler: No such file or directory (2)

The binary doesn't seem to exist for me.
Any ideas?

kraj, 2021-08-18:
It is built but not installed perhaps you can easily cook a patch to add an install command in do_install

MastaG, 2021-08-19:
Yes, I'll push a PR soon.

barndoor2276, 2021-08-19:
I was running into this issue this week also. I installed the binary via a do_install_append in a bbappend.
```
do_install_append() {
	install -m 0755 crashpad_handler ${D}${libdir}/chromium/
}
```
That worked but chromium still won't run and a new error pops up. Not sure if the error is related to crashpad or if it's just another error that was being masked.

`[0819/155701.539273:FATAL:semaphore_posix.cc(99)] Check failed: . sem_timedwait: Invalid argument (22)`

kraj, 2021-08-19:
@barndoor2276 which architecture/machine was targeted

barndoor2276, 2021-08-19:
@kraj 
arm
imx6q-dms-ba16

MastaG, 2021-08-19:
I'm on hardknott and it's working fine for me on armv7 (hard float).
Perhaps the crashpad_handler isn't built for each target (because not everyone is complaining), so I think we should guard it with some if-statement.
"""
if [ -e crashpad_handler ]
then
    install -m 0755 crashpad_handler ${D}${libdir}/chromium/
""""

kraj, 2021-08-19:
> I'm on hardknott and it's working fine for me on armv7 (hard float).
> Perhaps the crashpad_handler isn't built for each target (because not everyone is complaining), so I think we should guard it [...]

I think newer chromiums should all build it but its fine to see if the binary is existing before packaging it just to make robust.

kraj, 2021-08-19:
> I was running into this issue this week also. I installed the binary via a do_install_append in a bbappend.
>
> ```
> do_install_append() {
> 	install -m 0755 crashpad_handler ${D}${libdir}/chromium/
> } [...]

the fact that its working on hardknott means we have a bug in some dependent package's older version which is being used in dunfell.

barndoor2276, 2021-08-19:
@kraj Should I put my error in a separate issue? Is this a meta-browser defect or a chromium defect?

kraj, 2021-08-19:
> @kraj Should I put my error in a separate issue? Is this a meta-browser defect or a chromium defect?

Only if I knew :) but it is seen in chromium so open a separate issue here and see if it can be delved into for root causing it

zboszor, 2021-08-31:
I can reproduce it on Hardknott with chromium-x11 92.0.4515.131
```
Aug 31 08:23:59 dtd launch_browser[977]: [979:979:0831/082359.479482:FATAL:double_fork_and_exec.cc(131)] execv /usr/lib/chromium/crashpad_handler: No such file or directory (2)
```

rakuco, 2021-09-02:
I think we're hitting https://bugs.chromium.org/p/chromium/issues/detail?id=1176772. #552 should fix it.

closed 2021-09-02 by rakuco