OSSystems/meta-browser #543: why Convert to new override syntax?

jwzl, 2021-08-19:
I want update chromium to M92 in zeus, but the BB file convert to new override syntax? is any way to recovery  back the old override syntax ? thanks!

Best Regards
Qing

shr-project, 2021-08-19:
You can backport the bitbake change listed in that conversion PR to 1.44 branch

jwzl, 2021-08-23:
the M92 need which version of meta-clang  in yocto zeus?
thanks !

jwzl, 2021-08-23:
@otavio @shr-project

msisov, 2021-08-23:
@jwzl we no longer test Chromium with Zeus. See https://github.com/OSSystems/meta-browser/pull/531. It's EOL.

jwzl, 2021-08-23:
I wan't to which version of clang matched to build the M92. thanks!

msisov, 2021-08-23:
Emm. I'm not sure I understand you correctly. 

Let's take the #531 as an example. It states that the recipe was tested with

* chromium-x11
  * dunfell, clang, MACHINE="intel-core2-32"
  * dunfell, clang, MACHINE="intel-corei7-64"
  * dunfell, clang, MACHINE="raspberrypi3"
  * gatesgarth, MACHINE="raspberrypi3"

* chromium-ozone-wayland:
  * dunfell, clang, MACHINE="intel-corei7-64"
  * dunfell, clang, MACHINE="qemuarm64"
  * gatesgarth, clang, MACHINE="raspberrypi3"
  * dunfell, clang, MACHINE="m3ulcb"
  * hardknott, clang, MACHINE="raspebrrypi3"

Which means that dunfell yocto release was tested with meta-clang (dunfell branch), gatesgarth with meta-clang (gatesgarth) and hardknott with meta-clang (hardknott branch).

msisov, 2021-08-23:
Based on https://wiki.yoctoproject.org/wiki/Releases, we should drop gatesgarth and add Honister and Kirkstone.

shr-project, 2021-08-23:
> Based on https://wiki.yoctoproject.org/wiki/Releases, we should drop gatesgarth and add Honister and Kirkstone.

Only Honister, which is currently being developed in master branches. Kirkstone is future release which didn't even start to be developed.

msisov, 2021-08-23:
> > Based on https://wiki.yoctoproject.org/wiki/Releases, we should drop gatesgarth and add Honister and Kirkstone.
> 
> Only Honister, which is currently being developed in master branches. Kirkstone is future release which didn't even start to be developed.

Good to know. Thank you.

jwzl, 2021-08-25:
Emm, but I meet the error:
ERROR: Nothing PROVIDES 'clang-native-x86_64' (but virtual:native:/opt/work/yocto_zeus_2.3.3/sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb DEPENDS on or otherwise requires it). Close matches:
  clang-native
  slang-native

how to solve it ? thanks! @msisov @shr-project

ensc, 2021-09-03:
theoretically,  it would be possible to have one branch for both override styles.  You can write

```bitbake
_BITBAKE_OVSTYLE := "\${@['underline', 'colon'][bb.utils.vercmp_string('1.51.0', bb.__version__) <= 0]}"
...
require recipe_${_BITBAKE_OVSTYLE}.conf
```

But this becomes either very messy and/or causes lot of redundancy.

imo, it would be better and cleaner to add an `hardknott` branch (this is the last one with underline style(.  This branch does not need to exist forever; with EOL of dunfell (now + 7 months), it can disappear.

shr-project, 2021-09-03:
@ensc why not use latest revision from bitbake 1.50 as people should?

Or latest revision from hardknott branch in poky which contains necessary bitbake changes as well.

Whole point of these bitbake changes https://git.openembedded.org/bitbake/commit/?h=1.50&id=f8d1bc200460fc1cb5cbf7a7d1986fd86424b22d https://git.openembedded.org/bitbake/commit/?h=1.50&id=9b2d96b27f550da0fa68ba9ea96be98eb3a832a6 was to allow using new syntax since dunfell without the mess you're suggesting :).

ensc, 2021-09-03:
> @ensc why not use latest revision from bitbake 1.50 as people should?

Because `honister` (the `master` branch)  will use 1.51+ and the underline syntax is rejected there.

imo, whole transition happened too fast.  A bitbake accepting `:` should have been available for some time (6-9 months) before the changes to layers/OE happened.

But now, I do not see another way than keeping branches for new and old synta

shr-project, 2021-09-03:
> Because honister (the master branch) will use 1.51+ and the underline syntax is rejected there.

No, the same new syntax can be used everywhere from dunfell to honister. In dunfell-hardknott it's optional and only requires recent bitbake revision from corresponding branch (all 3 1.46, 1.48, 1.50 branches have the changes for compatibility with new syntax).

Newer honister builds not only support but also _require_ new syntax to be used.

shr-project, 2021-09-03:
If it's still not clear, then bitbake is accepting ':' in
dunfell with 1.46:
https://git.openembedded.org/bitbake/commit/?h=1.46&id=a6d5fb7554e3cf071e453db56a1e7469ac44277c
https://git.openembedded.org/bitbake/commit/?h=1.46&id=c5418eae56cc50dbae7951c055434a0c456c53a4

gatesgarth with 1.48:
https://git.openembedded.org/bitbake/commit/?h=1.48&id=aa9f7b80cfdb1119050af469a07ebd949829026c
https://git.openembedded.org/bitbake/commit/?h=1.48&id=14ae61205111383d5f609519c02476925184f6d1

hardknott with 1.50:
https://git.openembedded.org/bitbake/commit/?h=1.50&id=f8d1bc200460fc1cb5cbf7a7d1986fd86424b22d
https://git.openembedded.org/bitbake/commit/?h=1.50&id=9b2d96b27f550da0fa68ba9ea96be98eb3a832a6

honister with 1.51+:
https://git.openembedded.org/bitbake/commit/?id=0dbbb4547cb2570d2ce607e9a53459df3c0ac284
https://git.openembedded.org/bitbake/commit/?id=019251649a38754d5877759d13b664e28dea77de

If your builds don't accept new syntax, then you just need to update bitbake a bit.

The only exception are people still using unsupported Zeus release where they need to cherry-pick the 2 above commits to 1.44 branch on their own build (or ask RP do make and exception and merge it to 1.44 even after Zeus being EOL).