OSSystems/meta-browser #544: Builds currently broken on Poky from meta-browser/master

TonyTascioglu, 2021-08-19:
Hello,
When we try building chromium-x11 on Poky/master, we are greeted by the following build errors:

```
| python3 "../../build/toolchain/gcc_link_wrapper.py" --output="yocto_native/cddl" -- clang++ -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,--as-needed -Wl,t
| ld.lld: error: undefined symbol: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)                                                                          
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(ToUnderscoreId(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))                                    
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(ToCamelCase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))                                       
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(CppTypeToString[abi:cxx11](CppType const&))                                                                                                
| >>> referenced 80 more times                                                                                                                                                                                                        
|                                                                                                                                                                                                                                     
| ld.lld: error: undefined symbol: std::__throw_logic_error(char const*)                                                                                                                                                              
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(ToUnderscoreId(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))                                    
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(ToCamelCase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))                                       
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(WriteEnumEqualityOperatorSwitchCases(int, CppType const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >,)
| >>> referenced 26 more times                                                                                                                                                                                                        
|                                                                                                                                                                                                                                     
| ld.lld: error: undefined symbol: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::resize(unsigned long, char)                                                                                       
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(ToCamelCase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))                                       
|                                                                                                                                                                                                                                     
| ld.lld: error: undefined symbol: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::reserve(unsigned long)                                                                                            
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(CppTypeToString[abi:cxx11](CppType const&))                                                                                                
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(CppTypeToString[abi:cxx11](CppType const&))                                                                                                
| >>> referenced by main.cc                                                                                                                                                                                                           
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/main.o:(void Logger::WriteError<>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))                            
| >>> referenced 5 more times                                                                                                                                                                                                         
|                                                                                                                                                                                                                                     
| ld.lld: error: undefined symbol: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long)                                                                             
| >>> referenced by codegen.cc                                                                                                                                                                                                        
| >>>               yocto_native/obj/third_party/openscreen/src/tools/cddl/cddl/codegen.o:(CppTypeToString[abi:cxx11](CppType const&))                                                                                                
| >>> referenced by codegen.cc                                                                                                                                                                                                        
[... cut ...]
Has anyone else experienced a build error like this or have a solution? I also see @msisov has a new uprev underway so maybe it'll be fixed with the new version?

On the other hand, builds are also broken on Poky/hardknott since the recipe has migrated to the new syntax, which isn't backwards compatible on hardknott from my understanding. Are we going to make a separate master and hardknott branches to account for the change?

Thanks!

Tagging @rmacleod to the issue since he also found this issue.

TonyTascioglu, 2021-08-19:
It looks like the builds on master might be addressed by https://github.com/OSSystems/meta-browser/pull/542, I'm not sure if we're experiencing the same error, but it might be related.

schnitzeltony, 2021-08-19:
Your error looks familiar: #542 first commit should fix (and after that I ran into another issue which is solved by second commit)

rakuco, 2021-09-09:
@TonyTascioglu are you still having build issues? We've merged several PRs lately and I'm hoping everything is fine now.

rwmacleod, 2022-01-04:
As Khem said, this has been fixed for a while now. @TonyTascioglu  Please close the defect.

closed 2022-01-05 by rakuco