OSSystems/meta-browser #545: Chromium fails to load properly - sem_timedwait: Invalid argument

barndoor2276, 2021-08-19:
After installing missing 'crashpad_handler' binary (related to #541) starting with chromium 91 I am seeing the following error on chromium startup. Chromium starts and displays an all white screen but never finishes loading. After 10 minutes or so it crashes with a segmentation fault.

[0819/155701.539273:FATAL:semaphore_posix.cc(99)] Check failed: . sem_timedwait: Invalid argument (22)

Using poky/OE @ dunfell branch.
Arch: Arm
Machine: imx6q-dms-ba16

barndoor2276, 2021-08-19:
Other users in the thread for #541 state that on hardknott branch of poky/OE there was no issue.

MastaG, 2021-08-20:
Did you try running it witth --no-sandbox?

barndoor2276, 2021-08-20:
> 
> 
> Did you try running it witth --no-sandbox?

Yes I tried that while troubleshooting. I also ran chromium as a non root user and got the same results.

havok13888, 2021-09-14:
same issue, with gatesgarth

barndoor2276, 2022-02-22:
I fixed this by using clang 12 in the `dunfell-clang12` branch of meta-clang. See comments related here https://github.com/OSSystems/meta-browser/pull/602#issuecomment-1039211619. Of course I'm having new EGL issues now that I've successfully compiled chromium 97 but this issue can be closed.

closed 2022-02-22 by barndoor2276