OSSystems/meta-browser #549: how to disable chromium update notice to popping up?

gitgonewithwind, 2021-08-30:
Hi all,

I use imx8 maaxboad,iwave board. we use chromium  -kiosk  webpage  as user GUI(of course ,there is no menu,no settings ...(3 points ),no webpage bar....

but the  chromium  keep popping up update  notice to dispplay. how to disable it  when starting  up chromium with options or in setting config files in somewhere? (of course , can not use menu to change it manually).

I searched online, but do not find solutions  up to now.

 

any help?

 

Thank you very much.

r1mikey, 2021-08-30:
The following are the relevant flags we set.  Of these, the `--simulate-outdated-no-au` argument is the one you need for this issue, but we specify a bunch more to suppress behaviour that assumes a user present or is otherwise desirable in a kiosk.

```
exec /usr/bin/chromium \
  --kiosk --enable-fullscreen \
  --touch-events=enabled \
  --activate-on-launch \
  --noerrdialogs \
  --remote-debugging-port=9222 \
  --disable-pinch \
  --disable-features=OverscrollHistoryNavigation \
  --start-maximized \
  --disable-infobars \
  --disable-notifications \
  --disable-session-crashed-bubble \
  --no-first-run \
  --enable-offline-auto-reload \
  --autoplay-policy=no-user-gesture-required \
  --deny-permission-prompts \
  --disable-search-geolocation-disclosure \
  --enable-ipv6 $CHROMIUM_EXTRAS \
  --simulate-outdated-no-au='Tue, 31 Dec 2099 23:59:59 GMT' \
  --disable-component-update \
  --allow-browser-signin=false \
  "${KIOSK_URI}"
```

gitgonewithwind, 2021-08-30:
@r1mikey 
Thank you
I am going to try it (I just try --simulate-outdated-no-au='Tue, 31 Dec 2099 23:59:59 GMT'）

r1mikey, 2021-09-08:
@gitgonewithwind Did that work for you?

gitgonewithwind, 2021-09-08:
> @gitgonewithwind Did that work for you?

@r1mikey 
works!