OSSystems/meta-browser #558: chromium-ozone-wayland let weston crash in imx8mm

jwzl, 2021-09-08:
I used the M92.
and Run 2 WebGL aquarium demo, wait for a few seconds, all screen are black,  maybe weston crash!
I used  "/usr/bin/chromium --no-sandbox --disable-gpu-compositing  --test-type --enable-logging=stderr" to debug and no any other unexpected message!

restart chromium with commadline and is still black screen.
When I restart the weston, and restart the chromium, everything is okay!
thanks

jwzl, 2021-09-08:
It's seems that M91 no such issue!
but  M91 will exit unexpected at some times!

jwzl, 2021-09-08:
chromium is 92.0.4515.107 
weston is 9.0.0

jwzl, 2021-09-13:
@otavio @rakuco  thanks！

jwzl, 2021-09-16:
@msisov thanks！

msisov, 2021-09-16:
no idea. maybe imx8 specific. try the latest recipe that will be released tomorrow.

msisov, 2021-09-17:
Bot didn’t complete testing builds. Please wait until Monday