OSSystems/meta-browser #560: build issue

havok13888, 2021-09-09:
`ParseError at /home/neil/code/controltech/ct_yocto_pi/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_92.0.4515.159.bb:41: unparsed line: 'CHROMIUM_EXTRA_ARGS:append = " --ozone-platform=wayland"'`

" CHROMIUM_EXTRA_ARGS:append " is this new syntax? I couldn't find any references to it in the yocto manual. Or am I missing some includes. I am just trying to build the x11 version, not wayland.

rakuco, 2021-09-09:
It's _somewhat_ new syntax that's going to be mandatory in future releases, and since the other branches we support also allow it, we ended up landing the change. There's a bigger discussion about this in #543, with https://github.com/OSSystems/meta-browser/issues/543#issuecomment-912605725 providing a longer list of what branches support the syntax.

havok13888, 2021-09-09:
Updated to latest poky and I believe it's moved past that problem. Seeing something new now.

`ERROR: Nothing PROVIDES 'virtual/libc-native' (but virtual:native:meta-clang/recipes-devtools/clang/compiler-rt_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/libgl-native
  virtual/librpc-native
  virtual/libx11-native`
  
I am on gatesgarth for everything except meta-browser which is on master. I had built all of this a few months ago and worked fine then. So I'm assuming I'm missing something with new changes.

havok13888, 2021-09-10:
For now I have reverted to a commit prior to the overrides change to get everything work. I wonder if this works in Dunfell. Gatesgarth should be similar enough to Dunfell

shr-project, 2021-09-10:
Just to be clear, it's not because of overrides changes, but the commit just before them:
```
mjansa@kangaroo-build:~/mjansa/build/nodistro/gatesgarth/meta-browser$ git checkout 967f28f
Previous HEAD position was a5a5f27 chromium: update to 90.0.4430.93
HEAD is now at 967f28f gn-runtime: Fix build when using llvm runtime
```
fails
```
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'virtual/compilerlibs-native' (but virtual:native:/jenkins/mjansa/build/nodistro/gatesgarth/meta-clang/recipes-devtools/clang/compiler-rt_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/libsdl-native
  virtual/libusb0-native
  virtual/mesa-native
NOTE: Runtime target 'chromium-x11-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['chromium-x11-dev', 'gn-native', 'compiler-rt-native', 'virtual/compilerlibs-native']
NOTE: Runtime target 'chromium-x11' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['chromium-x11', 'gn-native', 'compiler-rt-native', 'virtual/compilerlibs-native']
ERROR: Nothing PROVIDES 'clang-native-x86_64' (but virtual:native:/jenkins/mjansa/build/nodistro/gatesgarth/meta-clang/recipes-devtools/clang/compiler-rt_git.bb DEPENDS on or otherwise requires it). Close matches:
  clang-native
  slang-native
ERROR: Nothing PROVIDES 'virtual/libc-native' (but virtual:native:/jenkins/mjansa/build/nodistro/gatesgarth/meta-clang/recipes-devtools/clang/compiler-rt_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/libgl-native
  virtual/librpc-native
  virtual/libx11-native
```

If you revert 967f28f from current master, then it still builds (or at least starts building - haven't finished yet).

So we would need few meta-clang changes to be backported from hardnott to gatesgarth.

shr-project, 2021-09-10:
@kraj already merged my PR for meta-clang/gatesgarth

@havok13888 please retest latest meta-clang/gatesgarth in your setup and hurry with upgrading from gatesgarth to hardknott or honister in case gatesgarth support is removed completely as suggested in: 
 https://github.com/OSSystems/meta-browser/issues/559

havok13888, 2021-09-14:
It's building now but I run into this issue #545 and I have a blank white screen. This is on a RPI4