OSSystems/meta-browser #562: Blank white screen RPI 4

havok13888, 2021-09-14:
All recipes on Gatesgarth, meta-browser on master.

chromium-x11_%.bbappend contains
`PACKAGECONFIG = "proprietary-codecs"`

Launching chromium with
export DISPLAY=:0
chromium --no-sandbox 'www.google.com'

All I get a a blank white screen

Startup logs:
[Chromium-X11-Start.txt](https://github.com/OSSystems/meta-browser/files/7162572/Chromium-X11-Start.txt)

Image:
[image]

barndoor2276, 2021-09-14:
Might be related to #545

havok13888, 2021-09-14:
This happened on a previous commit too. Maybe just something with my setup. It had a similar log with out 
`[0914/095916.502119:FATAL:semaphore_posix.cc(99)] Check failed: . sem_timedwait: Invalid argument (22)`

https://github.com/OSSystems/meta-browser/commit/967f28f0404cef3b74fb9d0d3c897b65b15adcdf

kraj, 2021-09-14:
does it work if you do not add `PACKAGECONFIG = "proprietary-codecs"`

havok13888, 2021-09-14:
Package config removed log: 
[NoPackageConfig.txt](https://github.com/OSSystems/meta-browser/files/7164774/NoPackageConfig.txt)

Same effect, blank white screen but new errors

I tried the same by remove --use-gl=egl from /usr/bin/chromium. The only thing that happens is the libGLESv2 errors are gone from the log
No EGL log
[NoEGL.txt](https://github.com/OSSystems/meta-browser/files/7164792/NoEGL.txt)

havok13888, 2021-09-14:
After a few more launched I'm seeing this 
[image]

It says

```
Your preferences are corrupt or invalid
Chromium is unable to recover you settings
```

havok13888, 2021-09-15:
I went all the way back to a5a5f277740a8434377f9844c62574efb17a4fa0 and it's working fine now

barndoor2276, 2021-09-15:
> 
> 
> I went all the way back to [a5a5f27](https://github.com/OSSystems/meta-browser/commit/a5a5f277740a8434377f9844c62574efb17a4fa0) (chromium: update to 90.0.4430.93) and it's working fine. I might try to see which commit actually breaks it.
> 
> startup log
> [Chromium90.txt](https://github.com/OSSystems/meta-browser/files/7170496/Chromium90.txt)

I've had to lock my build to same commit also.