OSSystems/meta-browser #565: chromium-ozone-wayland: build fails after updating layer to 39eef30 chromium: update to 93.0.4577.82

DWagenk, 2021-09-21:
On a yocto dunfell based build with a custom machine (imx6 based) I get this error after updating meta-browser to 39eef30 :
```
FAILED: obj/third_party/swiftshader/src/Vulkan/swiftshader_libvulkan/VkDescriptorPool.o 
arm-poky-linux-gnueabi-clang++ -target arm-poky-linux-gnueabi  -march=armv7-a  -mfpu=neon -mfloat-abi=hard -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/build/tmp/work/armv7at2hf-neon-imx-poky-linux-gnueabi/chromium-ozone-wayland/93.0.4577.82-r0/recipe-sysroot -MMD -MF obj/third_party/swiftshader/src/Vulkan/swiftshader_libvulkan/VkDescriptorPool.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DANGLE_DISABLE_TRACE -DNO_SANITIZE_FUNCTION=__attribute__\(\(no_sanitize\(\"function\"\)\)\) -DVK_USE_PLATFORM_XLIB_KHR -DVK_USE_PLATFORM_XCB_KHR -DVK_EXPORT=__attribute__\(\(visibility\(\"default\"\)\)\) -DSWIFTSHADER_ENABLE_ASTC -I../../third_party/swiftshader/src -I../../third_party/swiftshader/include -I../../third_party/vulkan-deps/spirv-tools/src/include -I../../third_party/vulkan-deps/spirv-headers/src/include -I../.. -Igen -I../../third_party/swiftshader/third_party/marl/include -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -Xclang -fdebug-compilation-dir -Xclang . -mfpu=neon -mthumb -O2 -fdata-sections -ffunction-sections -fno-unique-section-names -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wall -Wno-unused-variable -Wno-c++11-narrowing -Wno-misleading-indentation -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Os -Wno-unused-private-field -Wno-switch -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -std=c++17  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=/build/tmp/work/armv7at2hf-neon-imx-poky-linux-gnueabi/chromium-ozone-wayland/93.0.4577.82-r0=/usr/src/debug/chromium-ozone-wayland/93.0.4577.82-r0                      -fdebug-prefix-map=/build/tmp/work/armv7at2hf-neon-imx-poky-linux-gnueabi/chromium-ozone-wayland/93.0.4577.82-r0=/usr/src/debug/chromium-ozone-wayland/93.0.4577.82-r0                      -fdebug-prefix-map=/build/tmp/work/armv7at2hf-neon-imx-poky-linux-gnueabi/chromium-ozone-wayland/93.0.4577.82-r0/recipe-sysroot=                      -fdebug-prefix-map=/build/tmp/work/armv7at2hf-neon-imx-poky-linux-gnueabi/chromium-ozone-wayland/93.0.4577.82-r0/recipe-sysroot-native= -g1 -stdlib=libc++ -fvisibility-inlines-hidden -c ../../third_party/swiftshader/src/Vulkan/VkDescriptorPool.cpp -o obj/third_party/swiftshader/src/Vulkan/swiftshader_libvulkan/VkDescriptorPool.o
In file included from ../../third_party/swiftshader/src/Vulkan/VkDescriptorPool.cpp:15:
In file included from ../../third_party/swiftshader/src/Vulkan/VkDescriptorPool.hpp:18:
In file included from ../../third_party/swiftshader/src/Vulkan/VkObject.hpp:23:
In file included from ../../third_party/swiftshader/include/vulkan/vk_icd.h:26:
../../third_party/swiftshader/include/vulkan/vulkan.h:54:10: fatal error: 'xcb/xcb.h' file not found
#include <xcb/xcb.h>
         ^~~~~~~~~~~
1 error generated.
```

The same build with prior versions (tested with ee752be) work fine in the same build environment

The other layers are all at the current tip of the dunfell branch. The only custom modification I have in my layers is:
```
PACKAGECONFIG_append = " kiosk-mode"
```

```
bitbake chromium-ozone-wayland -e | grep -e ^DISTRO_FEATURES= -e ^TOOLCHAIN= -e ^CHROMIUM
CHROMIUM_EXTRA_ARGS="         --use-gl=egl         --kiosk --no-first-run --incognito  --ozone-platform=wayland"
DISTRO_FEATURES="acl alsa argp bluetooth ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci    vfat largefile opengl ptest multiarch wayland vulkan wayland pam pulseaudio sysvinit gobj
ect-introspection-data ldconfig"
TOOLCHAIN="clang"
```

Any ideas on what could cause this problem?

msisov, 2021-09-21:
vulkan includes xcb (x11 headers). it looks like my build bot  includes those by default and I didn't see the issue

DWagenk, 2021-09-22:
So adding `libxcb` to chromiums `DEPENDS` should fix this?
Is this correct even though the system uses wayland exclusively and X11 is not / should not be present in the image?

msisov, 2021-09-22:
no, it's not correct. I will fix that.