OSSystems/meta-browser #567: chromium-x11 93.0.4577.82 build fails

zboszor, 2021-09-23:
I got this error:
```
| In file included from ../../v8/src/heap/cppgc/allocation.cc:5:
| ../../v8/include/cppgc/allocation.h:168:39: error: no member named 'forward' in namespace 'std'
|     T* object = ::new (memory) T(std::forward<Args>(args)...);
|                                  ~~~~~^
| ../../v8/include/cppgc/allocation.h:168:47: error: 'Args' does not refer to a value
|     T* object = ::new (memory) T(std::forward<Args>(args)...);
|                                               ^
| ../../v8/include/cppgc/allocation.h:164:25: note: declared here
|   template <typename... Args>
|                         ^
| ../../v8/include/cppgc/allocation.h:178:39: error: no member named 'forward' in namespace 'std'
|     T* object = ::new (memory) T(std::forward<Args>(args)...);
|                                  ~~~~~^
| ../../v8/include/cppgc/allocation.h:178:47: error: 'Args' does not refer to a value
|     T* object = ::new (memory) T(std::forward<Args>(args)...);
|                                               ^
| ../../v8/include/cppgc/allocation.h:173:25: note: declared here
|   template <typename... Args>
|                         ^
| ../../v8/include/cppgc/allocation.h:206:55: error: no member named 'forward' in namespace 'std'
|       MakeGarbageCollectedTrait<T>::Call(handle, std::forward<Args>(args)...);
|                                                  ~~~~~^
| ../../v8/include/cppgc/allocation.h:206:63: error: 'Args' does not refer to a value
|       MakeGarbageCollectedTrait<T>::Call(handle, std::forward<Args>(args)...);
|                                                               ^
| ../../v8/include/cppgc/allocation.h:203:35: note: declared here
| template <typename T, typename... Args>
|                                   ^
| ../../v8/include/cppgc/allocation.h:225:55: error: no member named 'forward' in namespace 'std'
|                                                  std::forward<Args>(args)...);
|                                                  ~~~~~^
| ../../v8/include/cppgc/allocation.h:225:63: error: 'Args' does not refer to a value
|                                                  std::forward<Args>(args)...);
|                                                               ^
| ../../v8/include/cppgc/allocation.h:221:35: note: declared here
| template <typename T, typename... Args>
|                                   ^
| In file included from ../../v8/src/heap/cppgc/allocation.cc:10:
| In file included from ../../v8/src/heap/cppgc/object-allocator.h:13:
| In file included from ../../v8/src/heap/cppgc/heap-page.h:12:
| In file included from ../../v8/src/heap/cppgc/object-start-bitmap.h:15:
| ../../v8/src/base/bits.h:197:34: warning: implicit conversion loses integer precision: 'size_t' (aka 'unsigned long') to 'uint32_t' (aka 'unsigned int') [-Wshorten-64-to-32]
|     return RoundUpToPowerOfTwo32(value);
|            ~~~~~~~~~~~~~~~~~~~~~ ^~~~~
| 1 warning and 8 errors generated.
```

rakuco, 2021-09-23:
@zboszor can you paste the compiler invocation that led to this error?

Everything worked fine in our test builds, but this error was fixed upstream and @kraj backported the fix to musl in #566, we might have to make the patch apply for all libc implementations.

ensc, 2021-09-23:
me too with hardknott base

```
| aarch64-fsl-linux-clang++ -target aarch64-fsl-linux  -mcpu=cortex-a53 -march=armv8-a+crc+crypto -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=..../chromium-ozone-wayland/93.0.4577.82-r0/recipe-sysroot -MMD -MF obj/v8/cppgc_base/explicit-management.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -DOFFICIAL_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -DNO_UNWIND_TABLES -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64 -DENABLE_MINOR_MC -DV8_INTL_SUPPORT -DV8_USE_EXTERNAL_STARTUP_DATA -DV8_ATOMIC_OBJECT_FIELD_WRITES -DV8_ATOMIC_MARKING_STATE -DV8_ENABLE_LAZY_SOURCE_POSITIONS -DV8_SHARED_RO_HEAP -DV8_WIN64_UNWINDING_INFO -DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH -DV8_SNAPSHOT_COMPRESSION -DV8_SHORT_BUILTIN_CALLS -DV8_ENABLE_WEBASSEMBLY -DV8_ALLOCATION_FOLDING -DV8_ALLOCATION_SITE_TRACKING -DV8_COMPRESS_POINTERS -DV8_COMPRESS_POINTERS_IN_SHARED_CAGE -DV8_31BIT_SMIS_ON_64BIT_ARCH -DV8_DEPRECATION_WARNINGS -DCPPGC_CAGED_HEAP -DV8_TARGET_ARCH_ARM64 -DV8_HAVE_TARGET_OS -DV8_TARGET_OS_LINUX -DV8_RUNTIME_CALL_STATS -DDISABLE_UNTRUSTED_CODE_MITIGATIONS -I../.. -Igen -I../../v8 -I../../v8/include -Igen/v8 -Igen/v8/include -fno-delete-null-pointer-checks -fno-ident -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -mllvm -instcombine-lower-dbg-declare=0 -flto=thin -fsplit-lto-unit -fwhole-program-vtables -Xclang -fdebug-compilation-dir -Xclang . -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wenum-compare-conditional -Wno-unknown-warning-option -Wno-psabi -Wno-ignored-pragma-optimize -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -fno-omit-frame-pointer -g0 -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wmissing-field-initializers -Wunreachable-code -Wshorten-64-to-32 -O3 -fdata-sections -ffunction-sections -fno-unique-section-names -Wexit-time-destructors -std=c++14 -fno-trigraphs -Wno-trigraphs -fno-exceptions -fno-rtti -fvisibility-inlines-hidden  -O2 -pipe  -feliminate-unused-debug-types -fmacro-prefix-map=..../chromium-ozone-wayland/93.0.4577.82-r0=/usr/src/debug/chromium-ozone-wayland/93.0.4577.82-r0                      -fdebug-prefix-map=..../chromium-ozone-wayland/93.0.4577.82-r0=/usr/src/debug/chromium-ozone-wayland/93.0.4577.82-r0                      -fdebug-prefix-map=..../chromium-ozone-wayland/93.0.4577.82-r0/recipe-sysroot=                      -fdebug-prefix-map=..../chromium-ozone-wayland/93.0.4577.82-r0/recipe-sysroot-native= -g1 -fvisibility-inlines-hidden -stdlib=libc++ -c ../../v8/src/heap/cppgc/explicit-management.cc -o obj/v8/cppgc_base/explicit-management.o
```

```
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "fedora-34"
TARGET_SYS           = "aarch64-fsl-linux"

meta-clang           = "HEAD:8a7c0e808678a95dfdba6689b651f1a0ff0dc206"
meta-poky            = "HEAD:0f0682a1fe234d21ebd1e99838d4529b1a33b57b"
meta-freescale       = "HEAD:2b9a987af83ea1c6d1d1783d3bb7dda7b71c5645"
meta-freescale-3rdparty = "HEAD:f8150f3b37cb83cba1f9e2378e57bb63e02d4610"
meta-freescale-distro = "HEAD:a260c63ccb1aedc99b4858957ea539cd74c24797"
<meta-imx>              = "HEAD:426311a3f4e6936125c0599a34d8f6133dd2460d"
```

zboszor, 2021-09-23:
> @zboszor can you paste the compiler invocation that led to this error?
> 
> Everything worked fine in our test builds, but this error was fixed upstream and @kraj backported the fix to musl in #566, we might have to make the patch apply for all libc implementations.

Updated the original comment with the failing `x86_64-oe-linux-clang++` command.

rakuco, 2021-09-23:
> Everything worked fine in our test builds, but this error was fixed upstream and @kraj backported the fix to musl in #566, we might have to make the patch apply for all libc implementations.

Actually, the backport was added to all platforms. Can you update to 64c918e5a954527440393f0d3426fd29419e3e56 and see if it works?

rwmacleod, 2021-09-23:
This is tracked and fixed by: https://bugs.chromium.org/p/chromium/issues/detail?id=1237252&q=1237252&can=2
I have backported the trivial v8 fix: https://github.com/v8/v8/commit/69426180fac5d37643a28887b5e148f844e14ce6
and could submit a patch. It builds for meta-browser/master with other layers on master as well as hardknott for glibc. 
I guess I could test a musl build next if people are interested. I'm not sure why @kraj  's patch should be considered over the upstream solution, Khem?

rakuco, 2021-09-23:
My bad, I meant commit 983654305362e811d99b20720128cdc2ac10a4ee in meta-browser. That's a backport from the V8 commit.

kraj, 2021-09-23:
> This is tracked and fixed by: https://bugs.chromium.org/p/chromium/issues/detail?id=1237252&q=1237252&can=2
> I have backported the trivial v8 fix: [v8/v8@6942618](https://github.com/v8/v8/commit/69426180fac5d37643a28887b5e148f844e14ce6)
> and could submit a patch. It builds for meta-browser/master with other layers on master as well as hardknott for glibc.
> I guess I could test a musl build next if people are interested. I'm not sure why @kraj 's patch should be considered over the upstream solution, Khem?

Please look/use HEAD of meta-browser as @rakuco is suggesting and report back if it still fails.

rwmacleod, 2021-09-24:
master/HEAD builds and runs fine for qemux86. Thanks Khem.

rakuco, 2021-09-24:
Thanks all. Please reopen if this is still an issue.

closed 2021-09-24 by rakuco