OSSystems/meta-browser #569: chromium: Consider using bundled FreeType copy

rakuco, 2021-09-28:
Chromium project members contribute to FreeType itself, and are always actively tracking the latest upstream versions (include pre-release ones), depending on features included there and staying up-to-date with security issues.

This makes it difficult for us to get Chromium to build on multiple branches, including old ones like dunfell, where the FreeType version might lag behind what Chromium expects by quite a bit.

So far this means we've had commits like ac9d7200001434e692abad4052d5f34ba48f25e9 (with a patch that needs to be updated on every Chromium update), but according to #568 it no longer seems to be enough.

I'd like us to consider passing `use_system_freetype=false` to GN and using Chromium's own FreeType copy.

rakuco, 2021-09-28:
@kraj @msisov @shr-project @otavio

kraj, 2021-09-28:
> Chromium project members contribute to FreeType itself, and are always actively tracking the latest upstream versions (include pre-release ones), depending on features included there and staying up-to-date with security issues.
> 
> This makes it difficult for us to get Chromium to build on multiple branches, including old ones like dunfell, where the FreeType version might lag behind what Chromium expects by quite a bit.
> 
> So far this means we've had commits like [ac9d720](https://github.com/OSSystems/meta-browser/commit/ac9d7200001434e692abad4052d5f34ba48f25e9) (with a patch that needs to be updated on every Chromium update), but according to #568 it no longer seems to be enough.
> 
> I'd like us to consider passing `use_system_freetype=false` to GN and using Chromium's own FreeType copy.

I guess this is the approach we have to follow as chromium abstracts itself from platform more and more. Are there more liba that we should consider at this Time

rakuco, 2021-09-28:
> Are there more liba that we should consider at this Time

I don't think so, at least. `chromium-unbundle.inc` only unbundles flac, libjpeg and libxslt at the moment, and I don't see other patches for third-party libraries in `files/`.

kraj, 2021-09-28:
Ok looks good to me let’s start using bundled version

PR #570 (2021-09-29): chromium: use bundled freetype
* Fixes #569

* because the current skia codebase expects freetype newer than latest
  2.11 release from freetype upstream

* chromium-ozone-wayland as well as chromium-x11 with honister based builds:

  ../../third_party/skia/src/ports/SkFontHost_FreeType.cpp:1110:13: error: unknown type name 'FT_ClipBox'
              FT_ClipBox colrGlyphBbox;
              ^

  and

  ../../third_party/skia/src/ports/SkFontHost_FreeType_common.cpp:438:14: error: use of undeclared identifier 'FT_COLR_COMPOSITE_PLUS'
  ../../third_party/skia/src/ports/SkFontHost_FreeType_common. [... cut]
closed 2021-09-29 by rakuco