OSSystems/meta-browser #581: Chromium-x11: Build fails with latest dunfell

xemmits, 2021-10-31:
Hi,

I am trying to build chromium-x11 for TI am572x processor using poky with dunfell. I already had a working setup for zeus and now I am in the process of updating it. The error that I am encountering is somewhat as follows:

```
| [53265/53273] python3 "../../build/toolchain/gcc_link_wrapper.py" --output="./v8_context_snapshot_generator" -- arm-poky-linux-gnueabi-clang++ -target arm-poky-linux-gnueabi  -march=armv7-a  -mfpu=neon -mfloat-abi=hard -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/recipe-sysroot -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,--no-call-graph-profile-sort -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-z,defs -Wl,--as-needed -pie -Wl,--disable-new-dtags -Wl,--icf=none -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./v8_context_snapshot_generator" -Wl,--start-group @"./v8_context_snapshot_generator.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -ljpeg -lxslt -lxml2 -lresolv -lgio-2.0 -lexpat -luuid -lX11 -lXcomposite -lXdamage -lXext -lXfixes -lXrender -lXrandr -lXtst -lxcb -ldrm -lxkbcommon -lgbm -latk-1.0 -latk-bridge-2.0 -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lXi -lpci -ldbus-1 -latspi
| [53266/53273] python3 ../../build/gn_run_binary.py ./v8-qemu-wrapper.sh ./v8_context_snapshot_generator --output_file=v8_context_snapshot.bin
| FAILED: v8_context_snapshot.bin
| python3 ../../build/gn_run_binary.py ./v8-qemu-wrapper.sh ./v8_context_snapshot_generator --output_file=v8_context_snapshot.bin
| [1030/182628.092893:FATAL:cpu.cc(186)] Check failed: false.
| #0 0x0000fd685cf4 <unknown>
| #1 0x0000fd6166e8 <unknown>
| #2 0x0000fd61e0fc <unknown>
| #3 0x0000fd61e908 <unknown>
| #4 0x0000fd6141be <unknown>
| #5 0x0000fd61627c <unknown>
| #6 0x0000fd616578 <unknown>
| #7 0x0000fd6a9928 <unknown>
| #8 0x0000fd6a7fe6 <unknown>
| #9 0x0000fd68dac0 <unknown>
| 
| qemu: uncaught target signal 6 (Aborted) - core dumped
| ././v8-qemu-wrapper.sh: line 7:  5884 Aborted                 PSEUDO_UNLOAD=1 qemu-arm -r 3.2.0 -L /home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/recipe-sysroot -E LD_LIBRARY_PATH=/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/recipe-sysroot/usr/lib:/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/recipe-sysroot/lib "$@"
| ./v8-qemu-wrapper.sh failed with exit code 134
| ninja: build stopped: subcommand failed.
| WARNING: /home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/temp/run.do_compile.7843:1 exit 1 from 'ninja -v -j 64 chrome chrome_sandbox chromedriver.unstripped'
| ERROR: Execution of '/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/temp/run.do_compile.7843' failed with exit code 1
```
The relevant details for the build is as follows:

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "am57xx-phycore-gsi"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.11"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta-arm             
meta-arm-toolchain   = "HEAD:ce535dfb96de4d2529f091d7d85a7172c626001c"
meta-chromium        
meta-firefox         = "HEAD:550fb5ee686b8826c3cfa8a36c2e59c27b58f7ae"
meta-clang           = "HEAD:52c7b51ab9ebe426a77559802333d8479864912e"
meta-networking      
meta-oe              
meta-python          = "HEAD:814eec96c2a29172da57a425a3609f8b6fcc6afe"
meta-rust            = "HEAD:d4384db8c3f0aa01732627f92fc3d446359bf743"
meta-swupdate        = "HEAD:dfa09d7c9216894d89659e25fbf2de644c86292e"
meta-ti              = "HEAD:b2c515a5abd5a2270c54c0e0fec441c7c456c75f"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:3855395469fe7822bcac37f382735f4760f5a94e"
```

I have tried rolling back before meta-python2 dependency was removed and that has also shown similar problem. I would really appreciate if somebody can throw some light on the issue or at least point me to the last working commit for poky dunfell. Thanks in advance.

rakuco, 2021-11-01:
Hmm, I've just finished a dunfell build with MACHINE="qemuarm". Does removing those extra `GN_ARGS` help in any way? By the way, is there any particular reason for changing `fieldtrial_testing_like_official_build`, `is_official_build` and `is_chrome_branded`?

xemmits, 2021-11-01:
Thanks for the prompt reply. Some of the arguments are required but the ones you have mentioned may be removed in my opinion, but I have to confirm from a colleague. I tried builds without all the extra args and after removing just the aforementioned ones, leading me to similar linker errors. I also updated meta-clang from today.

```
| [13301/13301] python3 "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- arm-poky-linux-gnueabi-clang++ -target arm-poky-linux-gnueabi  -march=armv7-a  -mfpu=neon -mfloat-abi=hard -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -flto=thin -Wl,--thinlto-jobs=64 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -Wl,-mllvm,-import-instr-limit=5 -fwhole-program-vtables -march=armv7-a -Wl,--no-call-graph-profile-sort -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-z,defs -Wl,--as-needed -pie -Wl,--disable-new-dtags -Wl,--lto-O2 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -ldbus-1 -lresolv -ljpeg -lexpat -luuid -lxcb -ldrm -lxkbcommon -lX11 -lXcomposite -lXdamage -lXext -lXfixes -lXrender -lXrandr -lXtst -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lXi -lpci -latspi -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxshmfence -lxslt -lxml2
| FAILED: chrome
| python3 "../../build/toolchain/gcc_link_wrapper.py" --output="./chrome" -- arm-poky-linux-gnueabi-clang++ -target arm-poky-linux-gnueabi  -march=armv7-a  -mfpu=neon -mfloat-abi=hard -rtlib=compiler-rt --unwindlib=libgcc -stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/recipe-sysroot -Wl,--version-script=../../build/linux/chrome.map -fuse-ld=lld -Wl,--fatal-warnings -Wl,--build-id=sha1 -fPIC -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--icf=all -Wl,--color-diagnostics -Wl,-mllvm,-instcombine-lower-dbg-declare=0 -flto=thin -Wl,--thinlto-jobs=64 -Wl,--thinlto-cache-dir=thinlto-cache -Wl,--thinlto-cache-policy=cache_size=10\%:cache_size_bytes=40g:cache_size_files=100000 -Wl,-mllvm,-import-instr-limit=5 -fwhole-program-vtables -march=armv7-a -Wl,--no-call-graph-profile-sort -Wl,-O2 -Wl,--gc-sections -rdynamic -Wl,-z,defs -Wl,--as-needed -pie -Wl,--disable-new-dtags -Wl,--lto-O2 -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -o "./chrome" -Wl,--start-group @"./chrome.rsp"  -Wl,--end-group  -latomic -ldl -lpthread -lrt -lgmodule-2.0 -lgobject-2.0 -lgthread-2.0 -lglib-2.0 -lsmime3 -lnss3 -lsoftokn3 -lnssutil3 -lplds4 -lplc4 -lnspr4 -latk-1.0 -latk-bridge-2.0 -lgio-2.0 -ldbus-1 -lresolv -ljpeg -lexpat -luuid -lxcb -ldrm -lxkbcommon -lX11 -lXcomposite -lXdamage -lXext -lXfixes -lXrender -lXrandr -lXtst -lgbm -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -lcairo -lXi -lpci -latspi -lFLAC -lgtk-3 -lgdk-3 -lcairo-gobject -lgdk_pixbuf-2.0 -lxshmfence -lxslt -lxml2
| arm-poky-linux-gnueabi-ld.lld: error: undefined symbol: CastRemotingConnector::Get(content::WebContents*)
| >>> referenced by media_router_mojo_impl.cc:248 (../../chrome/browser/media/router/mojo/media_router_mojo_impl.cc:248)
| >>>               thinlto-cache/llvmcache-E4EB67EF3FDD544BF7A519D763C29955EA0B5BAB:(media_router::MediaRouterMojoImpl::CreateRoute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, url::Origin const&, content::WebContents*, base::OnceCallback<void (mojo::StructPtr<media_router::mojom::RoutePresentationConnection>, media_router::RouteRequestResult const&)>, base::TimeDelta, bool))
| >>> referenced by cast_mirroring_service_host.cc:437 (../../chrome/browser/media/cast_mirroring_service_host.cc:437)
| >>>               thinlto-cache/llvmcache-24409AADF14F949980750603BA17A544774A13E7:(mirroring::CastMirroringServiceHost::ConnectToRemotingSource(mojo::PendingRemote<media::mojom::Remoter>, mojo::PendingReceiver<media::mojom::RemotingSource>))
| 
| arm-poky-linux-gnueabi-ld.lld: error: undefined symbol: CastRemotingConnector::ResetRemotingPermission()
| >>> referenced by media_router_mojo_impl.cc:249 (../../chrome/browser/media/router/mojo/media_router_mojo_impl.cc:249)
| >>>               thinlto-cache/llvmcache-E4EB67EF3FDD544BF7A519D763C29955EA0B5BAB:(media_router::MediaRouterMojoImpl::CreateRoute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_trai [... cut ...] er.unstripped'
| ERROR: Execution of '/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/chromium-x11/94.0.4606.71-r0/temp/run.do_compile.17884' failed with exit code 1
ERROR: Task (/home/sappy/commoncontrol/yocto_update/meta-tr-gsi/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_94.0.4606.71.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7821 tasks of which 7744 didn't need to be rerun and 1 failed.

```

rakuco, 2021-11-01:
Hmm, those errors look related to the remoting-related GN args above IMO. Did you say you tried building without changing `GN_ARGS` or just without those specific arguments I mentioned above?

xemmits, 2021-11-01:
Ok. I did a cleanall and ran the build again after removing all my GN_ARGS. It resulted in some other errors in compile initially but finally succeeded when I added back the pulseaudio/alsa dependencies. So in general the build looks fine without my modifications. However, something has changed in the GN args since our last used version and that is for me to figure. Thanks for the support @rakuco .

Just an additional question. We are  trying to make the binary and yocto build as lightweight as possible, hence the GN_ARGS modifications, any other pointers? We are running a kiosk application on the chromium.

closed 2021-11-02 by xemmits

rakuco, 2021-11-02:
I'm glad to hear it worked out, @xemmits. As for the size optimizations: that's hard to say. The Chromium project in general favors not conditionally enabling features (unlike projects such as WebKit, for example), so even when there are some build toggles you can change it's possible they are not actively tested upstream and can result in build errors like those you encountered (especially if you mix multiple changes to the arguments). It depends on the use case for your deployment and what features you need enabled there

One thing that I advise though is to avoid setting `is_official_build=false`. This disables some build optimizations and enables some extra sanity checks that make Chromium significantly slower at runtime.