OSSystems/meta-browser #582: Version 92.0.4515.107 - ozone/wayland fails to boot weston 8.0.0

JeroenHiddink, 2021-11-01:
Hi,

I've build version 92.0.4515.107 and try to start Chromium on my platform (which is running iMX8M Mini, weston 8.0.0).
During boot (chromium --no-sandbox) , the following error occurs:

[2851030.448] wl_display@1.error(wl_display@1, 1, "invalid arguments for zwp_linux_explicit_synchronization_v1@5.get_synchronization
")

[clientlog.txt](https://github.com/OSSystems/meta-browser/files/7452848/clientlog.txt)

The weston serverlog:
[04:10:40.939] libwayland: unknown object (19), message get_synchronization(no)
[04:10:40.940] libwayland: error in client communication (pid 2536)
[04:10:40.953] libwayland: invalid object (6), type (wp_viewporter), message attach(?oii)
[04:10:40.953] libwayland: error in client communication (pid 2536)

Object (19) is the 'wl_surface@19' parameter in the zwp_linux_explicit_synchronization_v1@14.get_synchronization-call.
However, this object has been created just before that (client log): 
wl_compositor@4.create_surface(new id wl_surface@19)

So, I'm a bit confused what is going wrong here.

Eventually, after a few seconds, this error pops up:
[2536:2591:0101/041040.953989:ERROR:wayland_event_watcher.cc(249)] Fatal Wayland communication error: Broken pipe
And 
EGL: errno=22 (Invalid argument)

Also, nothing is shown on the display.

When I run Chromium with --disable-gpu-compositing, all works ok, except for the hardware acceleration which results in significant higher CPU-load.

Maybe anyone can help me with this?
For example, is this version of Chromium wayland compatible woth weston 8.0.0 / wayland protocol 1.18 ?

Thanks!

msisov, 2021-11-01:
https://chromium-review.googlesource.com/c/chromium/src/+/2964936 and https://chromium-review.googlesource.com/c/chromium/src/+/3160336 fixed the issue.

rakuco, 2021-11-01:
@msisov the first commit landed in M93 but the second is M96 material. Does it need to be backported?

msisov, 2021-11-01:
It'll be part of the next release.

rakuco, 2021-11-01:
OK, then I guess it makes sense to keep this issue open until then?

Also, does it mean it's unrelated to the Wayland/Weston versions?

msisov, 2021-11-01:
> OK, then I guess it makes sense to keep this issue open until then?
Ok.

> Also, does it mean it's unrelated to the Wayland/Weston versions?
It doesn't.

JeroenHid, 2021-11-01:
> https://chromium-review.googlesource.com/c/chromium/src/+/2964936 and https://chromium-review.googlesource.com/c/chromium/src/+/3160336 fixed the issue.

Thanks, I'll have a look there.

JeroenHid, 2021-11-25:
Building M96 for test, see also [#583 ](https://github.com/OSSystems/meta-browser/issues/583#issuecomment-979348787)