OSSystems/meta-browser #583: failed to boot chromium 94.0.4606.71.stable on raspberry 4b

jinxie0731, 2021-11-04:
Hi,

I'm trying to build chromimu from source code for raspberry 4b using chromium-ozone-wayland_94.0.4606.71.bb, build is sucessful but during the boot (click chrom icon) , the following error occurs, and if I boot chromium from command line(chromium --no-sandbox), similar errors are printed, so what's wrong and how to fix this issue?  

sh-5.0$ journal -f
Nov 04 02:57:46 raspberrypi4-64 systemd[606]: Started Application launched by gnome-shell.
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1330]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1330 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.330:23): auid=1000 uid=1000 gid=1000 ses=2 pid=1330 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1546.927454] audit: type=1701 audit(1635994667.330:23): auid=1000 uid=1000 gid=1000 ses=2 pid=1330 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.349938:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1332]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1332 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.362:24): auid=1000 uid=1000 gid=1000 ses=2 pid=1332 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1546.959667] audit: type=1701 audit(1635994667.362:24): auid=1000 uid=1000 gid=1000 ses=2 pid=1332 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.372963:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1334]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1334 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.398:25): auid=1000 uid=1000 gid=1000 ses=2 pid=1334 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1546.995733] audit: type=1701 audit(1635994667.398:25): auid=1000 uid=1000 gid=1000 ses=2 pid=1334 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.407800:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1340]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1340 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.546:26): auid=1000 uid=1000 gid=1000 ses=2 pid=1340 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1547.146182] audit: type=1701 audit(1635994667.546:26): auid=1000 uid=1000 gid=1000 ses=2 pid=1340 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.580931:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1344]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1344 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.606:27): auid=1000 uid=1000 gid=1000 ses=2 pid=1344 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1547.204168] audit: type=1701 audit(1635994667.606:27): auid=1000 uid=1000 gid=1000 ses=2 pid=1344 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.615758:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1345]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1345 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.630:28): auid=1000 uid=1000 gid=1000 ses=2 pid=1345 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1547.228988] audit: type=1701 audit(1635994667.630:28): auid=1000 uid=1000 gid=1000 ses=2 pid=1345 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1328:1104/025747.638583:ERROR:zygote_host_impl_linux.cc(263)] Failed to adjust OOM score of renderer with pid 1345: Permission denied (13)
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.642148:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1290:1104/025747.654860:ERROR:cursor_loader.cc(115)] Failed to load a platform cursor of type kNull
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1347]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1347 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.654:29): auid=1000 uid=1000 gid=1000 ses=2 pid=1347 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1547.252714] audit: type=1701 audit(1635994667.654:29): auid=1000 uid=1000 gid=1000 ses=2 pid=1347 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.663831:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1349]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1349 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.786:30): auid=1000 uid=1000 gid=1000 ses=2 pid=1349 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel[164]: [ 1547.383749] audit: type=1701 audit(1635994667.786:30): auid=1000 uid=1000 gid=1000 ses=2 pid=1349 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: [1290:1316:1104/025747.795316:ERROR:gpu_process_host.cc(956)] GPU process exited unexpectedly: exit_code=6
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
Nov 04 02:57:47 raspberrypi4-64 audit[1352]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1352 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.878:31): auid=1000 uid=1000 gid=1000 ses=2 pid=1352 comm="chromium-bin" exe="/usr/lib/chromium/chromium-bin" sig=6 res=1
Nov 04 02:57:47 raspberrypi4-64 chromium.desktop[1296]: *** stack smashing detected ***: terminated
[... cut ...]
Nov 04 02:57:48 raspberrypi4-64 kernel: audit: type=1701 audit(1635994667.998:32): auid=1000 uid=1000 gid=1000 ses=2 pid=1290 comm="Chrome_IOThread" exe="/usr/lib/chromium/chromium-bin" sig=5 res=1
Nov 04 02:57:48 raspberrypi4-64 kernel[164]: [ 1547.595487] audit: type=1701 audit(1635994667.998:32): auid=1000 uid=1000 gid=1000 ses=2 pid=1290 comm="Chrome_IOThread" exe="/usr/lib/chromium/chromium-bin" sig=5 res=1
Nov 04 02:57:48 raspberrypi4-64 audit[1331]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=1331 comm="Chrome_ChildIOT" exe="/usr/lib/chromium/chromium-bin" sig=4 res=1
Nov 04 02:57:48 raspberrypi4-64 systemd[606]: gnome-launched-chromium.desktop-1290.scope: Succeeded.

JeroenHid, 2021-11-04:
Hi,

I experience slightly the same problem on my platform (Dunfell, iMX8M Mini), same Chromium version.

Besides "Failed EGL API GetProcAddress" errors, I notice "*** stack smashing detected ***: terminated" errors and a Snap! in Chromium....

Any ideas?

[log.txt](https://github.com/OSSystems/meta-browser/files/7473142/log.txt)

rakuco, 2021-11-04:
cc @msisov for additional insight into all this GPU stuff

@JeroenHid @jinxie0731 do any of you get better results if you pass `--disable-gpu`?

JeroenHid, 2021-11-04:
@rakuco 

No more API errors, but stack smashing still occurring (including snap window):

imx8mm-cgt-sx8m:~# chromium --no-sandbox --disable-gpu
[656:690:0101/010844.280357:ERROR:bus.cc(392)] Failed to connect to the bus: Could not parse server address: Unknown address type (e
xamples of valid types are "tcp" and on UNIX "unix")
[656:690:0101/010844.281191:ERROR:bus.cc(392)] Failed to connect to the bus: Could not parse server address: Unknown address type (e
xamples of valid types are "tcp" and on UNIX "unix")
*** stack smashing detected ***: terminated
[656:656:0101/010844.509612:ERROR:cursor_loader.cc(115)] Failed to load a platform cursor of type kNull
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated

JeroenHid, 2021-11-04:
btw, I'm not very sure of the meta-clang version I've used to build Chromium. 
I ended up with using the dunfell-clang12 tag of https://github.com/kraj/meta-clang

kraj, 2021-11-04:
can you just use `dunfell` branch of meta-clang

JeroenHid, 2021-11-04:
I'll try...

Thanks for your quick reponses !

jinxie0731, 2021-11-04:
> cc @msisov for additional insight into all this GPU stuff
> 
> @JeroenHid @jinxie0731 do any of you get better results if you pass `--disable-gpu`?

@rakuco with option --disable-gpu, I got the same error !

msisov, 2021-11-04:
It's hard to say what's the problem there without a stack trace.

JeroenHid, 2021-11-04:
Hmm, that's true.
I need to build Chromium with stack traces enabled then ?

https://chromium.googlesource.com/chromiumos/docs/+/HEAD/stack_traces.md

rakuco, 2021-11-04:
@msisov just meant doing a debug build in the hopes of getting a stack trace with the failures and crashes.

rakuco, 2021-11-04:
I forgot what the incantations were for building a recipe in debug mode with poky, but be prepared to have a lot of memory and patience if you try to do it :-)

JeroenHid, 2021-11-04:
Memory and patience are available, now I need to find out how to make a debug build.

JeroenHid, 2021-11-04:
> can you just use `dunfell` branch of meta-clang

Building just finished by using the dunfell branch.
Errors regarding 'GetProcAddress' do still occur unfortunately.

[log2.txt](https://github.com/OSSystems/meta-browser/files/7478002/log2.txt)

 Also, the  '​stack smashing detected' otill occurs

msisov, 2021-11-05:
> @msisov just meant doing a debug build in the hopes of getting a stack trace with the failures and crashes.

One doesn't need to do a debug build. Simply start a gdbserver on a device and load the chrome binary via gdb-multiarch on the host. Then connect to the remote target inside the gdb session and start the chromium. It'll be enough to produce a meaningful stack trace.

jinxie0731, 2021-11-05:
For raspberry, it is not so easy to switch all meta to the dungell branch due to the compatibility of different meta_XXX. Even after the all these compatibility issues are solved, I still run into other build errors, so I'm preparing to drop this effort. Perhaps it makes more sense to trace the crash using GDB.

JeroenHid, 2021-11-05:
> One doesn't need to do a debug build. Simply start a gdbserver on a device and load the chrome binary via gdb-multiarch on the host. Then connect to the remote target inside the gdb session and [...]

See attached gdb and stdout logfiles, for  "chromium --no-sandbox"

I didn't catch a complete stack trace (like when a thread is really crashing), but I did see some other message in the logs, related to the API binding:

[467:520:0101/011919.220069:ERROR:gl_context_egl.cc(166)] eglBindApi failed with error EGL_BAD_PARAMETER
[467:520:0101/011919.220169:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[467:520:0101/011919.220206:ERROR:shared_image_stub.cc(524)] SharedImageStub: unable to create context
[467:520:0101/011919.220256:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub

Still, there's no additional information about the code location where the compiler check for array overflows occur (stack smashing detected).
Could this happen, because the proces ID being debugged is not the one actually crashing ?
Because I notice after boot there are 5 chromium processes, of which only one is crashing:

  33193 ttymxc1  00:00:07 chromium-bin
  33198 ttymxc1  00:00:00 chromium-bin
  33202 ttymxc1  00:00:02 chromium-bin
  33221 ttymxc1  00:00:00 chromium-bin
  **36714** ttymxc1  00:00:00 chromium-bin

*** stack smashing detected ***: terminated

  33193 ttymxc1  00:00:07 chromium-bin
  33198 ttymxc1  00:00:00 chromium-bin
  33202 ttymxc1  00:00:02 chromium-bin
  33221 ttymxc1  00:00:00 chromium-bin
  **36949** ttymxc1  00:00:00 chromium-bin

I tried to attach to this PID, but it's terminated before I can get the PID and attach the gdbserver.
[gdb session - stdout.txt](https://github.com/OSSystems/meta-browser/files/7482314/gdb.session.-.stdout.txt)
[gdb session.txt](https://github.com/OSSystems/meta-browser/files/7482315/gdb.session.txt)

JeroenHid, 2021-11-05:
> @msisov just meant doing a debug build in the hopes of getting a stack trace with the failures and crashes.
>
>
> One doesn't need to do a debug build. Simply start a gdbserver on a device and load the [...]

Tried debugging with --single-process:

Thread 5 "CompositorTileW" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 38513.38549]
0x0000aaaaaf751df8 in ?? ()

msisov, 2021-11-05:
eglBindApi fails with the EGL_BAD_PARAMETER parameter, which happens when "api is not one of the accepted tokens, or if the specified client API is not supported by the EGL implementation." according to the https://www.khronos.org/registry/EGL/sdk/docs/man/html/eglBindAPI.xhtml .

Something related to the egl implementation on your rpi4 gpu driver.

jinxie0731, 2021-11-07:
> One doesn't need to do a debug build. Simply start a gdbserver on a device and load the chrome binary via gdb-multiarch on the host. Then connect to the remote target inside the gdb session and [...]

@msisov could you look at this stack trace too?
[chromium-gdb-with-egl.log](https://github.com/OSSystems/meta-browser/files/7491799/chromium-gdb-with-egl.log)

jinxie0731, 2021-11-07:
Attach the stack traces that get from different chromium arguments:
1. --ozone-platform=wayland  --single-process
[chromium-gdb-single-process.log](https://github.com/OSSystems/meta-browser/files/7491824/chromium-gdb-single-process.log)
2. --ozone-platform=wayland -single-process -no-sandbox
[chromium-gdb-single-process-no-sandbox.log](https://github.com/OSSystems/meta-browser/files/7491826/chromium-gdb-single-process-no-sandbox.log)

msisov, 2021-11-08:
> drmPrimeFDToHandle() failed: Invalid argument
> Failed to create scanout resource.
> [1707:1743:1105/135209.215104:ERROR:gl_surface_egl.cc(780)] EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture
> [1707:1743:1105/135209.215502:ERROR:gl_image_egl.cc(42)] Error creating EGLImage: EGL_BAD_ALLOC
> [1707:1743:1105/135209.215973:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(267)] Failed to create GLImage 1280x1536, BGRA_8888, usage SCANOUT
> drmPrimeFDToHandle() failed: Invalid argument
> Failed to create scanout resource.

It looks the same as https://github.com/raspberrypi/linux/issues/3411

JeroenHid, 2021-11-08:
> eglBindApi fails with the EGL_BAD_PARAMETER parameter, which happens when "api is not one of the accepted tokens, or if the specified client API is not supported by the EGL implementation." [...]

That's strange then, I just updated from M92 to M94 and left the rest of the platform as-is.
During building M94 the following error occurs (while M92 builds fine), updating meta-clang (dunfell) fixed this, but at runtime these API bind failures occurs:

ERROR: Nothing PROVIDES 'virtual/libc-native' (but virtual:native:/yocto/vri-linux/build/../meta-clang/recipes-devtools/clang/compiler-rt_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/libgl-native
  virtual/librpc-native
  virtual/libx11-native
NOTE: Runtime target 'chromium-ozone-wayland' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['chromium-ozone-wayland', 'gn-native', 'compiler-rt-native', 'virtual/libc-native']
ERROR: Required build target 'vt3-mcu' has no buildable providers.
Missing or unbuildable dependency chain was: ['vt3-mcu', 'chromium-ozone-wayland', 'gn-native', 'compiler-rt-native', 'virtual/libc-native']

BTW, I'm on a iMX8M Mini.

WujianSun, 2021-11-10:
> That's strange then, I just updated from M92 to M94 and left the rest of the platform as-is. During building M94 the following error occurs (while M92 builds fine), updating meta-clang (dunfell) [...]

Hi @JeroenHid ,
The issue happend because of the chromiumV94 building generate libEGL.so and libGLESv2.so in /usr/lib/chromium/.
You can manually remove these two libraries.

JeroenHid, 2021-11-10:
Removing those libs worked, no more EGL Bind errors.

Still, during boot EGL errors occur:
[352637.430] wl_display@1.error(wl_display@1, 1, "invalid arguments for 
 wp_linux_explicit_synchronization_v1@5.get_synchronization")
wl_display@1: error 1: invalid arguments for 
 wp_linux_explicit_synchronization_v1@5.get_synchronization
EGL: errno=22 (Invalid argument)

Apart from the '*** stack smashing detected ***: terminated' error.

Ofc, no tabs are shown...

[log3.txt](https://github.com/OSSystems/meta-browser/files/7511954/log3.txt)

WujianSun, 2021-11-10:
Hi @JeroenHid ,
Seems the error happened about the wayland fence error, so could you please provide the weston log in /etc/xdg/weston/weston.log?
Also you can try using gl-renderer to check the issue as follow steps
1.vi /etc/xdg/weston/weston.ini
2. Add #use_g2d 
3. systemctl restart weston 
4. run chromium

JeroenHid, 2021-11-10:
Hi @WujianSun 

This is the weston log file:
[weston.log](https://github.com/OSSystems/meta-browser/files/7512017/weston.log)

I also changed weston.ini:
#use-g2d=1

restarted weston :
[weston2.log](https://github.com/OSSystems/meta-browser/files/7512044/weston2.log)
It is using GL renderer in stead of G2D now.

Same errors occur though:
wl_display@1: error 1: invalid arguments for zwp_linux_explicit_synchronization_v1@5.get_synchronization
EGL: errno=22 (Invalid argument)
EGL: errno=22 (Invalid argument)
EGL: errno=22 (Invalid argument)

weston log:
[01:04:14.226] libwayland: unknown object (24), message get_synchronization(no)
[01:04:14.227] libwayland: error in client communication (pid 619)
[01:04:21.629] libwayland: invalid object (6), type (wp_viewporter), message attach(?oii)
[01:04:22.531] libwayland: error in client communication (pid 619)

Any ideas ?

Thanks btw!

WujianSun, 2021-11-11:
Hi @JeroenHid ,
Please enbale gbm build in yotco and rebuild chromium, in mete-imx.

```
diff --git a/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend b/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend
index ac7f5d077..276dff74e 100644
--- a/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend
+++ b/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend
@@ -20,6 +20,5 @@ GN_ARGS += ' \
 # gbm is availiable only for mx8, but still drop it for 8MQ in 2020 Q4 release
 GN_ARGS_append_mx6 = " use_system_minigbm=false use_wayland_gbm=false"
 GN_ARGS_append_mx7 = " use_system_minigbm=false use_wayland_gbm=false"
-GN_ARGS_append_mx8 = " use_system_minigbm=false use_wayland_gbm=false"
 
 CHROMIUM_EXTRA_ARGS_append = " --disable-features=VizDisplayCompositor --in-process-gpu"
```

JeroenHid, 2021-11-12:
> Hi @JeroenHid , Please enable gbm build in yotco and rebuild chromium, in mete-imx.
> 
> ```
> diff --git a/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend b/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend
> index ac7f5d077..276dff74e 100644
> --- a/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend
> +++ b/meta-sdk/dynamic-layers/browser-layer/recipes-browser/chromium/chromium-ozone-wayland_%.bbappend
> @@ -20,6 +20,5 @@ GN_ARGS += ' \
>  # gbm is availiable only for mx8, but still drop it for 8MQ in 2020 Q4 release
>  GN_ARGS_append_mx6 = " use_system_minigbm=false use_wayland_gbm=false"
>  GN_ARGS_append_mx7 = " use_system_minigbm=false use_wayland_gbm=false"
> -GN_ARGS_append_mx8 = " use_system_minigbm=false use_wayland_gbm=false"
>  
>  CHROMIUM_EXTRA_ARGS_append = " --disable-features=VizDisplayCompositor --in-process-gpu"
> ```

Hi @WujianSun 

GBM was already enbaled in my buiold, if I'm correctly: 
GN_ARGS contains "use_wayland_gbm=true".

I checked chromium-ozone-wayland_%.bbappend of meta-sdk and the line with "GN_ARGS_append_mx8" was already removed I guess.

Maybe it's better to wait for release M95/M96 ?

rakuco, 2021-11-22:
FWIW, marex on Libera.Chat mentioned that the stack smashing messages do not show up when clang 13 (from honister) is used, while it happens with clang 12 on dunfell. Using clang 13 on dunfell also causes those messages to go away.

Can any of the affected users try to test this? We could try either backporting the required fixes to meta-clang's dunfell branch, or try moving to clang 13 in meta-clang's dunfell branch. It could also be the case that it's other changes to the `.conf` files in meta-clang's more recent branches that make this go away, I guess?

cc @kraj

kraj, 2021-11-22:
I think we perhaps would need `meta-clang-mixin` layer with clang13 in it

JeroenHid, 2021-11-22:
> FWIW, marex on Libera.Chat mentioned that the stack smashing messages do not show up when clang 13 (from honister) is used, while it happens with clang 12 on dunfell. Using clang 13 on dunfell also [...]

I could have a try at it, but because the build takes   around 12 hours expect results in a day.

JeroenHid, 2021-11-22:
> I could have a try at it, but because the build takes around 12 hours expect results in a day.

Hmmm, that's a bit hard. 
Tried to use clang 13 from honister in my Poky build (Dunfell 3.1.2), but in the Honister-branch of meta-clang, in the bb/bbappend files the underscore were replaced by semicolons.
E.g. "RDEPENDS_${PN}" became "RDEPENDS:${PN}", and "TUNE_CCARGS_remove_toolchain-clang" became "TUNE_CCARGS:remove:toolchain-clang"

My guess is I need to revert all these replacements in all meta-clang recipes;, perhaps any of you know a shortcut here ?

kraj, 2021-11-22:
yeah you need to adjust for override syntax changes, Marex mentioned on IRC that he did these adjustments locally as well. Perhaps you can reach out to him and get the changes to test.

shr-project, 2021-11-22:
> Tried to use clang 13 from honister in my Poky build (Dunfell 3.1.2), but in the Honister-branch of meta-clang, in the bb/bbappend files the underscore were replaced by semicolons.

New overrides syntax is supported in newer dunfell, why are you stuck on 3.1.2, can you update to 3.1.11 or latest dunfell?

JeroenHid, 2021-11-22:
Thanks, Ill be working on it (build is running).
We're still on 3.1.2 cause we were focussing on getting weston/wayland/chromium running with GPU rendering ;-)

Will update to latest Dunfell in the future.

marex, 2021-11-22:
The patch applies on honister branch commit `826f728 ("nativesdk-clang: Fix build when using RUNTIME = llvm")`:

```diff
diff --git a/conf/layer.conf b/conf/layer.conf
index 122cfbe..c051029 100644
--- a/conf/layer.conf
+++ b/conf/layer.conf
@@ -7,7 +7,7 @@ BBFILES += "${LAYERDIR}/recipes*/*/*.bb ${LAYERDIR}/recipes*/*/*.bbappend"
 BBFILE_COLLECTIONS += "clang-layer"
 BBFILE_PATTERN_clang-layer := "^${LAYERDIR}/"
 BBFILE_PRIORITY_clang-layer = "7"
-LAYERSERIES_COMPAT_clang-layer = "hardknott honister"
+LAYERSERIES_COMPAT_clang-layer = "dunfell hardknott honister"
 LAYERDEPENDS_clang-layer = "core"
 
 BBFILES_DYNAMIC += " \
diff --git a/recipes-core/busybox/busybox/0001-Turn-ptr_to_globals-and-bb_errno-to-be-non-const.patch b/recipes-core/busybox/busybox/0001-Turn-ptr_to_globals-and-bb_errno-to-be-non-const.patch
deleted file mode 100644
index 468a32d..0000000
diff --git a/recipes-core/busybox/busybox_1.34%.bbappend b/recipes-core/busybox/busybox_1.34%.bbappend
deleted file mode 100644
index 2c0af5e..0000000
```

JeroenHid, 2021-11-23:
> The patch applies on honister branch commit `826f728 ("nativesdk-clang: Fix build when using RUNTIME = llvm")`:
>
> ```diff
> diff --git a/conf/layer.conf b/conf/layer.conf
> index 122cfbe..c051029 100644 [...]

Thnx!

JeroenHid, 2021-11-25:
Hi,

Sorry for the delay.

Stack smashing errors have indeed disappeared when using clang 13 👍

Still, these errors remain on iMX8M:
- bind errors (Failed EGL API / ES_Common / GLES32 GetProcAddress), see log 1.
- removing libGLESv2.so / libEGL.so solves the bind-errors, but adds 
- error: "EGL: errno=22 (Invalid argument)", due to "Fatal Wayland communication error Broken pipe.", see log 2.

Weston log:
[01:10:41.702] libwayland: unknown object (21), message get_synchronization(no)
[01:10:41.702] libwayland: error in client communication (pid 743)
[01:10:41.711] libwayland: invalid object (6), type (wp_viewporter), message attach(?oii)
[01:10:41.712] libwayland: error in client communication (pid 743)

All cases are started with "CHROME_EXTRA_ARGS="         --use-gl=egl           --ozone-platform=wayland --disable-features=VizDisplayCompositor --in-process-gpu" in chromium-wrapper.

Tested with Chromium 95.0.4638.69

Any thoughts ?

[1 - bind errors.txt](https://github.com/OSSystems/meta-browser/files/7601456/1.-.bind.errors.txt)
[2 - without libs.txt](https://github.com/OSSystems/meta-browser/files/7601461/2.-.without.libs.txt)

JeroenHid, 2021-11-25:
Trying without GPU compositing will display the browser tab initially.
After that, it seems unresponsive. 
I think because of this Weston error: 
"[607:644:0101/010302.096274:ERROR:wayland_buffer_manager_gpu.cc(103)] WaylandBufferManagerGpu is not initialized. Can't register a surface."

[3 - without GPU compositing.txt](https://github.com/OSSystems/meta-browser/files/7601659/3.-.without.GPU.compositing.txt)

msisov, 2021-11-25:
> Hi,
> 
> Sorry for the delay.
> 
> Stack smashing errors have indeed disappeared when using clang 13 +1
> 
> Still, these errors remain on iMX8M:
> 
> * bind errors (Failed EGL API / ES_Common / GLES32 GetProcAddress), see log 1.
> * removing libGLESv2.so / libEGL.so solves the bind-errors, but adds
> * error: "EGL: errno=22 (Invalid argument)", due to "Fatal Wayland communication error Broken pipe.", see log 2.
> 
> Weston log: [01:10:41.702] libwayland: unknown object (21), message get_synchronization(no) [01:10:41.702] libwayland: error in client communication (pid 743) [01:10:41.711] libwayland: invalid object (6), type (wp_viewporter), message attach(?oii) [01:10:41.712] libwayland: error in client communication (pid 743)
> 
> All cases are started with "CHROME_EXTRA_ARGS=" --use-gl=egl --ozone-platform=wayland --disable-features=VizDisplayCompositor --in-process-gpu" in chromium-wrapper.
> 
> Tested with Chromium 95.0.4638.69
> 
> Any thoughts ? My goal is to run the browser with GPU rendering.
> 
> [1 - bind errors.txt](https://github.com/OSSystems/meta-browser/files/7601456/1.-.bind.errors.txt) [2 - without libs.txt](https://github.com/OSSystems/meta-browser/files/7601461/2.-.without.libs.txt)

The error is [521444.543] wl_display@1.error(wl_display@1, 1, "invalid arguments for zwp_linux_explicit_synchronization_v1@5.get_synchronization". 
M96 contains the necessary fix. Please use that.

msisov, 2021-11-25:
> Trying without GPU compositing will display the browser tab initially. After that, it seems unresponsive. I think because of this Weston error: [...]

I fixed that one week ago in the upstream. Let me backport the patch.

JeroenHid, 2021-11-25:
> 

Would be great!
I'll build M96 then after your backport patch

JeroenHid, 2021-11-25:
Build started....

JeroenHid, 2021-11-26:
Hi,

I just have build version "Chromium 96.0.4664.45 stable", and running it on iMX8M, dunfell.

Before I conclude that the reported error is/isn't resolved, I would like to ask a few questions about starting Chromium. 
Just to make sure I'm setting it up the right way before jumping to conclusions.

Question 1
What is the proper way to start Chromium: by using 'chromium', which links to the /usr/lib/chromium/chromium-wrapper, or by using chromium-bin ?
The first option sets the (lib-)path to /usr/lib/chromium e.g. and adds CHROME_EXTRA_ARGS.

Question 2 - API Bind failures
Can someone confirm I really need to delete libEGL.so and libGLESv2.so from /usr/lib/chromium and or /usr/lib/chromium.swiftshader to prevent these GetProcAddress-failures?
- Failed EGL API GetProcAddress
- Failed ES_Common API GetProcAddress
- Failed GLES32 API GetProcAddress

If so, which EGL/GLESv2 libs are being used instead?

After removing both libs, and starting Chromium, these errors occurs, and ANGLE is used, in stead of GL:
	[2706:2706:0101/041142.058890:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=139
	[2706:2706:0101/041143.096994:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=139
	[2706:2706:0101/041144.146424:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=139
	[2806:2806:0101/041144.155341:ERROR:egl_util.cc(74)] Failed to load GLES library: /usr/lib/chromium/libGLESv2.so: /usr/lib/chromium/
	libGLESv2.so: cannot open shared object file: No such file or directory
	[2806:2806:0101/041144.160323:ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
	[2813:2813:0101/041144.175723:ERROR:gpu_init.cc(457)] Passthrough is not supported, GL is disabled, ANGLE is

Seems obvious, since I removed the libs in the first place....

After these errors, chromium continues to boot, and looks ok.
But: there's no hardware acceleration, so CPU load is ~75% (quadcore iMX8M Mini) and the tabs load very slow. Sometimes the Page unresponsive -dialogue is shown, for an empty tab lol.
No Wayland errors though ;-)

I do have libEGL and libGLESv2 in my root file system, it just looks like chromium cannot find or use them ?

So, how to proceed to get EGL running (hardware acceleration).
I know this is a bit off-topic, regarding the original issue-report. 
I would appreciate any help, feel free to contact me by email.

kmeinhar, 2021-12-13:
We had a similar issue that Chromium was crashing with error message `stack smashing` ([Full Error Log](https://github.com/OSSystems/meta-browser/files/7702155/log2.txt)).

We are using a platform without a GPU and no DRI/DRM. Instead Weston uses the fbdev backend.
Our solution was to move to branch honister of the meta-clang layer and applying the patch from @marex .

> @kraj : I think we perhaps would need `meta-clang-mixin` layer with clang13 in it

This might be helpful for others that face the same problem. Depending on the implications it could also be useful to move the whole dunfell branch to clang13.

munoz0raul, 2022-01-19:
Hi @JeroenHid, Did you have any progress on that?

I'm trying chromium-ozone-wayland-96.0.4664.110-r0 on the iMX8M-mini evk. 

By just adding the layer and enabling it, it worked but still did not use the GPU.

Could someone please confirm if removing the `libEGL.so` and `libGLESv2.so` from `${D}${libdir}/chromium/ `is the way to go for iMX8?

https://github.com/OSSystems/meta-browser/blob/master/meta-chromium/recipes-browser/chromium/chromium-gn.inc#L455

I'm also confused about the `PACKAGECONFIG`. Does iMX8 support `use-vapi`? I tried to enable it but crashed. 

`| ../libva-2.10.0/va/glx/va_glx.h:29:10: fatal error: GL/glx.h: No such file or directory`

How about `proprietary-codecs`, should I enable it for iMX8?

msisov, 2022-01-19:
> Hi @JeroenHid, Did you have any progress on that?
>
> I'm trying `chromium-ozone-wayland-96.0.4664.110-r0` on the iMX8M-mini evk.
>
> By just adding the layer and enabling it, it worked but still did not [...]

Did you start the browser with --in-process-gpu flag? IIRC, iMX doesn't support drm render nodes, and, thus cannot use surfaceless path. Instead, it has to use Wayland EGL, which is only available with the flag I mentioned.

munoz0raul, 2022-01-19:
> Did you start the browser with --in-process-gpu flag?

Hi @msisov, thanks for your quick reply.

I tried with --in-process-gpu as well, see the log below:

```
chromium --use-gl=egl --ozone-platform=wayland --no-sandbox  --in-process-gpu
[729:784:0119/062355.149187:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[729:784:0119/062355.149425:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[729:729:0119/062355.264350:ERROR:cursor_loader.cc(114)] Failed to load a platform cursor of type kNull
[     1] ioctl(DRM_IOCTL_GEM_CLOSE) failed
[729:757:0119/062355.684474:ERROR:gbm_pixmap_wayland.cc(80)] Cannot create bo with format= BGRA_8888 and usage=SCANOUT
[729:757:0119/062355.684559:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(260)] Failed to create pixmap 1280x1024, BGRA_8888, usage SCANOUT
[     2] ioctl(DRM_IOCTL_GEM_CLOSE) failed
[729:757:0119/062355.684882:ERROR:gbm_pixmap_wayland.cc(80)] Cannot create bo with format= BGRA_8888 and usage=GPU_READ
[729:757:0119/062355.684924:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(260)] Failed to create pixmap 1280x1024, BGRA_8888, usage GPU_READ
[729:757:0119/062355.684949:ERROR:shared_image_backing_factory_gl_image.cc(351)] CreateSharedImage: Failed to create bindable image
[729:757:0119/062355.684971:ERROR:shared_image_factory.cc(584)] CreateSharedImage: could not create backing.
[729:769:0119/062355.685326:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.685469:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.685830:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:757:0119/062355.685873:ERROR:raster_decoder.cc(1233)]   RasterDecoderImpl: Context lost during MakeCurrent.
[729:769:0119/062355.686395:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.686560:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.686596:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.686624:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.686686:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.686738:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.686788:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687272:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687342:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687367:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687427:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687576:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687620:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687645:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:769:0119/062355.687667:ERROR:skia_renderer.cc(647)] Failed to create the promise sk image or get paint ops.
[729:757:0119/062355.688242:ERROR:skia_output_surface_impl_on_gpu.cc(1319)] Failed to make current.
[729:757:0119/062355.689424:ERROR:skia_output_surface_impl_on_gpu.cc(1319)] Failed to make current.
[729:757:0119/062355.690645:ERROR:skia_output_surface_impl_on_gpu.cc(1319)] Failed to make current.
[729:757:0119/062355.691329:ERROR:raster_decoder.cc(1233)]   RasterDecoderImpl: Context lost during MakeCurrent.
[729:757:0119/062355.693150:ERROR:shared_image_stub.cc(534)] SharedImageStub: context already lost
[729:757:0119/062355.700890:ERROR:shared_image_stub.cc(534)] SharedImageStub: context already lost
[     3] ioctl(DRM_IOCTL_GEM_CLOSE) failed
[729:757:0119/062355.820723:ERROR:gbm_pixmap_wayland.cc(80)] Cannot create bo with format= BGRA_8888 and usage=SCANOUT
[729:757:0119/062355.820770:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(260)] Failed to create pixmap 1280x1024, BGRA_8888, usage SCANOUT
[     4] ioctl(DRM_IOCTL_GEM_CLOSE) failed
[729:757:0119/062355.821039:ERROR:gbm_pixmap_wayland.cc(80)] Cannot create bo with format= BGRA_8888 and usage=GPU_READ
[729:757:0119/062355.821078:ERROR:gpu_memory_buffer_factory_native_pixmap.cc(260)] Failed to create pixmap 1280x1024, BGRA_8888, usage GPU_READ
[729:757:0119/062355.821104:ERROR:shared_image_backing_factory_gl_image.cc(351)] CreateSharedImage: Failed to create bindable image
[729:757:0119/062355.821125:ERROR:shared_image_factory.cc(584)] CreateSharedImage: could not create backing.
Trace/breakpoint trap
```
This log is my Yocto build + meta-chromium enabling `proprietary-codecs`

```
-PACKAGECONFIG ??= "upower use-egl"
+PACKAGECONFIG ??= "upower use-egl proprietary-codecs"
```
>IIRC, iMX doesn't support drm render nodes, and, thus cannot use surfaceless path. Instead, it has to use Wayland EGL, which is only available with the flag I mentioned.

Didn't understand this comment, is it related to `proprietary-codecs` or `use-vapi`? Should I enable or disable it for iMX8M?

msisov, 2022-01-19:
looks like there is drm render nodes as it still tries to use dmabuf (see gbm_pixmap_wayland logs), but fails for some reason.

you can explicitly disable that by adding "use_wayland_gbm = false" in your gn args.

JeroenHid, 2022-01-19:
> Did you start the browser with --in-process-gpu flag? IIRC, iMX doesn't support drm render nodes, and, thus cannot use surfaceless path. Instead, it has to use Wayland EGL, which is only available [...]

Hi @msisov !
No, no real progress on this. I was stuck at running Chromium without GPU support.
Also, your question "Could someone please confirm if removing the `libEGL.so` and `libGLESv2.so` from `${D}${libdir}/chromium/ `is the way to go for iMX8?" is a good one, I don't have a proper answer on it.

My strategy since a few weeks, was to wait a while for new updates, and try them as I was stuck trying to get Chromium on iMX8 with hw acceleration...

ensc, 2022-01-20:
fwiw: here, with iMX8MM + iMX8MP chromium runs fine after removing `libEGL.so` + `libGLESv2.so`.   BSP has recent hardknott branches on freescale layers and nxp kernel 5.10.72.    The https://webglsamples.org/aquarium/aquarium.html shows 9-10fp on iMX8MM and 13-14fps on iMX8MP (both 1080p).

There was/is a problem that chromium fails to start with only 2GiB of RAM (4GiB are ok).

My flags are

```
use_cups=false
ffmpeg_branding="Chrome"
proprietary_codecs=true
use_vaapi=false
is_component_build=false
use_gnome_keyring=false
use_kerberos=false
use_pulseaudio=true
use_system_libjpeg=true
use_system_freetype=false
enable_js_type_check=false
host_pkg_config="pkg-config-native"
is_debug=false
is_official_build=true
use_custom_libcxx=false
use_lld=true
use_gold=false
symbol_level=0
enable_remoting=false
enable_nacl=false
use_sysroot=false
treat_warnings_as_errors=false
is_cfi=false
disable_fieldtrial_testing_config=true
chrome_pgo_phase=0
google_api_key="invalid-api-key"
google_default_client_id="invalid-client-id"
google_default_client_secret="invalid-client-secret"
custom_toolchain="//build/toolchain/yocto:yocto_target"
gold_path=""
host_toolchain="//build/toolchain/yocto:yocto_native"
is_clang=true
clang_base_path="..../chromium-ozone-wayland/96.0.4664.45-r0/recipe-sysroot-native/usr/bin"
clang_use_chrome_plugins=false
target_cpu="arm64"
v8_snapshot_toolchain="//build/toolchain/yocto:yocto_target"
max_jobs_per_link="16"
use_cups=false
ffmpeg_branding="Chrome"
proprietary_codecs=true
use_vaapi=false
use_ozone=true
ozone_auto_platforms=false
ozone_platform_headless=true
ozone_platform_wayland=true
ozone_platform_x11=false
system_wayland_scanner_path="..../chromium-ozone-wayland/96.0.4664.45-r0/recipe-sysroot-native/usr/bin/wayland-scanner"
use_system_wayland_scanner=true
use_xkbcommon=true
use_system_libwayland=true
use_system_minigbm=true
use_system_libdrm=true
use_gtk=false
```

JeroenHid, 2022-01-20:
@ensc Thanks for sharing!

Can you confirm hardware acceleration is enabled (visit the about page in the browser to see details)?

ensc, 2022-01-20:
### Graphics Feature Status
- Canvas: Hardware accelerated
- Canvas out-of-process rasterization: Disabled
- Compositing: Hardware accelerated
- Multiple Raster Threads: Disabled
- Out-of-process Rasterization: Disabled
- OpenGL: Enabled
- Rasterization: Software only. Hardware acceleration disabled
- Raw Draw: Disabled
- Skia Renderer: Enabled
- Video Decode: Hardware accelerated
- Vulkan: Disabled
- WebGL: Hardware accelerated
- WebGL2: Hardware accelerated

### Driver Bug Workarounds
- clear_uniforms_before_first_program_use
- enable_webgl_timer_query_extensions
- exit_on_context_lost
- disabled_extension_GL_KHR_blend_equation_advanced
- disabled_extension_GL_KHR_blend_equation_advanced_coherent

### Problems Detected
- Raster is using a single thread.
- Disabled Features: multiple_raster_threads
- Clear uniforms before first program use on all platforms: 124764, 349137
- Applied Workarounds: clear_uniforms_before_first_program_use
- Disable KHR_blend_equation_advanced until cc shaders are updated: 661715
- Applied Workarounds: disable(GL_KHR_blend_equation_advanced), disable(GL_KHR_blend_equation_advanced_coherent)
- Expose WebGL's disjoint_timer_query extensions on platforms with site isolation: 808744, 870491
- Applied Workarounds: enable_webgl_timer_query_extensions
- Some drivers can't recover after OUT_OF_MEM and context lost: 893177
- Applied Workarounds: exit_on_context_lost

### Version Information

 | | |
 |-|-|
 | Data exported                                             | 2022-01-20T11:56:46.042Z                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  |
 | Chrome version                                            | Chrome/96.0.4664.45                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       |
 | Operating system                                          | Linux 5.10.72                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             |
 | Software rendering list URL                               | https://chromium.googlesource.com/chromium/src/+/76e4c1bb2ab4671b8beba3444e61c0f17584b2fc/gpu/config/software_rendering_list.json                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                       |

### Video Acceleration Information
 - Encoding	

### Vulkan Information

### Device Performance Information

JeroenHid, 2022-01-20:
@ensc Coll, thanks!
I'll give it a try myself

rakuco, 2022-01-21:
> Do you mind telling me how did you get all this information on the table?

That's from chrome://gpu

jimini-lumox, 2022-05-11:
Hi @ensc, is the Video Decode acceleration enabled for the iMX8MM or just the iMX8MP ?