OSSystems/meta-browser #588: EGL issues on IMX8MM + Wayland

denizlitr, 2021-11-22:
I built Chromium 95.0.4638.69 (Ozone Wayland) for Imx8MM with Vivante libs under Yocto Hardknott. 

When I launch Chromium (with or without --in-process-gpu), first I get alot of warnings like this:
`[     1] Failed EGL API GetProcAddress: eglLockSurfaceKHR !
[     2] Failed EGL API GetProcAddress: eglUnlockSurfaceKHR !
[     3] Failed EGL API GetProcAddress: eglBindWaylandDisplayWL !
[     4] Failed EGL API GetProcAddress: eglUnbindWaylandDisplayWL !
[     5] Failed EGL API GetProcAddress: eglQueryWaylandBufferWL !
[     6] Failed EGL API GetProcAddress: eglUpdateWaylandBufferWL !
[     7] Failed EGL API GetProcAddress: eglSetDamageRegionKHR !
[     8] Failed EGL API GetProcAddress: eglSwapBuffersWithDamageEXT !
[     9] Failed EGL API GetProcAddress: eglQueryDmaBufFormatsEXT !
[    10] Failed EGL API GetProcAddress: eglQueryDmaBufModifiersEXT !
[    11] Failed EGL API GetProcAddress: eglPatchID !
[    12] Failed ES_Common API GetProcAddress: forward_glActiveTexture !
[    13] Failed ES_Common API GetProcAddress: forward_glBindBuffer !
[    14] Failed ES_Common API GetProcAddress: forward_glBindTexture !
[    15] Failed ES_Common API GetProcAddress: forward_glBlendFunc !
[    16] Failed ES_Common API GetProcAddress: forward_glBufferData !
[    17] Failed ES_Common API GetProcAddress: forward_glBufferSubData !
[    18] Failed ES_Common API GetProcAddress: forward_glClear !
[    19] Failed ES_Common API GetProcAddress: forward_glClearColor !
[    20] Failed ES_Common API GetProcAddress: forward_glClearDepthf !
[    21] Failed ES_Common API GetProcAddress: forward_glClearStencil !
[    22] Failed ES_Common API GetProcAddress: forward_glColorMask !`

Notice that the list is very long, I just cut it. In later lines I also get:
`[    87] Failed GLES32 API GetProcAddress: glMultiDrawArraysIndirectEXT !
[    88] Failed GLES32 API GetProcAddress: glMultiDrawElementsIndirectEXT !
[    89] Failed GLES32 API GetProcAddress: glGetTexImage !
[    90] Failed ES Common GLES2X API GetProcAddress: glMultiDrawArraysEXT !
[373:410:1111/101830.754991:ERROR:gl_context_egl.cc(166)] eglBindApi failed with error EGL_BAD_PARAMETER
[373:410:1111/101830.773625:ERROR:gl_context_egl.cc(166)] eglBindApi failed with error EGL_BAD_PARAMETER
[373:410:1111/101830.773695:ERROR:gpu_info_collector.cc(90)] gl::init::CreateGLContext failed
[373:410:1111/101830.773713:ERROR:gpu_info_collector.cc(389)] Could not create context for info collection.
[373:410:1111/101830.774529:ERROR:gl_context_egl.cc(166)] eglBindApi failed with error EGL_BAD_PARAMETER`

And warnings like

`EGL: Warning: No default display support on wayland
[569:569:1111/101924.422021:ERROR:gl_surface_egl.cc(1350)] EGL display query failed with error EGL_SUCCESS
[569:569:1111/101924.422627:ERROR:gl_surface_egl.cc(1375)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[569:569:1111/101924.423025:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[569:569:1111/101924.445521:ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
[530:530:1111/101924.507133:ERROR:cursor_loader.cc(115)] Failed to load a platform cursor of type kNull
[584:584:1111/101924.926538:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is swiftshader, ANGLE is 
[530:600:1111/101928.669055:ERROR:chrome_browser_main_extra_parts_metrics.cc(228)] crbug.com/1216328: Checking Bluetooth availability started. Please report if there is no report that this ends.
[530:600:1111/101928.669610:ERROR:chrome_browser_main_extra_parts_metrics.cc(231)] crbug.com/1216328: Checking Bluetooth availability ended.
[530:600:1111/101928.669689:ERROR:chrome_browser_main_extra_parts_metrics.cc(234)] crbug.com/1216328: Checking default browser status started. Please report if there is no report that this ends.
[530:600:1111/101928.690454:ERROR:chrome_browser_main_extra_parts_metrics.cc(238)] crbug.com/1216328: Checking default browser status ended.
[530:530:1111/101932.899099:ERROR:cursor_loader.cc(115)] Failed to load a platform cursor of type kHand`

Is there a workaround for this or is this an issue in yocto configuration?

JeroenHid, 2021-11-22:
Hi,

I have the same issue, see [https://github.com/OSSystems/meta-browser/issues/583#issuecomment-964911899]

_The issue happend because of the chromiumV94 building generate libEGL.so and libGLESv2.so in /usr/lib/chromium/.
You can manually remove these two libraries._

Removing the libs solves the issue, although I wasn't able to get GPU-rendering working.... (so I'm stuck with high CPU)

denizlitr, 2021-11-22:
> Hi,
> 
> I have the same issue, see [https://github.com/[/issues/583](https://github.com/OSSystems/meta-browser/issues/583)#issuecomment-964911899]
> 
> _The issue happend because of the chromiumV94 building generate libEGL.so and libGLESv2.so in /usr/lib/chromium/. You can manually remove these two libraries._
> 
> Removing the libs solves the issue, although I wasn't able to get GPU-rendering working.... (so I'm stuck with high CPU)

I already tried this. My goal is to get high benchmark values in https://browserbench.org when I run "Motion Mark" tests. When I remove these libs from /usr/lib/chromium, I don't get these warnings, but I get 50% of the bench values (~15 instead of 30). 

These libs seem to come from swift shader.