OSSystems/meta-browser #591: chromium-ozone-wayland 96.0.4664.45 --in-process-gpu no longer working

MastaG, 2021-11-29:
Hi @msisov 

Previously you had fixed the use of --in-process-gpu for wayland when drivers do not support dmabuf (or any other modern features).
After upgrading to 96.0.4664.45 it doesn't seem to work any longer.
It now renders a black border around the window, however for some strange reason when taking a screenshot, it will fill the border with garbage.
[image]
.
I'm running it on Fedora 35 with: ```--use-gl=egl --ozone-platform=wayland --no-sandbox --ignore-gpu-blocklist --ignore-gpu-blacklist --in-process-gpu```

Full report:
[report.txt](https://github.com/OSSystems/meta-browser/files/7618578/report.txt)

MastaG, 2021-11-29:
Update:

When adding: ```--enable-native-gpu-memory-buffers --enable-gpu-rasterization --enable-impl-side-painting``` it seems to be fully accelerated and snappy again.

However the strange thick border remains.
It seems to be the the edge of the window which you can hover your mouse cursor over to resize it.

MastaG, 2021-12-15:
Hi @msisov 
Do you have any idea what could be the cause?
I am building with gtk support enabled btw (it's disabled by default for some reason), I could try to do a build without it.

msisov, 2021-12-15:
Sorry, I didn’t have time to work on this issue due to higher priority tasks. 

I wonder you can try chromium from the main (master) branch instead? There were some issues with frame insets. Maybe, this issue is one of those and have already been fixed.

MastaG, 2021-12-25:
Yep, it's fixed by using the latest git version.
Let's hope it will be backported soon.

closed 2021-12-25 by MastaG