OSSystems/meta-browser #599: No hardware acceleration with Chromium 96.0.4664.110 on Xilinx Zynq UltraScale 5EV (aarch64, Linux 5.10)

chris-laplante, 2021-12-22:
We are using the chromium-ozone-wayland recipe and running chromium with this command: `chromium --no-sandbox`

All forms of hardware acceleration seem to be disabled.

Here is the chrome://gpu report:

> Graphics Feature Status
> Canvas: Software only. Hardware acceleration disabled
> Canvas out-of-process rasterization: Disabled
> Compositing: Software only. Hardware acceleration disabled
> Multiple Raster Threads: Disabled
> Out-of-process Rasterization: Disabled
> OpenGL: Disabled
> Rasterization: Software only. Hardware acceleration disabled
> Raw Draw: Disabled
> Skia Renderer: Enabled
> Video Decode: Software only. Hardware acceleration disabled
> Vulkan: Disabled
> WebGL: Disabled
> WebGL2: Disabled
> Problems Detected
> Accelerated video decode has been disabled, either via blocklist, about:flags or the command line.
> Disabled Features: video_decode
> Gpu compositing has been disabled, either via blocklist, about:flags or the command line. The browser will fall back to software compositing and hardware acceleration will be unavailable.
> Disabled Features: gpu_compositing
> GPU process was unable to boot: GPU process crashed too many times with SwiftShader.
> Disabled Features: all
> Version Information
> Data exported	2021-12-22T15:25:32.126Z
> Chrome version	Chrome/96.0.4664.110
> Operating system	Linux 5.10.0-xilinx-v2021.1
> Software rendering list URL	https://chromium.googlesource.com/chromium/src/+/d5ef0e8214bc14c9b5bbf69a1515e431394c62a6/gpu/config/software_rendering_list.json
> Driver bug list URL	https://chromium.googlesource.com/chromium/src/+/d5ef0e8214bc14c9b5bbf69a1515e431394c62a6/gpu/config/gpu_driver_bug_list.json
> ANGLE commit id	unknown hash
> 2D graphics backend	Skia/96 5c2790986b220ac4f4dcc9d781f4164db30e4067
> Command Line	/usr/bin/chromium --use-gl=egl --ozone-platform=wayland --no-sandbox --enable-native-gpu-memory-buffers --enable-gpu-rasterization --enable-crashpad --flag-switches-begin --flag-switches-end
> Driver Information
> Initialization time	0
> In-process GPU	false
> Passthrough Command Decoder	false
> Sandboxed	false
> GPU0	VENDOR= 0x0000, DEVICE=0x0000
> Optimus	false
> AMD switchable	false
> Driver vendor	
> Driver version	
> GPU CUDA compute capability major version	0
> Pixel shader version	
> Vertex shader version	
> Max. MSAA samples	
> Machine model name	
> Machine model version	
> GL_VENDOR	Disabled
> GL_RENDERER	Disabled
> GL_VERSION	Disabled
> GL_EXTENSIONS	
> Disabled Extensions	
> Disabled WebGL Extensions	
> Window system binding vendor	
> Window system binding version	
> Window system binding extensions	
> Interfaces exposed by the Wayland compositor	wl_compositor:4 wl_subcompositor:1 wp_viewporter:1 zxdg_output_manager_v1:2 wp_presentation:1 zwp_relative_pointer_manager_v1:1 zwp_pointer_constraints_v1:1 zwp_input_timestamps_manager_v1:1 wl_data_device_manager:3 wl_shm:1 mali_buffer_sharing:2 wl_seat:7 weston_direct_display_v1:1 weston_content_protection:1 wl_output:3 zwp_input_panel_v1:1 zwp_input_method_v1:1 zwp_text_input_manager_v1:1 xdg_wm_base:1 zxdg_shell_v6:1 wl_shell:1 weston_desktop_shell:1 weston_screenshooter:1 zwp_linux_dmabuf_v1:3
> Ozone platform	wayland
> Direct rendering version	unknown
> Reset notification strategy	0x0000
> GPU process crash count	3
> gfx::BufferFormats supported for allocation and texturing	R_8: not supported, R_16: not supported, RG_88: not supported, BGR_565: not supported, RGBA_4444: not supported, RGBX_8888: not supported, RGBA_8888: not supported, BGRX_8888: not supported, BGRA_1010102: not supported, RGBA_1010102: not supported, BGRA_8888: not supported, RGBA_F16: not supported, YVU_420: not supported, YUV_420_BIPLANAR: not supported, P010: not supported
> Compositor Information
> Tile Update Mode	One-copy
> Partial Raster	Enabled
[... 91 lines cut ...]
> [1756:1756:1222/151817.139199:ERROR:gpu_init.cc(457)] : Passthrough is not supported, GL is disabled, ANGLE is

Note that the message log indicates some EGL issues. Here is the full output I get from chromium when launching:

```
root@fpdlink:~# chromium --no-sandbox
[2959:2992:1222/152837.118128:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2959:2992:1222/152837.120926:ERROR:bus.cc(393)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2959:2959:1222/152837.160176:ERROR:cursor_loader.cc(114)] Failed to load a platform cursor of type kNull
[2959:2959:1222/152837.262866:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=11
[2959:2959:1222/152837.341526:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=11
[2959:2959:1222/152837.393762:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=11
[3048:3048:1222/152837.447055:ERROR:angle_platform_impl.cc(44)] Display.cpp:894 (initialize): ANGLE Display::initialize error 0: Internal Vulkan error (-3): Initialization of an object could not be completed for implementation-specific reasons, in ../../third_party/angle/src/libANGLE/renderer/vulkan/RendererVk.cpp, initialize:1048.
[3048:3048:1222/152837.447350:ERROR:gl_surface_egl.cc(782)] EGL Driver message (Critical) eglInitialize: Internal Vulkan error (-3): Initialization of an object could not be completed for implementation-specific reasons, in ../../third_party/angle/src/libANGLE/renderer/vulkan/RendererVk.cpp, initialize:1048.
[3048:3048:1222/152837.447470:ERROR:gl_surface_egl.cc(1382)] eglInitialize SwANGLE failed with error EGL_NOT_INITIALIZED
[3048:3048:1222/152837.447585:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[3048:3048:1222/152837.453326:ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
[3061:3061:1222/152837.476122:ERROR:gpu_init.cc(457)] Passthrough is not supported, GL is disabled, ANGLE is
[2959:2976:1222/152840.469961:ERROR:chrome_browser_main_extra_parts_metrics.cc(226)] crbug.com/1216328: Checking Bluetooth availability started. Please report if there is no report that this ends.
[2959:2976:1222/152840.470038:ERROR:chrome_browser_main_extra_parts_metrics.cc(229)] crbug.com/1216328: Checking Bluetooth availability ended.
[2959:2976:1222/152840.470068:ERROR:chrome_browser_main_extra_parts_metrics.cc(232)] crbug.com/1216328: Checking default browser status started. Please report if there is no report that this ends.
[2959:2976:1222/152840.485716:ERROR:chrome_browser_main_extra_parts_metrics.cc(236)] crbug.com/1216328: Checking default browser status ended.
```

Any ideas?

FLeblanc-OTI, 2023-05-18:
I am having the same issue, with the same errors.
Here's more log output

```
root@zynq:~# chromium --no-sandbox --enable-logging=stderr --v=0 --vmodule=metrics=2
[2022:2022:0517/075505.265464:WARNING:wayland_object.cc(93)] Binding to wl_output version 2 but version 3 is available.
[2049:2049:0517/075505.288197:WARNING:ozone_platform_wayland.cc(262)] Failed to find drm render node path.
[2022:2022:0517/075505.385919:WARNING:wayland_surface.cc(116)] Server doesn't support zcr_alpha_compositing_v1.
[2022:2022:0517/075505.385991:WARNING:wayland_surface.cc(127)] Server doesn't support overlay_prioritizer.
[2022:2022:0517/075505.387108:ERROR:cursor_loader.cc(114)] Failed to load a platform cursor of type kNull
[2022:2022:0517/075505.512786:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=11
[2022:2022:0517/075505.513152:WARNING:gpu_process_host.cc(1273)] The GPU process has crashed 1 time(s)
[2101:2101:0517/075505.532199:WARNING:ozone_platform_wayland.cc(262)] Failed to find drm render node path.
[2022:2022:0517/075505.590926:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=11
[2022:2022:0517/075505.590988:WARNING:gpu_process_host.cc(1273)] The GPU process has crashed 2 time(s)
[2106:2106:0517/075505.603237:WARNING:ozone_platform_wayland.cc(262)] Failed to find drm render node path.
[2022:2022:0517/075505.634994:ERROR:gpu_process_host.cc(967)] GPU process exited unexpectedly: exit_code=11
[2022:2022:0517/075505.635079:WARNING:gpu_process_host.cc(1273)] The GPU process has crashed 3 time(s)
[2109:2109:0517/075505.642055:WARNING:ozone_platform_wayland.cc(262)] Failed to find drm render node path.
[2109:2109:0517/075505.663358:ERROR:angle_platform_impl.cc(44)] Display.cpp:894 (initialize): ANGLE Display::initialize error 0: Internal Vulkan error (-3): Initialization of an object could not be completed for implementation-specific reasons, in ../../third_party/angle/src/libANGLE/renderer/vulkan/RendererVk.cpp, initialize:1048.
[2109:2109:0517/075505.663685:ERROR:gl_surface_egl.cc(782)] EGL Driver message (Critical) eglInitialize: Internal Vulkan error (-3): Initialization of an object could not be completed for implementation-specific reasons, in ../../third_party/angle/src/libANGLE/renderer/vulkan/RendererVk.cpp, initialize:1048.
[2109:2109:0517/075505.663821:ERROR:gl_surface_egl.cc(1382)] eglInitialize SwANGLE failed with error EGL_NOT_INITIALIZED
[2109:2109:0517/075505.663953:ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[2109:2109:0517/075505.668413:ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
[2116:2116:0517/075505.683978:WARNING:ozone_platform_wayland.cc(262)] Failed to find drm render node path.
[2116:2116:0517/075505.684370:ERROR:gpu_init.cc(457)] Passthrough is not supported, GL is disabled, ANGLE is 
[2022:2022:0517/075505.689794:WARNING:gpu_process_host.cc(995)] Reinitialized the GPU process after a crash. The reported initialization time was 0 ms
[2022:2064:0517/075510.560146:ERROR:chrome_browser_main_extra_parts_metrics.cc(226)] crbug.com/1216328: Checking Bluetooth availability started. Please report if there is no report that this ends.
[2022:2064:0517/075510.560223:ERROR:chrome_browser_main_extra_parts_metrics.cc(229)] crbug.com/1216328: Checking Bluetooth availability ended.
[2022:2064:0517/075510.560286:ERROR:chrome_browser_main_extra_parts_metrics.cc(232)] crbug.com/1216328: Checking default browser status started. Please report if there is no report that this ends.
[2022:2064:0517/075510.588055:ERROR:chrome_browser_main_extra_parts_metrics.cc(236)] crbug.com/1216328: Checking default browser status ended.
```

And `glmark2-es2-wayland` works fine:
```
root@zynq:~# glmark2-es2-wayland
=======================================================
    glmark2 2021.02
=======================================================
    OpenGL Information
    GL_VENDOR:     ARM
    GL_RENDERER:   Mali-400 MP
    GL_VERSION:    OpenGL ES 2.0 "67dc026"
=======================================================
[build] use-vbo=false: FPS: 627 FrameTime: 1.595 ms
[build] use-vbo=true: FPS: 778 FrameTime: 1.285 ms
[texture] texture-filter=nearest: FPS: 938 FrameTime: 1.066 ms
[texture] texture-filter=linear: FPS: 842 FrameTime: 1.188 ms
[...]
```

chris-laplante, 2023-05-18:
@FLeblanc-OTI we ended up upgrading to Chromium 100.0.4896.88 (had to port the recipe(s) in Yocto, but it wasn't a big deal). With that, we got hardware acceleration working. Still needed to pass --in-process-gpu though.

FLeblanc-OTI, 2023-05-18:
@chris-laplante Thanks for the tip and the fast response!!