OSSystems/meta-browser #600: Building raspberrypi4 with branch jansa/master fails

HJK17, 2022-01-11:
ERROR: Task (/home/hjk/poky/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_93.0.4577.82.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5357 tasks of which 5352 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/hjk/poky/meta-browser/meta-chromium/recipes-browser/chromium/chromium-x11_93.0.4577.82.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

rakuco, 2022-01-11:
Are you using `jansa/master` instead of `master` for a specific reason? Chromium 93 is pretty old at this point, and a lot of changes have gone into this repository since then.

HJK17, 2022-01-11:
> Are you using `jansa/master` instead of `master` for a specific reason? Chromium 93 is pretty old at this point, and a lot of changes have gone into this repository since then.

I can't load the web page using the first tab of chromium96, and the local one can't load either. But the second tab can load the webpage normally. I will try to use chromium93 whether such a problem will occur.

HJK17, 2022-01-12:
> Are you using `jansa/master` instead of `master` for a specific reason? Chromium 93 is pretty old at this point, and a lot of changes have gone into this repository since then.

I have solved the problem that the webpage cannot be opened, just add `--disable-gpu` to start chromium

closed 2022-01-12 by HJK17