OSSystems/meta-browser #601: Stack smashing detected on imx8mm when starting Chromium

coldspark29, 2022-01-11:
With the current commit d25d8ee98a656b2534d0eec6138ef264529fab4f, I am getting stack smashing errors. Previously I was using 
a5a5f277740a8434377f9844c62574efb17a4fa0 and everything worked fine. Seems like some egl error, but the only commit hash that has changed is the meta-browser one.
```
chromium --no-sandbox --ozone-platform=wayland --start-fullscreen --app --remote-debugging-port=9222 --user-agent="Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.114 Safari/537.36 AdvancedInterface/1" --use-gl=egl --in-processpu --enable-oop-rasterization --enable-native-gpu-memory-buffers --enable-zero-copy http://localhost:8081
[     1] Failed EGL API GetProcAddress: eglLockSurfaceKHR !
[     2] Failed EGL API GetProcAddress: eglUnlockSurfaceKHR !
[     3] Failed EGL API GetProcAddress: eglBindWaylandDisplayWL !
[     4] Failed EGL API GetProcAddress: eglUnbindWaylandDisplayWL !
[     5] Failed EGL API GetProcAddress: eglQueryWaylandBufferWL !
[     6] Failed EGL API GetProcAddress: eglUpdateWaylandBufferWL !
[     7] Failed EGL API GetProcAddress: eglSetDamageRegionKHR !
[     8] Failed EGL API GetProcAddress: eglSwapBuffersWithDamageEXT !
[     9] Failed EGL API GetProcAddress: eglQueryDmaBufFormatsEXT !
[    10] Failed EGL API GetProcAddress: eglQueryDmaBufModifiersEXT !
[    11] Failed EGL API GetProcAddress: eglPatchID !
[    12] Failed ES_Common API GetProcAddress: forward_glActiveTexture !
[    13] Failed ES_Common API GetProcAddress: forward_glBindBuffer !
[    14] Failed ES_Common API GetProcAddress: forward_glBindTexture !
[    15] Failed ES_Common API GetProcAddress: forward_glBlendFunc !
[    16] Failed ES_Common API GetProcAddress: forward_glBufferData !
[    17] Failed ES_Common API GetProcAddress: forward_glBufferSubData !
[    18] Failed ES_Common API GetProcAddress: forward_glClear !
[    19] Failed ES_Common API GetProcAddress: forward_glClearColor !
[    20] Failed ES_Common API GetProcAddress: forward_glClearDepthf !
[    21] Failed ES_Common API GetProcAddress: forward_glClearStencil !
[    22] Failed ES_Common API GetProcAddress: forward_glColorMask !
[    23] Failed ES_Common API GetProcAddress: forward_glCompressedTexImage2D !
[    24] Failed ES_Common API GetProcAddress: forward_glCompressedTexSubImage2D !
[    25] Failed ES_Common API GetProcAddress: forward_glCopyTexImage2D !
[    26] Failed ES_Common API GetProcAddress: forward_glCopyTexSubImage2D !
[    27] Failed ES_Common API GetProcAddress: forward_glCullFace !
[    28] Failed ES_Common API GetProcAddress: forward_glDeleteBuffers !
[    29] Failed ES_Common API GetProcAddress: forward_glDeleteTextures !
[    30] Failed ES_Common API GetProcAddress: forward_glDepthFunc !
[    31] Failed ES_Common API GetProcAddress: forward_glDepthMask !
[    32] Failed ES_Common API GetProcAddress: forward_glDepthRangef !
[    33] Failed ES_Common API GetProcAddress: forward_glDisable !
[    34] Failed ES_Common API GetProcAddress: forward_glDrawArrays !
[    35] Failed ES_Common API GetProcAddress: forward_glDrawElements !
[    36] Failed ES_Common API GetProcAddress: forward_glEnable !
[    37] Failed ES_Common API GetProcAddress: forward_glFinish !
[    38] Failed ES_Common API GetProcAddress: forward_glFlush !
[    39] Failed ES_Common API GetProcAddress: forward_glFrontFace !
[    40] Failed ES_Common API GetProcAddress: forward_glGenBuffers !
[    41] Failed ES_Common API GetProcAddress: forward_glGenTextures !
[    42] Failed ES_Common API GetProcAddress: forward_glGetBooleanv !
[    43] Failed ES_Common API GetProcAddress: forward_glGetBufferParameteriv !
[    44] Failed ES_Common API GetProcAddress: forward_glGetError !
[    45] Failed ES_Common API GetProcAddress: forward_glGetFloatv !
[    46] Failed ES_Common API GetProcAddress: forward_glGetIntegerv !
[    47] Failed ES_Common API GetProcAddress: forward_glGetPointerv !
[    48] Failed ES_Common API GetProcAddress: forward_glGetString !
[    49] Failed ES_Common API GetProcAddress: forward_glGetTexParameterfv !
[    50] Failed ES_Common API GetProcAddress: forward_glGetTexParameteriv !
[    51] Failed ES_Common API GetProcAddress: forward_glHint !
[    52] Failed ES_Common API GetProcAddress: forward_glIsBuffer !
[    53] Failed ES_Common API GetProcAddress: forward_glIsEnabled !
[    54] Failed ES_Common API GetProcAddress: forward_glIsTexture !
[    55] Failed ES_Common API GetProcAddress: forward_glLineWidth !
[    56] Failed ES_Common API GetProcAddress: forward_glPixelStorei !
[    57] Failed ES_Common API GetProcAddress: forward_glPolygonOffset !
[    58] Failed ES_Common API GetProcAddress: forward_glReadPixels !
[    59] Failed ES_Common API GetProcAddress: forward_glSampleCoverage !
[    60] Failed ES_Common API GetProcAddress: forward_glScissor !
[    61] Failed ES_Common API GetProcAddress: forward_glStencilFunc !
[    62] Failed ES_Common API GetProcAddress: forward_glStencilMask !
[    63] Failed ES_Common API GetProcAddress: forward_glStencilOp !
[    64] Failed ES_Common API GetProcAddress: forward_glTexImage2D !
[    65] Failed ES_Common API GetProcAddress: forward_glTexParameterf !
[    66] Failed ES_Common API GetProcAddress: forward_glTexParameterfv !
[... 42 lines cut ...]
[66384:66410:0111/071252.185627:ERROR:gl_context_egl.cc(173)] eglBindApi failed with error EGL_BAD_PARAMETER
[66384:66410:0111/071252.185664:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[66384:66410:0111/071252.185686:ERROR:shared_image_stub.cc(552)] SharedImageStub: unable to create context
[66384:66410:0111/071252.185716:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub
[66384:66410:0111/071252.185794:ERROR:gl_context_egl.cc(173)] eglBindApi failed with error EGL_BAD_PARAMETER
[66384:66410:0111/071252.185819:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[66384:66410:0111/071252.185839:ERROR:shared_image_stub.cc(552)] SharedImageStub: unable to create context
[66384:66410:0111/071252.185863:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[66384:66410:0111/071252.287988:ERROR:gl_context_egl.cc(173)] eglBindApi failed with error EGL_BAD_PARAMETER
[66384:66410:0111/071252.288099:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[66384:66410:0111/071252.288130:ERROR:shared_image_stub.cc(552)] SharedImageStub: unable to create context
[66384:66410:0111/071252.288178:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub
[66384:66410:0111/071252.290791:ERROR:gl_context_egl.cc(173)] eglBindApi failed with error EGL_BAD_PARAMETER
[66384:66410:0111/071252.290884:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[66384:66410:0111/071252.290909:ERROR:shared_image_stub.cc(552)] SharedImageStub: unable to create context
[66384:66410:0111/071252.290951:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
```

msisov, 2022-01-11:
probably, one of the bugs related to SwAngle I fixed not that much time ago. Just for the sake of tests, can you compile without that feature - 

angle_enable_vulkan = false
angle_enable_swiftshader = false

coldspark29, 2022-01-11:
Will try tonight. Where do I add those arguments?

rakuco, 2022-01-11:
> Where do I add those arguments?

They're GN arguments, so you can e.g. add them to `GN_ARGS` in `chromium-gn.inc` or append to the variable in your `local.conf`.

coldspark29, 2022-01-11:
The build failed with an overwhelmingly massive error message

https://pastebin.com/1dHvPEhV

msisov, 2022-01-12:
Please add 

`angle_enable_null = true`

coldspark29, 2022-01-12:
Still getting that huge error

https://pastebin.com/uMi1MqRJ

JeroenHid, 2022-01-19:
See https://github.com/OSSystems/meta-browser/issues/583#issuecomment-978963620

I've tried clang 13, and concluded "Stack smashing errors have indeed disappeared when using clang 13".
It was a while ago, but maybe it's something to try ?

coldspark29, 2022-01-31:
I will try soon. Am dealing with other problems atm

closed 2022-11-21 by jclsn