OSSystems/meta-browser #607: Chromium 97 - eglCreateContext failed with error EGL_BAD_MATCH

barndoor2276, 2022-02-22:
We just upgraded our meta-browser layer from v90 to v97 and started using clang 12 in order to successfully compile. Chrome launches but displays nothing and GPU doesn't start properly. Error below is repeated thousands of times until chrome is forcibly closed.

Using poky/OE @ dunfell branch.
Using meta-clang @ dunfell-clang12 branch.
Arch: Arm
Machine: imx6q-dms-ba16

[2300:2300:0222/170633.715923:ERROR:gl_surface_egl.cc(783)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[2300:2300:0222/170633.716250:ERROR:gl_context_egl.cc(297)] eglCreateContext failed with error EGL_BAD_MATCH
[2300:2300:0222/170633.716507:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[2300:2300:0222/170633.716752:ERROR:shared_image_stub.cc(552)] SharedImageStub: unable to create context
[2300:2300:0222/170633.717045:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub
[2300:2300:0222/170633.761588:ERROR:gl_surface_egl.cc(783)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[2300:2300:0222/170633.761971:ERROR:gl_context_egl.cc(297)] eglCreateContext failed with error EGL_BAD_MATCH
[2300:2300:0222/170633.762321:ERROR:gpu_channel_manager.cc(819)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[2300:2300:0222/170633.762658:ERROR:shared_image_stub.cc(552)] SharedImageStub: unable to create context
[2300:2300:0222/170633.762930:ERROR:gpu_channel.cc(568)] GpuChannel: Failed to create SharedImageStub

jwzl, 2022-03-01:
when I disabled the gpu, it has same issue!

eodabash, 2022-04-15:
I recently ran into a similar problem (eglCreateContext w/shared context failing with EGL_BAD_MATCH) and came across this github issue while searching for solutions. My issue turned out to be that the context I was trying to share with was created with EGL_CONTEXT_OPENGL_NO_ERROR_KHR, but I was not including the same flag with my new context attributes.