OSSystems/meta-browser #609: Layer clang-layer

MonsieurHBA, 2022-02-28:
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/bx1/yocto/poky/rpi-build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 7485 at 2022-02-28 11:09:24.271474 ---
ERROR: Layer clang-layer is not compatible with the core layer which only supports these series: dunfell (layer is compatible with kirkstone)

rakuco, 2022-02-28:
Looks like you need to switch your meta-clang checkout to the clang12 branch (`cd meta-clang && git checkout -b clang12 origin/clang12`, for example).

MonsieurHBA, 2022-02-28:
ERROR: Layer clang-layer is not compatible with the core layer which only supports these series: dunfell (layer is compatible with kirkstone)

i try to add dunfell in meta-clang/conf/layer.conf   : LAYERSERIES_COMPAT_clang-layer = "dunfell kirkstone" and i got an other error

rakuco, 2022-02-28:
See my comment above for how you should try to fix this.

MonsieurHBA, 2022-02-28:
yes i did but i got the same ERROR :  ERROR: Layer clang-layer is not compatible with the core layer which only supports these series: dunfell (layer is compatible with kirkstone)

rakuco, 2022-02-28:
To confirm: is your `meta-clang` repository at commit `e24b1235e82083d2088df5f8f93b1277daf17ff2`?

MonsieurHBA, 2022-02-28:
how can i find the commit of my meta-clang

rakuco, 2022-02-28:
`cd /path/to/meta-clang && git log -1`, for example.

MonsieurHBA, 2022-02-28:
commit 6d4ebdff6099f80716acc99efc3e49b8e9ea9a39

rakuco, 2022-02-28:
That commit is part of master, not the clang12 branch, which indicates you haven't switched branches as indicated above.

MonsieurHBA, 2022-02-28:
how can i solve the problem , please HELP!

rakuco, 2022-02-28:
Sorry but I've already said how it's supposed to be solved above. I feel it's important that you familiarize yourself with git more and understand how branches work and how to switch between them. See https://git-scm.com/book/en/v2 and https://docs.github.com/en for example.

closed 2022-02-28 by rakuco