OSSystems/meta-browser #610: chromium will be blocked when open multi-tab after some time

jwzl, 2022-03-01:
I have update chromium to M97,
and I have removed  libEGL.so and libGLESv2.so in /usr/lib/chromium/.
when I open multi-tab after some time, chromium will be blocked, and the message as below:
 
[1182:1207:0301/032855.493733:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.812318:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.813291:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.814118:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.814842:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.815541:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.816238:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.817012:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.817792:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.818573:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.819533:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.820535:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.821484:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.822652:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.823626:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.825705:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.827348:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1207:0301/032855.828201:ERROR:shared_image_manager.cc(226)] SharedImageManager::ProduceSkia: Trying to Produce a Skia representation from a non-existent mailbox.
[1182:1235:0301/032857.092323:ERROR:wayland_event_watcher.cc(265)] Fatal Wayland communication error Broken pipe.

Best Regards
Qing

jwzl, 2022-03-02:
@msisov  Thanks!

jwzl, 2022-03-03:
@msisov hello

msisov, 2022-03-03:
We have improved error logging recently and some patches are required to be applied. Otherwise, I can't seem to see where the problem is in the log.

Will you be able to apply https://chromium-review.googlesource.com/c/chromium/src/+/3386963, https://chromium-review.googlesource.com/c/chromium/src/+/3419856 and https://chromium-review.googlesource.com/c/chromium/src/+/3447304.

If you can't, I will include those in the coming meta-browser update.

jwzl, 2022-03-04:
okay, I will try it, thanks!

Regards,

jwzl, 2022-03-09:
I has meet  some build error? Do you has update these patch into M98 ? @msisov

msisov, 2022-03-09:
nope, it'll part of m99

jwzl, 2022-03-11:
@msisov 
Dear msisov：
let's changed the description:
that is when I play the video on yutube, it will crash after some times!

and log as below:
[putty2.log](https://github.com/OSSystems/meta-browser/files/8231123/putty2.log)

msisov, 2022-03-11:
tbh, I don't have any ideas why creation of egl context fails for you. support on embedded devices has always been complicated and chromium is not officially tested on these devices (except android).