OSSystems/meta-browser #612: Error in progress of building core-image-weston with meta-chromium and meta-clang (dunfell branch )

MonsieurHBA, 2022-03-02:
Hello guys :+1: 
So is that will affect my image or chromium will not work after build ?

[image]

MonsieurHBA, 2022-03-02:
[image]
this my config file

rakuco, 2022-03-02:
Sorry, but we don't maintain that recipe and can't help you further with that. Please check who the maintainer of that recipe is and what the instructions for filing bugs is. Just one side note: meta-clang needs to track the `clang12` branch, not the `dunfell` branch. Good luck!

closed 2022-03-02 by rakuco

kraj, 2022-03-02:
yeah I think we need to fix other `dunfell` layers to build with clang12. 
@MonsieurHBA can you add
```
CFLAGS:append:pn-kbd:toolchain-clang = " -Wno-error"
```

to `meta-clang/conf/nonclangable.conf`