OSSystems/meta-browser #614: Error chromium-ozone-wayland

MonsieurHBA, 2022-03-05:
ERROR: chromium-ozone-wayland-97.0.4692.71-r0 do_compile: Execution of '/home/bx1/yocto/poky/rpi-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/chromium-ozone-wayland/97.0.4692.71-r0/temp/run.do_compile.17090' failed with exit code 1
ERROR: Logfile of failure stored in: /home/bx1/yocto/poky/rpi-build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/chromium-ozone-wayland/97.0.4692.71-r0/temp/log.do_compile.17090
ERROR: Task (/home/bx1/yocto/poky/meta-browser/meta-chromium/recipes-browser/chromium/chromium-ozone-wayland_97.0.4692.71.bb:do_compile) failed with exit code '1'
[image]
[image]

[image]

Hello guys ..
can someone help me to fix that , 
I use core-image-weston and I want to add chromium there.

nros, 2023-01-25:
There is no way to detect the cause of the issue with the limited information you have provided. So, I just give some general advices how you might proceed to find it yourself.

1. It is better to post the texts/file-contents instead of images. It enables others to copy those and try them for you. Please use "source code"-block to add the files or texts. Don't external URLs as those usually vanish too fast.
2. The provided error is just a general error, that the subcommand failed. The cause of the failure is not on display and you need to figure that out first. For debugging, open a "development shell" for the package:
    ```sh
    bitbake chromium-ozone-wayland -c devshell
    ```
    This opens a `screen`-based separate shell inside the package build directory with all necessary environment variables set, which you can leave by pressing `CRTL+D` (=closing the input stream "STDIN"). Now you should execute the command that has failed as displayed by `ninja` (I won't manually type it, though, as reading from screenshot is too long and too error prune! Just copy and paste from your screen):
    ```sh
    clang++ -MMD ......
    ```
    Now you will see the real cause of the failure as a starting point for more scrutiny. If you want to try your fixes, just run the build process manually with the command:
    ```sh
    ninja
    ```

Since this issue is very old, you might consider closing it now :)

MonsieurHBA, 2023-01-25:
Hii,

Thank you for your feedback, but this issue has been resolved, you can
close it please.
Thank you again!
Have a great day

Best regards
Houssem.

nros, 2023-01-25:
@MonsieurHBA  I think you need to close it - I lack required permissions as I am not involved in this project - I am just a "consumer".

closed 2023-01-25 by rakuco