OSSystems/meta-browser #618: Chromium 98, WIP?

DeHuilendeHacker, 2022-03-11:
Hi there,

Thanks to all the current and previous maintainers of this repository for the hard work and concurrent updates.

I'm using meta-chromium right now on a CM4 with the Dunfell LTS release with custom distro and machine configuration.
Integrating Chromium went quite well, only one issue left to solve..

But the reason I'm writing this issue is that I was wondering what kind of release strategy there now is? 

Are all commits to master stable and tested, or is nothing tested at all?
The most recent branch named to one of the yocto releases is "zeus", are the more recent versions not supported or will there just no new releases on yocto-named-branches?  

I noticed the last commit of @msisov being called WIP, what stands for WorkInProgress for as far as I know, that made me wonder the above.

Thanks again

rakuco, 2022-03-11:
Hi there.

> Are all commits to master stable and tested, or is nothing tested at all?

We try to create branches only when necessary (i.e. when something stops working with master etc), so by default everything is tested in master and it _should_ be stable. We always build version updates to master with multiple branches, and some of those builds are also run.

> I noticed the last commit of @msisov being called WIP, what stands for WorkInProgress for as far as I know, that made me wonder the above.

That was a hiccup on our side :( The pull request was originally a WIP one, which later became ready to be merge d but we forgot to update the commit message title (see the discussion in #611).

kraj, 2022-03-11:
> Hi there,
>
> Thanks to all the current and previous maintainers of this repository for the hard work and concurrent updates.
>
> I'm using meta-chromium right now on a CM4 with the Dunfell LTS release [...]

good progress

> But the reason I'm writing this issue is that I was wondering what kind of release strategy there now is?
>
> Are all commits to master stable and tested, or is nothing tested at all? The most recent [...]

testing chromium is a big ordeal since it requires a beefy build machine and hours of build time. All of us do not have access to such machines so we pool testing for limited platforms. These are usually listed in the pull to do major release bumps. For same reason, we only test latest release and maintained LTS, which as of now means `dunfell` and future `kirkstone` and latest release is `honister`, some of us do limited test it with master branch of yocto as well.  For same reason, we try to not branch because the test matrix doubles and again resources are limited so we try to maintain master branch across yocto releases. 

> I noticed the last commit of @msisov being called WIP, what stands for WorkInProgress for as far as I know, that made me wonder the above.

That commit was ready to commit but @msisov forgot to edit the PR subject, so do not read too much into it. We will also mark the pulls as `drafts` when they go through testing and validation in future.

> 
> Thanks again

DeHuilendeHacker, 2022-03-11:
Thanks for the quick updates! 
Everything is clear to me now, also no more issues and nothing to be solved on my chromium build right now :)

Quite a heavy task for my notebook, so I see there is not enough power (or time) to compile and evaluate :)

Thanks again! 
Will keep using the most recent master every now and then, to see if something comes up at my end!

closed 2022-03-11 by DeHuilendeHacker