OSSystems/meta-browser #624: chrome_crashpad_handler: symbol lookup error: /lib/libc++abi.so.1: undefined symbol: __cxa_end_cleanup

MastaG, 2022-04-02:
I'm building chromium-ozone-wayland using the honister branches of openembedded-core and meta-openembedded and meta-clang with bitbake 1.52.

But after upgrading my OS to Fedora 36 (armv7) it doesn't seem to run any longer.
The crashpad_handler seems to fail with:
```
/usr/local/lib/chromium-wayland/chrome_crashpad_handler: symbol lookup error: /lib/libc++abi.so.1: undefined symbol: __cxa_end_cleanup
```
Now it seems honister builds using LLVM/clang 13, but Fedora 36 seems to ship mixed versions:
```
libcxxabi-14.0.0-1.fc36.armv7hl
libcxx-14.0.0-1.fc36.armv7hl
llvm-libs-14.0.0-1.fc36.armv7hl
llvm13-libs-13.0.1-1.fc36.armv7hl
```
It does provide version 13 for the LLVM libraries, but libcxx is only available as version 14 now.
But it seems the library does contain the symbol:
```
$ grep __cxa_end_cleanup /lib/libc++abi.so.1
grep: /lib/libc++abi.so.1: binary file matches
$ rpm -qf /lib/libc++abi.so.1
libcxxabi-14.0.0-1.fc36.armv7hl
```

Any ideas on how to solve this?

kraj, 2022-04-02:
are you building chromium using OE/Yocto and then running it with fedora runtime ? If so, I am afraid this is not much tested and even supported combo. You should also use runtime from the oe/yocto builds. We have no control over how fedora would have compiled with llvm runtime.

MastaG, 2022-04-03:
Yes that's correct. I only use oe/yocto to build it and then move it to my Fedora installation.
Would it be possible to link libc++ and libc++abi statically instead?
I believe the original Chrome browser does this as well?

kraj, 2022-04-03:
> Yes that's correct. I only use oe/yocto to build it and then move it to my Fedora installation. Would it be possible to link libc++ and libc++abi statically instead? I believe the original Chrome browser does this as well?

that is possible and perhaps what we could do as long as we don't call C++ system libraries we should be fine, although this will mean a bit of duplication of libraries in case folks are already using LLVM runtime with yocto

anyway, I have posted a patch to make this setting as a packageconfig. So you can try it out to enable this knob in settings and report back

https://github.com/OSSystems/meta-browser/pull/625

MastaG, 2022-04-04:
I first tried to rebuild everything (oe/yocto) from master (because meta-clang master builds LLVM/clang 14 which matches the runtime on my Fedora 36 host), but I still got the same ```symbol lookup error: /lib/libc++abi.so.1: undefined symbol: __cxa_end_cleanup``` error.
Fedora must have broken something related to their libc++ packages.

Then I tried your PR https://github.com/OSSystems/meta-browser/pull/625 and rebuild with ```custom-libcxx``` and it now runs fine again.
The only thing I noticed is that the the chromium binaries then require ```libunwind.so.1``` to load.
But Fedora seems to provide that from the ```llvm-libunwind-13.0.1-1.fc36.armv7hl``` package.
So I simply set: ```LD_PRELOAD=/usr/lib/llvm-unwind/libunwind.so.1``` and all is solved :+1:

closed 2022-04-11 by MastaG